zephyrproject-rtos/zephyr #35220: tests: dma: memory-to-memory transfer fails on stm32f746zg nucleo board

FRASTM, 2021-05-12:
**Describe the bug**
test cases `tests/drivers/dma/chan_blen_transfer`/ and `tests/drivers/dma/loop_transfer` fail on nucleo_f746zg board

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f746zg tests/drivers/dma/chan_blen_transfer/

**Expected behavior**
test PASSED

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - Commit SHA f05ea67

FRASTM, 2021-05-12:
FIXED by https://github.com/zephyrproject-rtos/zephyr/pull/35165

galak, 2021-05-18:
Need to validate if #35258 fixes this.

FRASTM, 2021-05-20:
The pb of the #35220 is **not** fixed with this PR. The
Because the the soc/arm/st_stm32/stm32f7 does not define the `CONFIG_INIT_ARCH_HW_AT_BOOT`
then the Cache memory  is not disabled here in the `arch/arm/core/aarch32/cortex_m/scb.c`

galak, 2021-05-20:
> The pb is **not** fixed with this PR #35258
> Because the the soc/arm/st_stm32/stm32f7 does not define the `CONFIG_INIT_ARCH_HW_AT_BOOT`
> then the Cache memory is not disabled here in the `arch/arm/core/aarch32/cortex_m/scb.c`

Shouldn't the cache get invalidated though in `z_arm_mpu_init()`?

FRASTM, 2021-05-20:
This is not enough since in the `soc/arm/st_stm32_f7` st_stm32f7_init  function already enables the cache.So that
1)  	`select INIT_ARCH_HW_AT_BOOT` in the soc/arm/st_stm32/stm32f7/Kconfig.series
2) remove the SCB_EnableICache/SCB_EnableDCache  in the soc.c of the stm32F7 (and probalbly also of the stm32H7

galak, 2021-05-20:
@FRASTM I dont follow.  It seems if I'm reading the code correctly that `z_arm_mpu_init()` will get called before `st_stm32f7_init`.

FRASTM, 2021-05-27:
FIXED by #35165
tests are now PASSED

closed 2021-05-27 by FRASTM