zephyrproject-rtos/zephyr #35229: twister log mixing between tests

ABOSTM, 2021-05-12:
**Describe the bug**
Running stm32 automatic test bench, sometimes, log are mixed between tests:
for example log of gpio_api_1pin test shows uart results (see log below)
Problem is not systematic and occurrence among the different tests/samples seems quite low: for now, only gpio_api_1pin test has been identified as failed, probably because this is visible only when test is failed (when test is passed nobody check the logs)
But occurrence seems to be higher since 1 or 2 weeks, with almost 1 board failed each day out of 14 boards available on the bench.

This issue looks very similar to #25719

**To Reproduce**
Steps to reproduce the behavior:
1. run daily tests on STM32 bench (I know it is not accessible, but I have no simple way to reproduce)
2.  command is:
twister --ninja --runtime-artifact-cleanup --hardware-map /local/ztb/zephyrproject/zephyr/../map.yaml --device-testing --outdir /local/ztb/zephyr-test-bench/../logs/210512/log_dt_master_fb88c77ac2_0941
3. See error

**Logs and console output**
```
$ cat nucleo_g071rb/tests/drivers/gpio/gpio_api_1pin/peripheral.gpio.1pin/handler.log

===================================================================
START - test_forever_timeout
W: RX was already enabled
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:442: test_forever_timeout: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
FAIL - test_forever_timeout in 3.120 seconds
===================================================================
Test suite uart_async_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA fb88c77ac2

PerMac, 2021-05-17:
@ABOSTM  In your example, do you see the same error reported for the actual failing test suite (here tests/drivers/uart/uart_async_api)? If yes, it is possible that the previous app was not erased due to some flashing error and the next test (here the one from tests/drivers/gpio/gpio_api_1pin) will again see the same output.

ABOSTM, 2021-05-17:
@PerMac,
goo point. 
No uart_async_api is not reporting failure, but the real reason for this is that log show truncated (1st line is incomplete) wdt_basic_api output which is successful :
```
st suite wdt_basic_test
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_bad_window_max
PASS - test_wdt in 0.8 seconds
===================================================================
Test suite wdt_basic_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```
So I look at wdt_basic_api log which seems correct and complete

```
Running test suite wdt_basic_test
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Testcase passed
Testcase: test_wdt_bad_window_max
PASS - test_wdt in 0.8 seconds
===================================================================
Test suite wdt_basic_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

Finally, this is wdt_basic_api log which is partially duplicated, creating some kind of shift in logs versus tests.
But among all the logs, I could not found reference to gpio testing. So it seems shift ends at gpio test.

PerMac, 2021-05-17:
I think this is the issue I was facing: https://github.com/zephyrproject-rtos/zephyr/issues/33509
I ended up deleting watchdog test and this sample from my tree before executing test suite, so no other tests are corrupted.
I also added a quarantine feature to the twister, because this was driving me crazy. You can add this test to quarantine and see if the issue is gone. https://docs.zephyrproject.org/latest/guides/test/twister.html#quarantine

ABOSTM, 2021-05-17:
@PerMac, Thanks it is really interesting, I will study that

PR #36003 (2021-06-08): tests: drivers: watchdog: Pause during debugging
Enable the option to pause the watchdog if the MCU is halted by a
debugger.

This is similar to https://github.com/zephyrproject-rtos/zephyr/pull/35484
fixes #35229
closed 2021-06-08 by nashif