zephyrproject-rtos/zephyr #35236: tests: doc: Document generation process FAILS with valid module documentation

gregshue, 2021-05-12:
**Describe the bug**
When building standard Zephyr documentation within an `example-application` (external manifest+module) workspace,
the Zephyr documentation generation process reports "FAILED: CMakeFiles/html" with the summary statement "free(): invalid pointer".  

It does not fail if the `samples:` configuration is removed from `example-application/zephyr/module.yml`. 

**To Reproduce**
Steps to reproduce the behavior:
west init -m https://github.com/gregshue-rtos/example-application --mr feature/33962_subsys_module_hello my-workspace
cd my-workspace
west update
cd zephyr
source zephyr-env.sh
cd doc
make

**Expected behavior**
I expect the document generation process to produce the same output as when generating documentation after copying the (2) sources of documentation into the parallel location within the Zephyr repository itself.

**Impact**
As a Zephyr user, this is a central to my development and product artifact-generation processes. I need to be able to generate at least independent documentation for my local tweaked fork of Zephyr.  The presence of the application module in the workspace should not break that, especially in the LTS.

**Logs and console output**
```
zephyr@zephyr-VirtualBox:~/exapp_ws/zephyr/doc$ make htmldocs-fast
make htmldocs KCONFIG_TURBO_MODE=1
make[1]: Entering directory '/home/zephyr/exapp_ws/zephyr/doc'
cmake -GNinja -B_build -S. -DDOC_TAG=development \
	-DSPHINXOPTS=-q \
	-DKCONFIG_TURBO_MODE=1
-- Zephyr base: /home/zephyr/exapp_ws/zephyr
-- West: /home/zephyr/.local/bin/west (west --version: "West version: v0.11.0")
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.6") found components: Interpreter 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.18") found components: doxygen 
-- Found LATEX: /usr/bin/latex   
-- Zephyr version: 2.6.0-rc1 (/home/zephyr/exapp_ws/zephyr)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zephyr/exapp_ws/zephyr/doc/_build
cmake --build _build -- htmldocs  # -v # VERBOSE=1
[0/4] Running gen_devicetree_rest.py /home/zephyr/exapp_ws/zephyr/doc/_build/rst/doc/reference/devicetree
[4/4] Generating HTML documentation with /home/zephyr/.local/bin/sphinx-build -q
FAILED: CMakeFiles/html 
cd /home/zephyr/exapp_ws/zephyr/doc && /usr/bin/cmake -E env ZEPHYR_BASE=/home/zephyr/exapp_ws/zephyr ZEPHYR_BUILD=/home/zephyr/exapp_ws/zephyr/doc/_build DOXYGEN_EXECUTABLE=/usr/bin/doxygen /home/zephyr/.local/bin/sphinx-build -W -N -t development -b html -d /home/zephyr/exapp_ws/zephyr/doc/_build/doctrees -q /home/zephyr/exapp_ws/zephyr/doc/_build/rst/doc /home/zephyr/exapp_ws/zephyr/doc/_build/html
free(): invalid pointer
Child aborted
ninja: build stopped: subcommand failed.
make[1]: *** [Makefile:19: htmldocs] Error 1
make[1]: Leaving directory '/home/zephyr/exapp_ws/zephyr/doc'
make: *** [Makefile:16: htmldocs-fast] Error 2
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used
   https://github.com/gregshue/example-application@7cd4d7ebcfd9e488d15e46e3f0666642da0e21eb
   which imports https://github.com/zephyrproject-rtos/zephyr@929c573d31d9c4a09f92b0f40cd4fc0ae1c49c69

**Additional context**

gmarull, 2021-05-19:
the provided steps can't be reproduced as the repository doesn't exist

gregshue, 2021-05-20:
@gmarull, strange. I verified the repo has public visibility and is readable from my wife's computer.  The link without the @# works.  You will find that commit at the HEAD of branch `bug/35236_doc_gen_FAILS`.  What makes you think the *repository* doesn't exist?

gmarull, 2021-05-20:
I can't reproduce the issue following the listed steps:
```
(.venv) ➜  doc git:(aa8e34302a) make         
make htmldocs KCONFIG_TURBO_MODE=1
make[1]: Entering directory '/home/gmarull/ws/tmp/my-workspace/zephyr/doc'
cmake -GNinja -B_build -S. -DDOC_TAG=development \
        -DSPHINXOPTS=-q \
        -DKCONFIG_TURBO_MODE=1
-- Zephyr base: /home/gmarull/ws/tmp/my-workspace/zephyr
-- West: /home/gmarull/ws/tmp/my-workspace/zephyr/.venv/bin/west (west --version: "West version: v0.11.0")
-- Found Python3: /home/gmarull/ws/tmp/my-workspace/zephyr/.venv/bin/python3 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen 
-- Found LATEX: /usr/bin/latex   
-- Zephyr version: 2.6.0-rc1 (/home/gmarull/ws/tmp/my-workspace/zephyr)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gmarull/ws/tmp/my-workspace/zephyr/doc/_build
cmake --build _build -- htmldocs  # -v # VERBOSE=1
[0/4] Running gen_devicetree_rest.py /home/gmarull/ws/tmp/my-workspace/zephyr/doc/_build/rst/doc/reference/devicetree
[4/4] Generating HTML documentation with /home/gmarull/ws/tmp/my-workspace/zephyr/.venv/bin/sphinx-build -q
make[1]: Leaving directory '/home/gmarull/ws/tmp/my-workspace/zephyr/doc'
(.venv) ➜  doc git:(aa8e34302a) 
```

I'd suggest to build the docs in a isolated Python venv, ie

```
python3 -m venv .venv
source .venv/bin/activate
pip install -U pip # make sure using latest pip!
pip install -r scripts/requirements-base.txt
pip install -r scripts/requirements-doc.txt
cd doc
make
```

gregshue, 2021-05-26:
@gmarull After reproducing the issue 3x prior to submission, I am no longer able to reproduce the issue (without even attempting the venv).  Thank you for looking in to this.

closed 2021-05-26 by gregshue