zephyrproject-rtos/zephyr #35237: build: Enhance twister to follow all module.yml in module list

gregshue, 2021-05-12:
**Is your enhancement proposal related to a problem? Please describe.**
`twister` seems to only look for tests and samples beneath ZEPHYR_BASE and wherever it is explicitly told on the command line. It seems to ignore the optional `tests:` and `subsys:` list settings in `zephyr/module.yml`.  This prevents automatically running CI on the complete workspace.  Doing so is highly desirable for user-provided CI.

**Describe the solution you'd like**
`twister` should default to searching all modules for tests and samples as configured by each `zephyr/module.yml`.  A command line option should be added to support looking only over specific module(s).

**Describe alternatives you've considered**
I have not considered any other options, as this appears to be a gap in support of existing extensibility mechanisms.

**Additional context**
Please add this in or before the LTS.

nashif, 2021-06-23:
```
./scripts/zephyr_module.py --twister-out t.args
./scripts/twister +t.args
```

we do not want twister dealing with manifests directly, this is up to the user to run tests on modules with tests.

gregshue, 2021-06-23:
@nashif 

> we do not want twister dealing with manifests directly,  ...

I completely agree, and this fundamentally has nothing to do with manifests.  Manifest parsing should be left to `west`, and any ability to kickoff off `twister` with a generated list of module directories from manifests should be left to a west extension located in `zephyr/scripts/west_commands/`.

This is about `twister`:
1. following a provided list of directories (e.g., `ZEPHYR[_EXTRA]_MODULES, or a new command line option)
2. detecting and validating any existing `zephyr/module.yml` 
3. following the _existing_ `tests:` and `samples:` in valid `zephyr/module.yml` to determine where to (and where not to) look for testcase_base values

Since I haven't been able to think of any other client that would use the `tests:` field in `module.yml`, this seems like `twister` was missed in the original implementation.

nashif, 2021-06-23:
I still think twister should not deal with modules directly. We should have and we have the tools to cover modules in CI, but we do not want to be parsing module.yaml files in twister. If you want to test something in a module, point to it directly.

nashif, 2022-11-19:
we have a method for testing module with twister which is part of CI. twister can be pointed to any directory with tests already.