zephyrproject-rtos/zephyr #35244: tests/crypto/mbedtls/crypto.mbedtls build failure for native_posix

yashi, 2021-05-13:
**Describe the bug**
While I was testing mbedtls for #32667, I found that `tests/crypto/mbedtls/crypto.mbedtls` failed to build for native_posix. I couldn't find a good starting point for `git bisect`.  v2.4 and v2.5 failed for another reason.

**To Reproduce**
Steps to reproduce the behavior:
```
$ scripts/twister -N -p native_posix -T tests/crypto/mbedtls -v -j3
INFO    - Zephyr version: v2.6.0-rc1-73-gd3ae4e9aa392
INFO    - JOBS: 3
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 2 test scenarios (2 configurations) selected, 1 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/1 native_posix              tests/crypto/mbedtls/crypto.mbedtls                FAILED Build failure (native)
ERROR   - see: /home/yashi/src/zephyr-project/zephyr/twister-out/native_posix/tests/crypto/mbedtls/crypto.mbedtls/build.log

INFO    - 0 of 1 test configurations passed (0.00%), 1 failed, 1 skipped with 0 warnings in 3.69 seconds
INFO    - In total 1 test cases were executed, 1 skipped on 1 out of total 373 platforms (0.27%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing xunit report /home/yashi/src/zephyr-project/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/yashi/src/zephyr-project/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

```

**Expected behavior**
`tests/crypto/mbedtls/crypto.mbedtls` pass its tests.

**Logs and console output**
```
$ tail twister-out/native_posix/tests/crypto/mbedtls/crypto.mbedtls/build.log
[180/188] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[181/188] Linking C static library modules/mbedtls/libmodules__mbedtls.a
[182/188] Linking C static library zephyr/kernel/libkernel.a
[183/188] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map 
: && ccache /usr/lib/ccache/gcc   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/home/yashi/src/zephyr-project/zephyr/twister-out/native_posix/tests/crypto/mbedtls/crypto.mbedtls/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/posix/core/libarch__posix__core.a  zephyr/soc/posix/inf_clock/libsoc__posix__inf_clock.a  zephyr/boards/posix/native_posix/libboards__posix__native_posix.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/posix/core/offsets/offsets.c.obj  -L/home/yashi/src/zephyr-project/zephyr/twister-out/native_posix/tests/crypto/mbedtls/crypto.mbedtls/zephyr  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -Wl,--fatal-warnings  -m32  -ldl  -pthread  -lm && cd /home/yashi/src/zephyr-project/zephyr/twister-out/native_posix/tests/crypto/mbedtls/crypto.mbedtls/zephyr && /usr/bin/cmake -E echo
/usr/bin/ld: app/libapp.a(main.c.obj): warning: relocation in read-only section `.text.test_main'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.0 alpha 1
 - Commit SHA: v2.6.0-rc1-73-gd3ae4e9aa3

galak, 2021-05-13:
Not able to reproduce this myself locally.  What distro are you running?

galak, 2021-05-13:
> Not able to reproduce this myself locally. What distro are you running?

Is this w/ or w/o the changes in the rand/srand PR?

yashi, 2021-05-13:
I'm running Debian Sid.  This is without the rand PR.
```
$ git describe
v2.6.0-rc1-73-gd3ae4e9aa3
```
```
$ west update
   :
=== updating mbedtls (modules/crypto/mbedtls):
HEAD is now at 5765cb7 mbedtls: update README with laterst version information
```
Attaching [build.log](https://github.com/zephyrproject-rtos/zephyr/files/6472405/build.log).

yashi, 2021-05-27:
https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=fa6ecf940581b4be26165351bb0473427d31c7d8
https://bugzilla.kernel.org/show_bug.cgi?id=204495

yashi, 2021-05-27:
It seems all tests fail to build for `native_posix` with binutils version 2.35.  But it's just that twister treats the warning and mark them `FAILED`.  You can build and run the tests without a problem if you aren't using twister.

d3zd3z, 2021-05-28:
I was not able to reproduce this on NixOS with binutils 2.35.1. I think one of us will have to setup a debian-sid environment to be able to reproduce.

closed 2021-08-11 by github-actions[bot]

yashi, 2021-09-27:
https://github.com/zephyrproject-rtos/zephyr/pull/38730#issuecomment-928049682

yperess, 2021-09-28:
The issue is 2 lines in the root CMakeLists.txt on Debian systems:

```
zephyr_cc_option(-fno-pie)
zephyr_cc_option(-fno-pic)
```

I just found a fix for it. Pushing a PR...

tejlmand, 2021-09-28:
it's most likely related to this binutils commit:
https://sourceware.org/git/?p=binutils-gdb.git;a=commit;h=fa6ecf940581b4be26165351bb0473427d31c7d8

but note, this is only a linker `warning`, but because all warnings are treated as errors per default when using twister the end result is an error.

yashi, 2021-10-01:
https://lists.zephyrproject.org/g/devel/topic/building_on_gentoo_creating_a/19873560

Found this mailing list archive.