zephyrproject-rtos/zephyr #35250: Twister is not reading the serial line output completely

ananthjaisankar95, 2021-05-13:
**Describe the bug**
We are trying to contribute for efr32bg22 from silicon labs. When we running twister , tests/kernel/timer/timer_api/kernel.timer.tickless test is failing. But if I run that using west, it is working fine. From the logs I can see that twister is missing "PROJECT EXECUTION SUCCESSFUL" line.

What have you tried to diagnose or workaround this issue?
I have put print(serial_line) after https://github.com/zephyrproject-rtos/zephyr/blob/f8ac3a49ec95f4bf98bbd1f2b827c421f654d48b/scripts/pylib/twister/twisterlib.py#L638 and I can see it is getting "PROJECT EXECUTION SUCCESSFUL" line. So, looks like twister is not reading the serial output properly that leads to mark the test as failed.

**Expected behavior**
tests/kernel/timer/timer_api/kernel.timer.tickless should pass

**Logs and console output**
Twister output: https://gist.github.com/ananthjaisankar95/97cb01a47388ca91d58eb5a782c9fbce
West output: https://gist.github.com/ananthjaisankar95/c550b9fb888f7f1237eb4bc74fe809e9

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain : gcc-arm-none-eabi-10-2020-q4-major
 - 2.5.99

ananthjaisankar95, 2021-05-24:
Hi,
 we have tried to debug this issue further and observed the below key points
 1) when we use twister to run the test case(tests/kernel/timer_api), after this line (https://github.com/zephyrproject-rtos/zephyr/blob/374629af906a24add294b1c3b945128cb6a486e8/subsys/testsuite/include/tc_util.h#L189) soc is entering into lower energy modes( UART wont work into lower energy modes) as CONFIG_PM is enabled in the test case.
2) As "PROJECT EXECUTION SUCCESSFUL" is not transmitted through UART because of UART wont working in lower energy modes, twister is timed out waiting for the message.
3) But the same test case is working and pushing "PROJECT EXECUTION SUCCESSFUL" through UART in west Build & Flash
@nashif @carlescufi

nashif, 2021-05-26:
You should not enable power management by default and for running tests. This has lots of side effects and the tests were not designed to be run with power management enabled. Does it work for you with PM disabled?

ananthjaisankar95, 2021-05-27:
Yes, The test is getting passed if PM is disabled. And FYI the particular testcase has COMFIG_PM=y in prj_tickless.conf itself and we did not add them.

closed 2021-08-10 by github-actions[bot]