zephyrproject-rtos/zephyr #35253: lvgl font generation

glory-man, 2021-05-13:
**Is your enhancement proposal related to a problem? Please describe.**
Lvgl-library support utf. Zephyr uses some pregenerated font-files based on montserrat font (need to note that montserrat_8.c & montserrat_10.c available for selection during zephyr config are absent). But those files contained glyphs only for base ascii symbols. 

**Describe the solution you'd like**
It will be helpfull to generate used in application font files during build time and also be able to configure additional characters to be included in the file (for example range 0x400-0x4FF for cyrillic, 0x100-0x17F for latin extended).

alexanderwachter, 2021-06-23:
@nashif no idea abou lvlg 😅
Sorry

glory-man, 2021-08-06:
LVGL sources includes font-generator script (built_in_font_gen.py), which could be executed in build-time and generate font c-files. This script has an argument '--range' which could be used to select range of generated symbols. Also we can change font-family (for example use roboto fonts instead of montserrat) using '--range' argument.

fabiobaltieri, 2023-08-23:
Hey this has been sitting for ages and reading about the last update I think we can close it down. Feel free to reopen if there's something actionable still.

closed 2023-08-23 by fabiobaltieri