zephyrproject-rtos/zephyr #35255: Don't fail on resubmit while delayed works are canceled

pdunaj, 2021-05-13:
**Is your enhancement proposal related to a problem? Please describe.**
In new delayable work implementation as long as work is in CANCELING state, the work cannot be submitted again.
For canceling to complete workqueue thread need to reach finalization routine and it's impossible to tell how much this can take, especially in preemptive configuration.

The problem may occur in the following scenario:
1. some protocol uses delayable work to set a timeout for request it wait for
2. request is fulfilled so delayable work is canceled
3. workqueue thread does not have the CPU to finalize the cancelation
4. new request arrives and work is submitted again - submission fails (silently or not - see further)
4a. timeout is K_NOWAIT - submission to queue occurs now and problem will be seen immediately (so module knows there was something wrong assuming an error is checked)
4b. timeout is not K_NOWAIT - resubmit will success as it can set work's timeout even if work is canceling; after this point there is a race between workqueue thread and timeout subsystem - if timeout hits before workqueue finalizes cancelation work will not be submitted and there will be no indication that something wrong happened.

In short the problem will hit when canceling process is not complete at the time work is actually being submitted to the queue. Since Zephyr is mostly used with cooperative threads, plus you would need to sufficiently long request timeout plus you will need to have CPU busy enough so canceling cannot complete, there is very slim chance it will be seen.
However since there is no bound for any of above we should assume that this can and will happen eventually.

**Describe the solution you'd like**
I think it would be best if submit to queue is allowed while cancel is in progress. It could work similarly to the case when work is in running state. In both running and canceling state such submission refers to the subsequent occurrence of work not the one that is "pending".

**Describe alternatives you've considered**
At the moment to be 100% sure everything goes well one would have to cancel_sync, that is block the thread until cancel is done. The block can be right after cancel or some additional thread could be spawned to track it. Both cases have serious drawbacks.
Alternatively you can simply avoid to use the cancel feature.

**Additional context**
N/A

pdunaj, 2021-05-13:
@pabigot

pabigot, 2021-05-13:
I don't think your use case is detailed enough to illuminate the potential solutions.  I tried to make it precise, and stalled at:

> Some unspecified local event **T** (trigger) occurs, after which an event **R** (response) is expected within time **D** (delay).  **T** events can occur asynchronously.  Exactly (?) one **R_i** event is expected for each **T_i** event, but delivery is not reliable so **R_i** may not be received, or it may show up "late" (more than **D** after **T_i**).  An **R_i** event does not (?) carry sufficient information to determine which **T_i** event it corresponds to.  Solution: On **T** schedule work after **D**; on **R** cancel work; work handler unconditionally recovers from failure to receive **R**.

But even this isn't enough.  Is it true that **R_i** cannot be associated with **T_i**?  Can **R_i** events be received out of order?  Is there a minimum interval between **T_i** events?  Can a new **T_j** (**j** > **i**) event be received before the expected time of **R_i**?

Depending on details a work queue entry may not be the best approach.  Certainly it's not obvious that canceling a work item is necessary: you may be able maintain a count or set of unreceived **R** events, reschedule the work item based on the longest outstanding delay, and have the work handler detect whether any **R_i** it was waiting for is too late.  Many ways to solve this, all depending on details.

In any case, here's some history that should inform a redesign to change the behavior again:

The new Zephyr workqueue API was [designed](https://github.com/zephyrproject-rtos/zephyr/pull/29618#issuecomment-718193197) to follow the Linux [concurrency managed workqueue](https://www.kernel.org/doc/html/latest/core-api/workqueue.html) with certain limits because Linux has much more capability than can be done in Zephyr (including per-CPU work pools).  The design was informed by these conclusions:

* Linux permits synchronous cancellation of delayable and non-delayable work items.
* Linux does not permit asynchronous cancellation of non-delayable work items.
* Linux permits ISR (asynchronous) attempts to cancel a delayable work item, but doesn't guarantee such a cancellation will complete: the work item may still be running.

A Linux work item that is cancelling may be "pending", which prevents any submission or schedule operation on the work item from being performed.  See https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/kernel/workqueue.c?h=v5.12.3#n607.  But that appearance may be misleading; the documentation is not clear on what must happen if work item that is being canceled is submitted or scheduled, and the implementation is too complex to easily infer the actual behavior (e.g. can a cancelling item not be pending?).

A large number of problems with Zephyr's work queue result from the decision to support cancelling work items that have been submitted, while failing to account for cases where the cancellation could not be performed immediately.  It's years to late to change that, but I strongly suspect that a better design would have justified cancelling a work item only in a situation where a subsystem that uses the work queue is being shut down.  In that case, you want to cancel, and confirm cancellation, to be sure it's safe to release any resources in the subsystem---or that it's safe to restart it.  This is what synchronous cancellation is for.

It comes down to a question: If you cancel a work item, do you mean to cancel all past submissions/schedules of that item, or do you want to force the work item to an idle state where its handler will not be invoked again?  These are very different goals.  For the latter, resubmission of a still-running item must be rejected otherwise it may never become idle.  For the former, you could probably accomplish the same thing by recording additional state specific to the work item and the events that cause it to be scheduled/submitted.

I don't currently believe it's necessary to change the work queue behavior, but if this is to be done it really should be based on a clear understanding of *all* use cases and alternative approaches. I don't think there's more that I can contribute to such a redesign.