zephyrproject-rtos/zephyr #35278: LittleFs Sample will not build for qemu_riscv64 sample target

bob2oneil, 2021-05-14:
For the sample code as zephyrproject/zephyr/samples/subsys/fs, while this project builds and
operates properly for the posix_native_64 board target, it will not build for the qemu_riscv64 board target.  

Perhaps it is not possible to emulate flash and the LittleFs within the QEMU environment?  If so, I want to understand why that is true and if there are any workarounds.

**To Reproduce**
Steps to reproduce the behavior:
1. Change to this projects directory
2. west build -p always -b qemu_riscv64 ./

**Expected behavior**
For the project build as it does for the native_posix_64 board target.

**Impact**
I need this capability to read-write persistent content.

Environment is Ubuntu 20 with Zephyr latest as 2.6.0.rc1

Build output is as follows:

robocom@scnuc-1:~/zephyrproject/zephyr/samples/subsys/fs/littlefs$ west build -p always -b qemu_riscv64 ./
-- west build: making build dir /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/robocom/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs
-- Zephyr version: 2.6.0-rc1 (/home/robocom/zephyrproject/zephyr), build: v2.6.0-rc1-61-g8eea9ebc3a88
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: qemu_riscv64
-- Cache files will be written to: /home/robocom/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.4 (/home/robocom/zephyr-sdk-0.12.4)
-- Found dtc: /home/robocom/zephyr-sdk-0.12.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/robocom/zephyrproject/zephyr/boards/riscv/qemu_riscv64/qemu_riscv64.dts
-- Generated zephyr.dts: /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build/zephyr/include/generated/device_extern.h
Parsing /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/Kconfig
Loaded configuration '/home/robocom/zephyrproject/zephyr/boards/riscv/qemu_riscv64/qemu_riscv64_defconfig'
Merged configuration '/home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/prj.conf'
Configuration saved to '/home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build/zephyr/.config'
Kconfig header saved to '/home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build/zephyr/include/generated/autoconf.h'

warning: FLASH_PAGE_LAYOUT (defined at drivers/flash/Kconfig:54) was assigned the value 'y' but got
the value 'n'. Check these unsatisfied dependencies: FLASH_HAS_PAGE_LAYOUT (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_FLASH_PAGE_LAYOUT.html and/or look up
FLASH_PAGE_LAYOUT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: FILE_SYSTEM_LITTLEFS (defined at subsys/fs/Kconfig.littlefs:6) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: FLASH_MAP (=n), FLASH_PAGE_LAYOUT (=n).
See http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_FILE_SYSTEM_LITTLEFS.html and/or
look up FILE_SYSTEM_LITTLEFS in the menuconfig/guiconfig interface. The Application Development
Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual
might be helpful too.

warning: FLASH_MAP (defined at subsys/storage/flash_map/Kconfig:10) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: FLASH_HAS_DRIVER_ENABLED (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_FLASH_MAP.html and/or look up
FLASH_MAP in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/robocom/zephyr-sdk-0.12.4/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build
-- west build: building application
[1/121] Preparing syscall dependency handling

[16/121] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /home/robocom/zephyr-sdk-0.12.4/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc -DBUILD_VERSION=v2.6.0-rc1-61-g8eea9ebc3a88 -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/home/robocom/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/robocom/zephyrproject/zephyr/soc/riscv/riscv-privilege/virt -I/home/robocom/zephyrproject/zephyr/soc/riscv/riscv-privilege/common/. -isystem /home/robocom/zephyrproject/zephyr/lib/libc/minimal/include -isystem /home/robocom/zephyr-sdk-0.12.4/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/include -isystem /home/robocom/zephyr-sdk-0.12.4/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/10.2.0/include-fixed -Og -imacros /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -fdiagnostics-color=always -mcmodel=medany -mabi=lp64 -march=rv64imac -imacros /home/robocom/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/robocom/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/robocom/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c ../src/main.c
In file included from ../src/main.c:14:
/home/robocom/zephyrproject/zephyr/include/fs/littlefs.h:14:10: fatal error: lfs.h: No such file or directory
   14 | #include <lfs.h>
      |          ^~~~~~~
compilation terminated.
[25/121] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/robocom/zephyrproject/zephyr/samples/subsys/fs/littlefs/build

**Logs and console output**

Laczen, 2021-05-14:
Hi @bob2oneil , for littlefs to work a flash device driver is needed. Such a flash driver is not available for the qemu platform.  It is possible to emulate flash like in qemu_x86, but the simulation is done in ram, so there is no persistent storage.

closed 2021-05-14 by galak

bob2oneil, 2021-05-14:
Many thanks galak for your quick response, this is what I needed to know, that is that it is simply not possible.