zephyrproject-rtos/zephyr #35286: New logging breaks eclipse

xhpohanka, 2021-05-14:
**Describe the bug**
With recent changes in logging API (logging v2) using Eclipse IDE for zephyr projects is almost unpossible.
It could be an issue in eclipse parser but AFAIK it is not much maintained these days. 

**To Reproduce**
Steps to reproduce the behavior:
1. create an eclipse project `west build -b MY_BOARD -- -G "Eclipse CDT4 - Ninja"`
2. insert multiple LOG_XXX statements
3. see how eclipse is slow and lagging on most of editting

**Additional context**
After last cleanup for zephyr 2.6 eclipse states that it needs 674 steps to expand `LOG_ERR` macro and results with around 600 lines below.
With first commit where I noticed this behaviour (f1bb20f6b43b8b241e45f3f132f0e7bbfc65401b) there was even around 1900 expand steps and 2000 lines. It is regardless of which logging version is selected through Kconfig.
Originaly it was to about 70 lines.

```c
do { \
	if (!(0 &&					    \
	((1U <= 0U) \
	||								    \
	((0 == 0) &&		    \
	(1U <= __log_level) &&					    \
	(1U <= 0U)				    \
	)								    \
	))) { \
		break; \
	} \
	if (0) { \
		do { \
	z_log_minimal_printk("%c: " "test" "\n", \
			     z_log_minimal_level_to_char(1U)); \
} while (0); \
		break; \
	} \
	\
	_Bool is_user_context = k_is_user_context(); \
	uint32_t filters = 0 ? \
						(__log_current_dynamic_data)->filters : 0;\
	if (0 && !is_user_context && \
	    1U > ((*(&filters) >> (3U * (0))) & ((1UL << (3U)) - 1U))) { \
		break; \
	} \
	if (0) { \
		int _mode; \
		void *_src = 0 ? \
			(void *)__log_current_dynamic_data : (void *)__log_current_const_data; \
		do { \
	; \
	if (({\
	_Pragma("GCC diagnostic push") \
	_Pragma("GCC diagnostic ignored \"-Wpointer-arith\"") \
	int _rv = 0; \
	_Pragma("GCC diagnostic pop")\
	_rv; \
})) { \
		;\
		z_log_msg2_runtime_create(0U, (void *)_src, \
					  1U, (uint8_t *)((void *)0), 0,\
					  "test");\
		_mode = Z_LOG_MSG2_MODE_RUNTIME; \
	} else if (0 && 1 && ((0) == 0)) {\
		;\
		do { \
	int _plen; \
	do { \
	_Pragma("GCC diagnostic push") \
	_Pragma("GCC diagnostic ignored \"-Wpointer-arith\"") \
	 \
	_Static_assert(!0 || \
		     (0 && \
[... 523 lines cut ...]
		0 : 0; \
	struct log_msg_ids src_level = { \
		.level = 1U, \
		.domain_id = 0U, \
		.source_id = src_id \
	}; \
	do { \
	if (is_user_context) { \
		log_from_user(src_level, "test"); \
	} else if (0) { \
		log_string_sync(src_level, "test"); \
	} else { \
		log_0("test", src_level); \
	} \
} while (0); \
} while (0);\
	} \
	if (0) { \
		/* Arguments checker present but never evaluated.*/ \
		/* Placed here to ensure that __VA_ARGS__ are*/ \
		/* evaluated once when log is enabled.*/ \
		z_log_printf_arg_checker("test"); \
	} \
} while (0)
```

xhpohanka, 2021-05-14:
as a quick workaround allowing me to work I have inserted this hack at the bottom of `logging/log.h`

```c
#if defined(__CDT_PARSER__)
#undef LOG_ERR
#undef LOG_WRN
#undef LOG_INF
#undef LOG_DBG

#undef LOG_HEXDUMP_ERR
#undef LOG_HEXDUMP_WRN
#undef LOG_HEXDUMP_INF
#undef LOG_HEXDUMP_DBG

#define LOG_ERR(...) (void) 0
#define LOG_WRN(...) (void) 0
#define LOG_DBG(...) (void) 0
#define LOG_INF(...) (void) 0

#define LOG_HEXDUMP_ERR(...) (void) 0
#define LOG_HEXDUMP_WRN(...) (void) 0
#define LOG_HEXDUMP_DBG(...) (void) 0
#define LOG_HEXDUMP_INF(...) (void) 0
#endif
```

galak, 2021-05-18:
@xhpohanka can you send a PR with this proposed fix.

xhpohanka, 2021-05-19:
I can, but it's more a workaround than a fix.

@nordic-krch can you please confirm, that the example of expanded code in eclipse is somewhat expected?

nordic-krch, 2021-05-19:
@xhpohanka it is expected that logging macros expands in multiple steps, there are macros applied for each argument in the macro message so if this is a problem for eclipse parser then the best would be to suppress that with what you proposed with additional comment explaining why it is being done.

galak, 2021-06-01:
@xhpohanka ping on sending a PR.

PR #35851 (2021-06-02): Eclipse CDT parser workaround for new logging
Recent rework in logging api brings complicated macros which can
overhelm Eclipse CDT parser.
This workaround disables expansion of LOG_xxx macros in Eclipse.

Fixes #35286.
closed 2021-06-02 by galak