zephyrproject-rtos/zephyr #35293: Sporadic boot failure

lyakh, 2021-05-14:
**Describe the bug**
Sporadically the firmware fails to boot with "error: firmware boot failure"

**To Reproduce**
The bug occurs very rarely, to attempt reproduction the SOF driver just has to be loaded, after which kernel messages have to be checked

**Expected behavior**
A successful start of firmware booting looks approximately like this:
```
[   26.932551] [32] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware download successful, booting...
[   26.943387] [1465] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx: 0x70000000: FW_READY
[   26.943406] [1465] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc: DSP is ready 0x70000000 offset 0x81000
[   26.943519] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware info: version 1:8:0-fd0db
[   26.943528] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware: ABI 3:18:1 Kernel ABI 3:18:0
[   26.943829] [1465] sof-audio-pci-intel-cnl 0000:00:1f.3:  mailbox upstream 0x81000 - size 0x1000
[   26.943844] [1465] sof-audio-pci-intel-cnl 0000:00:1f.3:  mailbox downstream 0xa0000 - size 0x2000
[   26.943855] [1465] sof-audio-pci-intel-cnl 0000:00:1f.3:  stream region 0xc1000 - size 0x1000
[   26.943867] [1465] sof-audio-pci-intel-cnl 0000:00:1f.3:  debug region 0xc0000 - size 0x800
[   26.943890] [1465] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc rx done: 0x70000000: FW_READY
[   26.943948] [32] sof-audio-pci-intel-cnl 0000:00:1f.3: firmware boot complete
```

**Impact**
When the firmware fails to load, in general a system reboot is required to recover

**Logs and console output**
A failure case looks like this:
```
[  129.320935] [95] sof-audio-pci-intel-cnl 0000:00:1f.3: Firmware download successful, booting...
[  131.376178] sof-audio-pci-intel-cnl 0000:00:1f.3: error: firmware boot failure
[  131.376417] sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[  131.376442] sof-audio-pci-intel-cnl 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[  131.376486] sof-audio-pci-intel-cnl 0000:00:1f.3: error: extended rom status:  0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0
[  131.376504] sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[  131.376516] sof-audio-pci-intel-cnl 0000:00:1f.3: error: failed to boot DSP firmware -5
```

**Environment (please complete the following information):**
The environment approximately matches Zephyr commit 4208bf1927 and SOF commit 22731744c but newer snapshots are expected to have the same issue. Tested on CNL / cAVS 1.8

andyross, 2021-05-18:
I missed this when submitted, sorry.  Anything in the trace buffer when this happens?  Looks like the "fw entered" status got hit, which means it definitely entered our code.  Does this change with/without SMP=y?

lyakh, 2021-06-18:
haven't been observed in a while, let's close it for now

closed 2021-06-18 by lyakh