zephyrproject-rtos/zephyr #35297: STM32 SPI - wrong behavior after PR 34731

Delsian, 2021-05-14:
McuBoot fails to read JEDEC ID over SPI:
```
[00:00:00.166,000] <err> spi_flash_at45: Wrong JEDEC ID: 00 1F 26, expected: 1F 26 00
```

This bug occurs after applying PR https://github.com/zephyrproject-rtos/zephyr/pull/34731
Commit hash 0129884062055156bc79fd8faafaf13e601438b4
Removing this change solves the problem.

McuBoot settings:
```
...
CONFIG_FLASH=y
CONFIG_SPI=y
CONFIG_SPI_FLASH_AT45=y
...
```

SPI settings:
```
&spi1 {
	status = "okay";
	pinctrl-0 = <&spi1_nss_pa4 &spi1_sck_pa5
		     &spi1_miso_pa6 &spi1_mosi_pa7>;
};
```

Dataflash settings:
```
&spi1 {
	cs-gpios = <&gpioa 4 GPIO_ACTIVE_LOW>;

	dataflash0: at45db321e@0 {
		compatible = "atmel,at45";
		reset-gpios = <&gpioc 4 GPIO_ACTIVE_LOW>;
		wp-gpios = <&gpioa 3 GPIO_ACTIVE_LOW>;
		reg = <0>;
		spi-max-frequency = <1000000>;
		label = "DATAFLASH_0";
		jedec-id = [1f 26 00];
		size = <16777216>;
		sector-size = <65536>;
		block-size = <4096>;
		page-size = <512>;
		use-udpd;
		enter-dpd-delay = <1000>;
		exit-dpd-delay = <180000>;
	};
};
```

@galak @FRASTM @affrinpinhero-2356 FYI.
I did no more checks, just adding a patch locally to remove this PR from my project. Looks like https://github.com/zephyrproject-rtos/zephyr/pull/34731 breaks SPI RX functionality. In my project some errors periodically occurs without this patch:
```
[00:00:03.939,000] <err> spi_ll_stm32: spi_stm32_get_err: err=64
[00:00:03.943,000] <err> pl360_hw: SPI error -5

[00:00:04.735,000] <err> spi_ll_stm32: spi_stm32_get_err: err=64
[00:00:04.740,000] <err> pl360_hw: SPI error -5

[00:00:04.835,000] <err> spi_ll_stm32: spi_stm32_get_err: err=64
[00:00:04.840,000] <err> pl360_hw: SPI error -5

[00:00:05.141,000] <err> spi_ll_stm32: spi_stm32_get_err: err=64
[00:00:05.142,000] <err> pl360_hw: SPI error -5

[00:00:05.938,000] <err> spi_ll_stm32: spi_stm32_get_err: err=64
[00:00:05.942,000] <err> pl360_hw: SPI error -5

[00:00:07.129,000] <err> spi_ll_stm32: spi_stm32_get_err: err=64
[00:00:07.133,000] <err> pl360_hw: SPI error -5
```

avisconti, 2021-05-19:
I also have an issue with  commit 012988406205.
In my case I'm using Nucleo F401RE or F411RE with x_nucleo_iks02a1 or x_nucleo_iks01a3 shield. I did a modification to the samples to test both sensorhub and standard mode with mult-instance. The instance on DIL24 is attached thru SPI, and with this commit STM32 is not able to retrieve even sensor chip id. 

I saved my branch on my github: iis2iclx-fix-triggers-multi-instance-test

avisconti, 2021-05-19:
I just saw that same issue is happening on sensortile.box board using latest master. Again, if I remove 0129884062055156bc79fd8faafaf13e601438b4 the SPI starts working again (by the way I have another issue on that board to STTS751 device, but it is on I2C, so totally different story)

```
0:00:00.003,000] <inf> HTS221: Cannot enable trigger without drdy-gpios
[00:00:00.003,000] <dbg> IIS3DHHC.iis3dhhc_spi_init: SPI GPIO CS configured on GPIOE:10
[00:00:00.004,000] <dbg> LIS2DW12.lis2dw12_spi_init: SPI GPIO CS configured on GPIOE:11
[00:00:00.004,000] <dbg> LIS2DW12.lis2dw12_set_power_mode: Apply default Power Mode
[00:00:00.005,000] <dbg> LIS2MDL.lis2mdl_spi_init: SPI GPIO CS configured on GPIOA:15
[00:00:00.007,000] <inf> LSM6DSO: Initialize device LSM6DSO
[00:00:00.007,000] <inf> LSM6DSO: chip id 0x6c
[00:00:00.509,000] <dbg> STTS751.stts751_init_chip: Failed reading chip id
[00:00:00.509,000] <dbg> STTS751.stts751_init: Failed to initialize chip
[00:00:00.565,000] <inf> usb_cdc_acm: Device suspended
[00:00:00.983,000] <inf> usb_cdc_acm: Device configured
SensorTile.box test!!
Could not get STTS751 device
```

You can find the sensortile.box sample here:
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/boards/sensortile_box

board:
https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/arm/sensortile_box

StefJar, 2021-05-21:
I am having problems with the stm32 spi as well. After migrating my fw from an older zephyr version to the latest main branch the SPI does not seem to work.
I checked the CS etc to make sure that everything is fine and setup.
It does not really matter if I am using the DMA or not. If no DMA is used I am getting `spi_ll_stm32: spi_stm32_get_err: err=64`.

My setup is:
two SPI slaves on my SPI3. They are setup with different speeds, `SPI_MODE_CPOL` and `SPI_MODE_CPHA` parameters. So the SPI driver needs to switch almost every time its config. 

Happy to test quickly any changes on that issue.

@galak anyone working on that?

erwango, 2021-05-21:
> @galak anyone working on that?

Yes, we are.
(Hint: You can check the assignee field on top of each issue to see who's working on what)

avisconti, 2021-05-21:
> I am having problems with the stm32 spi as well. After migrating my fw from an older zephyr version to the latest main branch the SPI does not seem to work.
> I checked the CS etc to make sure that [...]

Can you to check again after a git revert 01298840 ?
Just to be sure that we are talking about same issue...

StefJar, 2021-05-21:
> > I am having problems with the stm32 spi as well. After migrating my fw from an older zephyr version to the latest main branch the SPI does not seem to work.
> > I checked the CS etc to make sure that everything is fine and setup.
> > It does not really matter if I am using the DMA or not. If no DMA is used I am getting `spi_ll_stm32: spi_stm32_get_err: err=64`.
> > My setup is:
> > two SPI slaves on my SPI3. They are setup with different speeds, `SPI_MODE_CPOL` and `SPI_MODE_CPHA` parameters. So the SPI driver needs to switch almost every time its config.
> > Happy to test quickly any changes on that issue.
> > @galak anyone working on that?
> 
> Can you try to check it again after a git revert [0129884](https://github.com/zephyrproject-rtos/zephyr/commit/0129884062055156bc79fd8faafaf13e601438b4) ?
> Just to be sure that we are talking about same issue...

Done the changes, but it seems to me that they don't have a real effect. Guess my issue is maybe different.

erwango, 2021-05-21:
@StefJar can you raise a specific issue with all possible details ?

StefJar, 2021-05-21:
> @StefJar can you raise a specific issue with all possible details ?

Hard for me to nail to down. My setup is
SPI1 -> ext nor flash
SPI3 -> movement sensor(node 0) & eeg sensor(node 1)

When starting with a hard reset I am seeing several crashes before the system becomes stable.

the flash related exception is:
```
00:00:00.495,312] <err> os: ***** Reserved Exception ( -16) *****

[00:00:00.505,782] <err> os: r0/a1:  0x2000d580  r1/a2:  0x0801bc31  r2/a3:  0x08029929

[00:00:00.519,181] <err> os: r3/a4:  0x0801bc5d r12/ip:  0x0801bc5d r14/lr:  0x0801bc5d

[00:00:00.532,570] <err> os:  xpsr:  0x00000000

[00:00:00.540,554] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000

[00:00:00.556,502] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000

[00:00:00.572,449] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000

[00:00:00.588,412] <err> os: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000

[00:00:00.604,390] <err> os: fpscr:  0x0801bbf1

[00:00:00.612,433] <err> os: Faulting instruction address (r15/pc): 0x0801bc5d

[00:00:00.624,542] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0

[00:00:00.636,474] <err> os: Current thread: 0x200034f8 (unknown)�[00:00:00.005,212] <err> spi_nor: Device id 00 00 00 does not match config bf 26 41

```
So after this kind of exception the system comes up with having problems finding the flash (on spi1)

On spi3 the situation is different. Sometime the SPI transfers to both peripherals is failing. Sometimes only one.
Sorry that is very unspecific. It feels for me that somehow a reset is might putting the SPIx into some weird state, from which they don't really recover.

I have working firmware without Zephyr based on the STM32cubeMX API and a working firmware on a somehow modded Zephyr from 2.5 years ago.

erwango, 2021-05-21:
@StefJar ok, thanks. But if not directly linked with current issue, let's discuss this elsewhere in order not to confuse things.

Delsian, 2021-05-21:
> Can you try to check it again after a git revert [0129884](https://github.com/zephyrproject-rtos/zephyr/commit/0129884062055156bc79fd8faafaf13e601438b4) ?
> Just to be sure that we are talking about same issue...

@avisconti  Just checked again. On branch 6ce0f2ee6606915c75e17753e34db71cb053c119 bug still present,  reverting 0129884062055156bc79fd8faafaf13e601438b4 resolves my problem.

StefJar, 2021-05-24:
@erwango I moved my possible DMA/INTR related issues to an extra ticket(https://github.com/zephyrproject-rtos/zephyr/issues)/35581

galak, 2021-05-26:
Fixed by #35661

closed 2021-05-26 by galak