zephyrproject-rtos/zephyr #35299: PM suspend IPC message not delivered sporadically

lyakh, 2021-05-14:
**Describe the bug**
Sometimes the PM suspend `SOF_IPC_PM_CTX_SAVE` IPC messages fails to be delivered to the IPC handler, which results in an IPC timeout error message in the kernel SOF driver.

**To Reproduce**
Wait for 3 seconds after loading the SOF driver or streaming audio to let the driver enter automatic runtime suspend. Most often the problem occurs after playing audio on an HDMI interface. Also building with `CONFIG_SMP=n` makes the problem appear much more often.

**Expected behavior**
Usually `SOF_IPC_PM_CTX_SAVE` is received by the firmware and replied to accordingly, producing kernel output similar to:
```
[ 1101.433556] [1523] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x40010000
[ 1101.434069] [1523] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx succeeded: 0x40010000
[ 1101.434109] [1523] sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0x1010f0f successful
[ 1101.434683] [1523] sof-audio-pci-intel-cnl 0000:00:1f.3: FW Poll Status: reg[0x4]=0xf0f successful
[ 1101.434697] [1523] sof-audio-pci-intel-cnl 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 1101.434780] [1523] sof-audio-pci-intel-cnl 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[ 1101.434788] [1523] sof-audio-pci-intel-cnl 0000:00:1f.3: Debug PCIW: 00000010 at  00000044
[ 1101.434799] [1523] sof-audio-pci-intel-cnl 0000:00:1f.3: Current DSP power state: D3
```

**Impact**
When this failure occurs, the kernel driver still goes on to power the DSP down, after which it can be powered up again and used as usual.

**Logs and console output**
In an error case the kernel log looks like:
```
[  296.073271] [22] sof-audio-pci-intel-cnl 0000:00:1f.3: ipc tx: 0x40010000
[  296.579303] sof-audio-pci-intel-cnl 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[  296.579324] sof-audio-pci-intel-cnl 0000:00:1f.3: status: fw entered - code 00000005
[  296.579583] sof-audio-pci-intel-cnl 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00000000
[  296.579598] sof-audio-pci-intel-cnl 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  296.579605] sof-audio-pci-intel-cnl 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  296.579618] sof-audio-pci-intel-cnl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  296.579625] sof-audio-pci-intel-cnl 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
```

**Environment (please complete the following information):**
The environment approximately matches Zephyr commit 4208bf1 and SOF commit 22731744c but newer snapshots are expected to have the same issue. Tested on CNL / cAVS 1.8

kv2019i, 2021-05-21:
@lyakh This is fixed right? At least based on my testing on TGL, SOF patch https://github.com/thesofproject/sof/pull/4208 fixes this.

lyakh, 2021-05-25:
@kv2019i thanks for the confirmation, let's close it then.

closed 2021-05-25 by lyakh