zephyrproject-rtos/zephyr #35305: Linking order when using both TF-M and Mbed TLS

urutva, 2021-05-14:
**Describe the bug**
When both TF-M and Mbed TLS (on NS side) modules are enabled, all the PSA crypto API calls are linked against Mbed TLS library instead of TF-M. This is caused by incorrect linking order of modules where Mbed TLS comes before TF-M.

Please note that this is not a linking failure but incorrect linking of APIs.

**To Reproduce**
Steps to reproduce the behavior:
1. Enabled TF-M and Mbed TLS
`samples/tfm_integration/psa_level_1/prj.conf`
```
CONFIG_BUILD_WITH_TFM=y

CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=32768
CONFIG_MBEDTLS_USER_CONFIG_ENABLE=y
CONFIG_MBEDTLS_USER_CONFIG_FILE="user-tls.conf"
CONFIG_MBEDTLS_ENTROPY_ENABLED=y
CONFIG_MBEDTLS_PK_WRITE_C=y
CONFIG_MBEDTLS_MD=y
CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_ENABLED=n
CONFIG_MBEDTLS_TLS_VERSION_1_2=n
```
`zephyr/samples/tfm_integration/psa_level_1/CMakeLists.txt`
```
zephyr_include_directories(${APPLICATION_SOURCE_DIR}/src/tls_config)
```
`zephyr/samples/tfm_integration/psa_level_1/src/tls_config/user-tls.conf`
```
#define MBEDTLS_USE_PSA_CRYPTO
#define MBEDTLS_PSA_CRYPTO_C

#define MBEDTLS_ECP_C
#define MBEDTLS_ECP_DP_SECP256R1_ENABLED

#define MBEDTLS_TEST_NULL_ENTROPY

#define MBEDTLS_X509_CSR_WRITE_C
#define MBEDTLS_PEM_WRITE_C
#define MBEDTLS_X509_CREATE_C
#define MBEDTLS_BIGNUM_C
#define MBEDTLS_OID_C
#define MBEDTLS_ASN1_PARSE_C
#define MBEDTLS_ASN1_WRITE_C

#define MBEDTLS_BASE64_C
```
2. Call any PSA crypto API (`psa_generate_random`) in samples/tfm_integration/psa_level_1/src/main.c.
3. west build -p -b mps2_an521_nonsecure samples/tfm_integration/psa_level_1 -t run

**Expected behavior**
PSA APIs are linked against TF-M library.

**Impact**
showstopper

**Logs and console output**
Linking order:
`build/build.ninja`
```
#############################################
# Link the executable zephyr/zephyr_prebuilt.elf

build zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map: C_EXECUTABLE_LINKER__zephyr_prebuilt_ zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj | zephyr/linker_zephyr_prebuilt.cmd app/libapp.a zephyr/libzephyr.a zephyr/arch/common/libarch__common.a zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a zephyr/arch/arch/arm/core/aarch32/cortex_m/cmse/libarch__arm__core__aarch32__cortex_m__cmse.a zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/lib/posix/liblib__posix.a zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a zephyr/boards/arm/mps2_an521/libboards__arm__mps2_an521.a zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/i2c/libdrivers__i2c.a zephyr/drivers/serial/libdrivers__serial.a modules/mbedtls/libmodules__mbedtls.a modules/trusted-firmware-m/libtfm_api.a zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj zephyr/arch/common/libisr_tables.a tfm/platform/libplatform_ns.a tfm/app/libtfm_api_ns.a tfm/secure_fw/s_veneers.o zephyr/linker_zephyr_prebuilt.cmd || app/libapp.a modules/mbedtls/libmodules__mbedtls.a modules/trusted-firmware-m/libtfm_api.a modules/trusted-firmware-m/tfm zephyr/arch/arch/arm/core/aarch32/cortex_m/cmse/libarch__arm__core__aarch32__cortex_m__cmse.a zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a zephyr/arch/common/libarch__common.a zephyr/arch/common/libisr_tables.a zephyr/boards/arm/mps2_an521/libboards__arm__mps2_an521.a zephyr/driver_validation_h_target zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/i2c/libdrivers__i2c.a zephyr/drivers/serial/libdrivers__serial.a zephyr/kernel/libkernel.a zephyr/kobj_types_h_target zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/lib/posix/liblib__posix.a zephyr/libzephyr.a zephyr/linker_zephyr_prebuilt_script_target zephyr/offsets zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a zephyr/syscall_list_h_target
  LINK_LIBRARIES = -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/Users/devran01/Documents/devaraj/linaro/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/cmse/libarch__arm__core__aarch32__cortex_m__cmse.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/boards/arm/mps2_an521/libboards__arm__mps2_an521.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/serial/libdrivers__serial.a  modules/mbedtls/libmodules__mbedtls.a  modules/trusted-firmware-m/libtfm_api.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"/Users/devran01/Documents/devaraj/linaro/compilers/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v8-m.main/nofp"  -L/Users/devran01/Documents/devaraj/linaro/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  tfm/platform/libplatform_ns.a  tfm/app/libtfm_api_ns.a  tfm/secure_fw/s_veneers.o  -mcpu=cortex-m33+nodsp  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn
```

The library `modules/mbedtls/libmodules__mbedtls.a` comes before`tfm/app/libtfm_api_ns.a` which is causing all the PSA crypto APIs to be linked against Mbed TLS

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: CrossTool-NG
 - 5f5fb7d7925be3b40b4a963b11fb8b4cfedf03a1

**Additional context**
Add any other context about the problem here.

urutva, 2021-05-14:
@microbuilder

microbuilder, 2021-05-18:
@tejlmand Any suggestions on forcing the right order for linking here?

tejlmand, 2021-05-19:
> @tejlmand Any suggestions on forcing the right order for linking here?

unfortunately not.
The mbed TLS library is linked inside the whole-archive, which means we cannot control the order in the usual CMake way.
So without #8451 this is difficult to manage.

That said, having the same APIs provided by two libraries and relying on correct linking order is in general a bad habit, and in high risk of suddenly fail if other changes / dependencies causes the link order to change.

In this case, the best thing would be to ensure that the mbed TLS does not provide the PSA APIs, which could be done if the granularity of mbedTLS Kconfig is fine enough so we could specify something like:
```
config MBEDTLS_PSA_SOME_SETTING
        depends on !BUILD_WITH_TFM
```

microbuilder, 2021-05-19:
Given that the PSA API support in mbedtls 2.26.0 is still incomplete, and a (likely) LTS release of mbedtls will happen during the Zephyr 2.7.0 dev cycle, my vote is to disable use of the PSA backend for now in mbedtls, and reference mbedtls as a purely stand-alone module with no dependencies on TF-M.

We can revisit mbedtls + PSA integration in the 2.7.0 dev cycle, with an improved release of mbedtls for PSA support, but we still have access to features in mbedtls today such as the X.509 functions, etc., that aren't exposed by TF-M.

microbuilder, 2021-05-19:
@urutva Does disabling these two defines result in the PSA APIs not being included in the mbedtls build:

```
#define MBEDTLS_USE_PSA_CRYPTO
#define MBEDTLS_PSA_CRYPTO_C
```

@galak No samples in Zephyr 2.6.0 currently use this combination, and it will only be a requirement in 2.7.0, so we can document this as a known issue ("mbedtls 2.26.0 PSA Crypto APIs can not be used when CONFIG_BUILD_WITH_TFM is enabled.") and punt to 2.7.0 where we'll have more time to improve the integration of mbedtls and TF-M? I think this combination needs more testing.

microbuilder, 2021-05-19:
## Test Results

### With `MBEDTLS_USA_PSA_CRYPTO` (Fails)

```
#define MBEDTLS_USE_PSA_CRYPTO
#define MBEDTLS_PSA_CRYPTO_C

#define MBEDTLS_ECP_C
#define MBEDTLS_ECP_DP_SECP256R1_ENABLED

#define MBEDTLS_TEST_NULL_ENTROPY

#define MBEDTLS_X509_CSR_WRITE_C
#define MBEDTLS_PEM_WRITE_C
#define MBEDTLS_X509_CREATE_C
#define MBEDTLS_BIGNUM_C
#define MBEDTLS_OID_C
#define MBEDTLS_ASN1_PARSE_C
#define MBEDTLS_ASN1_WRITE_C

#define MBEDTLS_BASE64_C
```

```bash
$ west build -p -b mps2_an521_nonsecure samples/tfm_integration/psa_level_1 -t run
*** Booting Zephyr OS build zephyr-v2.5.0-3821-g41afd838aabd  ***
[00:00:00.023,000] <inf> app: app_cfg: Creating new config file with UID 0x1055CFDA7A
[00:00:03.948,000] <inf> app: att: System IAT size is: 545 bytes.
[00:00:03.948,000] <inf> app: att: Requesting IAT with 64 byte challenge.
[00:00:05.942,000] <inf> app: att: IAT data received: 545 bytes.
          0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
00000000 D2 84 43 A1 01 26 A0 59 01 D5 AA 3A 00 01 24 FF ..C..&.Y...:..$.
00000010 58 40 00 11 22 33 44 55 66 77 88 99 AA BB CC DD X@.."3DUfw......
00000020 EE FF 00 11 22 33 44 55 66 77 88 99 AA BB CC DD ...."3DUfw......
00000030 EE FF 00 11 22 33 44 55 66 77 88 99 AA BB CC DD ...."3DUfw......
00000040 EE FF 00 11 22 33 44 55 66 77 88 99 AA BB CC DD ...."3DUfw......
00000050 EE FF 3A 00 01 24 FB 58 20 A0 A1 A2 A3 A4 A5 A6 ..:..$.X .......
00000060 A7 A8 A9 AA AB AC AD AE AF B0 B1 B2 B3 B4 B5 B6 ................
00000070 B7 B8 B9 BA BB BC BD BE BF 3A 00 01 25 00 58 21 .........:..%.X!
00000080 01 FA 58 75 5F 65 86 27 CE 54 60 F2 9B 75 29 67 ..Xu_e.'.T`..u)g
00000090 13 24 8C AE 7A D9 E2 98 4B 90 28 0E FC BC B5 02 .$..z...K.(.....
000000A0 48 3A 00 01 24 FA 58 20 AA AA AA AA AA AA AA AA H:..$.X ........
000000B0 BB BB BB BB BB BB BB BB CC CC CC CC CC CC CC CC ................
000000C0 DD DD DD DD DD DD DD DD 3A 00 01 24 F8 20 3A 00 ........:..$. :.
000000D0 01 24 F9 19 30 00 3A 00 01 24 FD 82 A5 01 63 53 .$..0.:..$....cS
000000E0 50 45 04 65 30 2E 30 2E 30 05 58 20 BF E6 D8 6F PE.e0.0.0.X ...o
000000F0 88 26 F4 FF 97 FB 96 C4 E6 FB C4 99 3E 46 19 FC .&..........>F..
00000100 56 5D A2 6A DF 34 C3 29 48 9A DC 38 06 66 53 48 V].j.4.)H..8.fSH
00000110 41 32 35 36 02 58 20 D9 8B A3 AB AC 63 D2 8A 2F A256.X .....c../
00000120 24 75 A7 66 0B 9A 08 E9 46 F1 70 3E F1 7F FF B5 $u.f....F.p>....
00000130 01 1D 98 EF F2 5C A5 A5 01 64 4E 53 50 45 04 65 .....\...dNSPE.e
00000140 30 2E 30 2E 30 05 58 20 B3 60 CA F5 C9 8C 6B 94 0.0.0.X .`....k.
00000150 2A 48 82 FA 9D 48 23 EF B1 66 A9 EF 6A 6E 4A A3 *H...H#..f..jnJ.
00000160 7C 19 19 ED 1F CC C0 49 06 66 53 48 41 32 35 36 |......I.fSHA256
00000170 02 58 20 6B DA DB AE 40 EF A6 3F 27 14 E3 1D 68 .X k...@..?'...h
00000180 3C F6 B3 22 31 64 A6 AD A8 03 FE 28 5B 43 42 21 <.."1d.....([CB!
00000190 9F EA 43 3A 00 01 25 01 77 77 77 77 2E 74 72 75 ..C:..%.wwww.tru
000001A0 73 74 65 64 66 69 72 6D 77 61 72 65 2E 6F 72 67 stedfirmware.org
000001B0 3A 00 01 24 F7 71 50 53 41 5F 49 4F 54 5F 50 52 :..$.qPSA_IOT_PR
000001C0 4F 46 49 4C 45 5F 31 3A 00 01 24 FC 72 30 36 30 OFILE_1:..$.r060
000001D0 34 35 36 35 32 37 32 38 32 39 31 30 30 31 30 58 456527282910010X
000001E0 40 53 A1 B7 9B 18 45 D4 15 4D 84 8C A6 D6 0C 10 @S....E..M......
000001F0 A3 88 17 E7 E7 C9 39 72 DC 32 ED A0 DB FB EA 06 ......9r.2......
00000200 19 68 0F 4C D7 66 87 80 D4 B4 40 6B 37 3F E3 C8 .h.L.f....@k7?..
00000210 02 89 AF 5E 24 FD 3A 9A 5A DC F0 B7 E5 D4 8C BC ...^$.:.Z.......
00000220 34                                              4
[00:00:06.004,000] <err> app: Bad state
[00:00:06.004,000] <err> app: Function: 'crp_test_rng'
[00:00:06.004,000] <inf> app: Generating 256 bytes of random data.
[... 82 lines cut ...]
00000210 6D E0 12 3D 05 C9 80 CD 3C 16 C8 B1 17 BE 6B DF m..=....<.....k.
00000220 83                                              .
[00:00:06.007,000] <inf> app: Generating 256 bytes of random data.
          0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
00000000 6F 35 D1 58 2C C0 CD 28 2A 25 C2 AD 71 99 4D D5 o5.X,..(*%..q.M.
00000010 A2 3D 99 8B 54 78 1A 46 C5 FE A6 6C 66 F2 43 26 .=..Tx.F...lf.C&
00000020 68 47 DD A7 9C BA A0 98 B7 89 34 C9 99 36 88 08 hG........4..6..
00000030 3A E3 9F BA 94 39 D5 72 F1 8F 95 3C 53 2C 2A F2 :....9.r...<S,*.
00000040 59 79 24 12 53 D2 A5 C1 4F DD E5 93 11 3D 25 E4 Yy$.S...O....=%.
00000050 03 50 9A 08 0F 7E 9C 4E 44 F1 12 F3 F4 35 03 D8 .P...~.ND....5..
00000060 7A AF CB 06 B7 5F AD 5C 4C 7E EC A5 F1 39 6A CE z...._.\L~...9j.
00000070 6D 16 27 F8 4F A3 58 D3 F3 1B A1 D0 43 42 AF 2D m.'.O.X.....CB.-
00000080 F6 92 B3 94 69 0B 9D AB E0 1A EE 2A 02 7A 0B A0 ....i......*.z..
00000090 AA 18 A7 5A CD 3E 71 B9 E7 57 9B 21 A8 D1 28 B6 ...Z.>q..W.!..(.
000000A0 B7 76 E4 19 62 B3 4A 9D EC 46 C1 48 CA 4E 0B E3 .v..b.J..F.H.N..
000000B0 58 1B 76 27 CF 25 9F 70 6B B2 09 90 B2 A1 AD FA X.v'.%.pk.......
000000C0 61 4A C4 13 59 3F 13 6D A0 DC 4E 50 69 AC 12 B0 aJ..Y?.m..NPi...
000000D0 99 5A AA F9 42 D3 DF A0 0A 48 61 BC A8 E0 D3 4C .Z..B....Ha....L
000000E0 D2 98 AD DD DC 6E 6F 88 60 FD 50 B8 25 B9 1A 82 .....no.`.P.%...
000000F0 44 65 23 20 15 0C B0 62 D1 0A 6C 88 4D 9C 88 3A De# ...b..l.M..:
[00:00:06.035,000] <inf> app: Calculating SHA-256 hash of value.
          0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
00000000 E3 B0 C4 42 98 FC 1C 14 9A FB F4 C8 99 6F B9 24 
00000010 27 AE 41 E4 64 9B 93 4C A4 95 99 1B 78 52 B8 55 
```

microbuilder, 2021-05-20:
I've been trying to find a solution to the PSA config issue to not allow `CONFIG_BUILD_WITH_TFM` and mbedtls PSA APIs at the same time (since it leads to ambiguous linking), with some variation of a new PSA Kconfig menu and a default `config-psa.h` file similar to what we do for TLS:

```bash
menu "PSA configuration"
	depends on MBEDTLS_BUILTIN && MBEDTLS_CFG_FILE = "config-psa.h"

config MBEDTLS_USE_PSA_CRYPTO
	bool "Enable PSA for X.509 and TLS cryptographic operations"
    depends on !BUILD_WITH_TFM
	help
	  The mbedtls library will use Platform Security Architecture (PSA)
	  API for the X.509 and TLS cryptographic operations. This option is
	  not enabled by default, and can not be used when TF-M is enabled.

...

endmenu
```

This will only prevent potential linker issues when using this config file, though, not against users defining their own config file (as happens in this issue) and enabling the PSA APIs plus TF-M  there.

The ideal solution here is to expose all PSA settings via Kconfig to avoid the need for a custom mbedtls config file, but given that no samples currently use this functionality it's a new feature that should be tackled in Zephyr 2.7.0.

IMO, we should note this as a known issue that PSA APIs can not be used together with TF-M in mbedtls 2.26.0 and Zephyr 2.6.0 (since we can't reasonably prevent this at present without sacrificing mbedtls entirely), and integrate PSA config into the next release along with a new sample that makes use of that functionality, with proper testing of the various flags involved. This will also allow for a newer release of mbedtls to be used with better PSA API support.

microbuilder, 2021-05-20:
As per a conversation with @urutva, the problem is as follows:

- `MBEDTLS_USE_PSA_CRYPTO` defines TF-M as a backend for Mbed TLS
- `MBEDTLS_PSA_CRYPTO_C` controls the TF-M crypto implementation in Mbed TLS.

Unfortunately, both configs are tied to each other, so one cannot enable only `MBEDTLS_USE_PSA_CRYPTO` (to use TF-M as a backend) ... we also have to enable `MBEDTLS_PSA_CRYPTO_C`, which pulls in it's own TF-M Crypto implementation.  In that case we have two PSA crypto implementations resulting in arbitrary linker issues.

ioannisg, 2021-05-20:
> Unfortunately, both configs are tied to each other, so one cannot enable only MBEDTLS_USE_PSA_CRYPTO (to use TF-M as a backend) ... we also have to enable MBEDTLS_PSA_CRYPTO_C, which pulls in it's [...]

@microbuilder this is going to be addressed in the mbedtls 3.0, is that correct? If so, I think we will have time until the Zephyr LTS 2 release to fix this problem.

microbuilder, 2021-05-20:
@ioannisg I think this is something to resolve in the 2.7 cycle with an update to mbedtls, yes, and we can bring the conflict up with the mbedtls community. I don't think this should be considered a bug to solve for 2.6.0. See #35513 for a release note describing this limitation.

microbuilder, 2021-05-21:
Closing this as unable to fix with the current limitations of the build system and mbedtls 2.26.0, but will be addressed in the 2.7.0 dev cycle, coordinating with the upstream mbedtls project for any required changes there.

Enhancement can be tracked via https://github.com/zephyrproject-rtos/zephyr/issues/35552

closed 2021-05-21 by microbuilder