zephyrproject-rtos/zephyr #35316: log_panic() hangs kernel

sslupsky, 2021-05-15:
**Describe the bug**
I've been experiencing this problem for several years now using Zephyr with sam0.  Whenever I encounter a panic situation, the kernel simply hangs.  On very rare occasions, I have seen an actual panic message and a dump.  But most often, the mcu simply hangs waiting for what appears to be a semaphore.

**To Reproduce**
In my situation, cause a panic condition.

**Expected behavior**
Kernel should handle the panic and output panic message and halt.

**Impact**
Kernel panics lead to hang.

**Logs and console output**
Here is a stack trace showing the initial fault and the sequence to the hang.
```
z_add_timeout@0x0001b5ae (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/timeout.c:119)
z_add_thread_timeout@0x0001a90c (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/include/timeout_q.h:49)
add_thread_timeout@0x0001a90c (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/sched.c:594)
pend@0x00023d12 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/sched.c:605)
z_pend_curr@0x0001adfa (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/sched.c:679)
z_impl_k_sem_take@0x0001b0c6 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/sem.c:158)
k_sem_take@0x0000bc20 (/Users/stevenslupsky/Documents/source/zephyrproject/dock1310/build/zephyr/include/generated/syscalls/kernel.h:756)
cdc_acm_poll_out@0x0000bc20 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/usb/class/cdc_acm.c:950)
z_impl_uart_poll_out@0x0001eee0 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/include/drivers/uart.h:645)
uart_poll_out@0x0001eee0 (/Users/stevenslupsky/Documents/source/zephyrproject/dock1310/build/zephyr/include/generated/syscalls/uart.h:112)
write@0x0001eee0 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_uart.c:225)
shell_write@0x0001ec7a (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_ops.c:380)
shell_print_stream@0x0001ec9c (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_ops.c:399)
shell_fprintf_buffer_flush@0x0001e886 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_fprintf.c:56)
shell_fprintf_fmt@0x00009530 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_fprintf.c:49)
shell_raw_fprintf@0x0001eafc (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_ops.h:24)
shell_op_cursor_vert_move@0x0000994a (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_ops.c:13)
panic@0x0001f410 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_log_backend.c:281)
log_backend_panic@0x000072f0 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/include/logging/log_backend.h:178)
z_impl_log_panic@0x000072f0 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/logging/log_core.c:597)
log_panic@0x0001a0f8 (/Users/stevenslupsky/Documents/source/zephyrproject/dock1310/build/zephyr/include/generated/syscalls/log_ctrl.h:34)
k_sys_fatal_error_handler@0x0001a0f8 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/fatal.c:41)
z_fatal_error@0x0001a172 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/fatal.c:123)
z_arm_fatal_error@0x00020abe (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/arch/arm/core/aarch32/fatal.c:47)
z_arm_fault@0x00010ae4 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:977)
z_arm_hard_fault@0x00010964 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:85)
<signal handler called>@0xfffffff1 (Unknown Source:0)
z_check_stack_sentinel@0x0001b262 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/thread.c:369)
z_swap_irqlock@0x0001adfe (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/include/kswap.h:175)
z_swap@0x0001adfe (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/include/kswap.h:175)
z_pend_curr@0x0001adfe (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/sched.c:680)
z_impl_k_sem_take@0x0001b0c6 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/sem.c:158)
k_sem_take@0x0000bc20 (/Users/stevenslupsky/Documents/source/zephyrproject/dock1310/build/zephyr/include/generated/syscalls/kernel.h:756)
cdc_acm_poll_out@0x0000bc20 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/usb/class/cdc_acm.c:950)
z_impl_uart_poll_out@0x0001eee0 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/include/drivers/uart.h:645)
uart_poll_out@0x0001eee0 (/Users/stevenslupsky/Documents/source/zephyrproject/dock1310/build/zephyr/include/generated/syscalls/uart.h:112)
write@0x0001eee0 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_uart.c:225)
shell_write@0x0001ec7a (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_ops.c:380)
shell_print_stream@0x0001ec9c (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_ops.c:399)
shell_fprintf_buffer_flush@0x0001e886 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_fprintf.c:56)
shell_fprintf_fmt@0x00009530 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_fprintf.c:49)
shell_raw_fprintf@0x0001eafc (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_ops.h:24)
shell_op_cursor_vert_move@0x0000994a (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_ops.c:13)
panic@0x0001f410 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/shell/shell_log_backend.c:281)
log_backend_panic@0x000072f0 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/include/logging/log_backend.h:178)
z_impl_log_panic@0x000072f0 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/subsys/logging/log_core.c:597)
log_panic@0x0001a0f8 (/Users/stevenslupsky/Documents/source/zephyrproject/dock1310/build/zephyr/include/generated/syscalls/log_ctrl.h:34)
k_sys_fatal_error_handler@0x0001a0f8 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/fatal.c:41)
z_fatal_error@0x0001a172 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/kernel/fatal.c:123)
z_arm_fatal_error@0x00020abe (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/arch/arm/core/aarch32/fatal.c:47)
z_do_kernel_oops@0x00020aca (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/arch/arm/core/aarch32/fatal.c:86)
_oops@0x000106b6 (/Users/stevenslupsky/Documents/source/zephyrproject/zephyr/arch/arm/core/aarch32/swap_helper.S:465)
<signal handler called>@0xfffffffd (Unknown Source:0)
??@0xaaaaaaaa (Unknown Source:0)
```

**Environment (please complete the following information):**
v2.3.99

The "reason" in the k_sys_fatal_error_handler() after the oops is "2".  I cannot seem to find anywhere where the "reasons" are documented as to what they mean.

This may be related to issues #34372 and #31761.  Both of those issues appear to have been closed without actually resolving the underlying issue(s).

@carlescufi I think the GitHub "auto close" needs to be turned off on this repo.  There are far too many issues being closed that should not be.

sslupsky, 2021-05-15:
Part way through the stack trace there appears to be an exception generated by the stack sentinel.  This appears to imply that the stack of the outgoing thread experienced an overrun. I believe the current thread is the kernel thread since an exception (oops) is being processed ... is that correct?

Further on, the shell appears to be emitting some output through the usb_cdc.  Interestingly, in this particular instance, the usb cdc initialization is not complete.  In fact, usb has not even been enabled yet.  Here is main where the first exception occurs:
```
	/* start logging to files */
	log.begin(&soc_storage_mnt, &spi_storage_mnt, &dock1310.device.boot, K_SECONDS(WITAP_LOG_SYNC_TIMEOUT));

	k_work_init(&usb_fsm_work, usb_fsm_handler);
	k_delayed_work_init(&usb_disconnect_work, usb_disconnect_handler);
	ret = usb_enable(usb_status_cb);
```

log.being() causes the error (some problem with littlefs causes the oops).  The usb interface is enabled after this point.

sslupsky, 2021-05-15:
I suspect because the usb is not enabled that the usb_cdc data structures are not yet properly initialized.  So, the shell panic() it is likely using the usb_cdc when it should not be.

Likely, the shell panic() function should check that the cdc is properly initialized before attempting to use it?

sslupsky, 2021-05-15:
It appears shell panic() is attempting to enable the interface here:

```
	err = shell->iface->api->enable(shell->iface, true);
```

I do not understand how it is possible to reliably enable the interface (usb cdc) here since the underlying interface (usb) has not been enabled yet. I expected that enable() here should have returned an error because usb is not enabled but instead it returns zero and panic() continues which leads to the hang.

sslupsky, 2021-05-15:
The interface being used by the shell is:
```
 0x200009b0 <shell_transport_uart>
```

Which I believe is connected to the usb_cdc in shell_uart.c write() here:
```
uart_poll_out(sh_uart->ctrl_blk->dev, data8[i]);
```

In this case, sh_uart->ctrl_blk->dev is 
```
0x20000f24 <__device_cdc_acm_0>
```

caco3, 2021-05-15:
As mentioned in https://github.com/zephyrproject-rtos/zephyr/issues/34372#issuecomment-825635189, I also see occational Crashes.

Recently I similair to @sslupsky saw it happen around a `log_panic()` call.

I have a battery driven device and when the battery is empty, I simply switch it off. In such case BLE communication might still be in progress while we call following function:
```
void powerOff(void) {
    /* Cut Power off, try infinitely in case it fails */
    log_panic(); // Disable log buffer, all messages get flushed immediately
    LOG_INF("\n\n*** POWER OFF ***");
    while (1) {
       // <= Here we use I2C to disconnect the power chip, expecting to lose power to the MCU.
       //    How ever sporadically it never reaches this I2C call and instead runs into a Watchdog reset.
        k_msleep(100);
        printk("Still powered, trying again...\n");
    }
}

```

nordic-krch, 2021-05-19:
One of the reasons for that crash is that usb is used as uart driver. Docs (https://docs.zephyrproject.org/latest/reference/logging/index.html#logging-backends) states that

```
Additionally, backend is notfied when logging enter panic mode with log_backend_panic(). On that call backend should switch to synchronous, interrupt-less operation or shut down itself if that is not supported. 
```

cdc_acm based uart probably does not support interrupt-less initialization and should be shut down in that case.

sslupsky, 2021-05-20:
That might work if I can continue to use RTT?

How do I shut down cdc acm?  It should not be enabled since USB is not yet enabled?

sslupsky, 2021-05-21:
@nordic-krch Is there something we can do about this?  It looks like something gets messed up in z_add_timeout()?  When it hangs, the cpu appears to be infinitely looping in here.
```
		for (t = first(); t != NULL; t = next(t)) {
			__ASSERT(t->dticks >= 0, "");

			if (t->dticks > to->dticks) {
				t->dticks -= to->dticks;
				sys_dlist_insert(&t->node, &to->node);
				break;
			}
			to->dticks -= t->dticks;
		}
```

galak, 2021-06-01:
@nordic-krch what's the latest here?

sslupsky, 2021-10-18:
@carlescufi Has this issue been resolved?

carlescufi, 2021-12-20:
> @carlescufi Has this issue been resolved?

No, it has not actually. Let me reopen it.

carlescufi, 2021-12-20:
@jfischer-no could you comment on @nordic-krch's questions?

nordic-krch, 2022-01-31:
@sslupsky  USB CDC ACM poll_out function has been reworked to be callable from interrupt #37503. That should impact it (potentially fix it). Do you still see the issue?

sslupsky, 2022-02-01:
@nordic-krch I haven't looked at this in a while.  I will have a look.

carlescufi, 2022-03-30:
> @nordic-krch I haven't looked at this in a while. I will have a look.

@sslupsky ping

jfischer-no, 2022-05-25:
This should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/37503.

closed 2022-05-25 by jfischer-no

DuMaM, 2022-12-12:
It still happening for me. 
I'm using nrf52840 and nrf sdk 2.1.2 with zephry 3.1.99