zephyrproject-rtos/zephyr #35321: IMPROVE STM32: SERIAL DRIVER,HANDLE UART MODE PER INSTANCE

manoj153, 2021-05-16:
The goal is to have each serial (instance)  mode to be configured independently as either, interrupt, or async.

 the issue I noticed is, the shell uses interrupt API, while when we enable the`CONFIG_UART_ASYNC_API=y` there is no mean to make a particular UART instance driven using interrupt mechanism, as such the shell is completely not functional.

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/issues/34672 
The issue above describes the problem related to the shell with the current arrangement of serial driver

I can contribute a PR to address this, I would like to hear some inputs.

shlomow, 2021-05-17:
@manoj153 I fixed the issue. The bug is described in the commit message. In general the only place that has shared code between interrupt api and async api is the isr function. So when there is a strange issue between the two modes the isr is the place to look

manoj153, 2021-05-18:
@shlomow I'll do a testing a verify the commit

erwango, 2021-05-18:
> IMPROVE STM32: SERIAL DRIVER,HANDLE UART MODE PER INSTANCE

@manoj153 caps lock ? ;-)

manoj153, 2021-05-18:
> > IMPROVE STM32: SERIAL DRIVER,HANDLE UART MODE PER INSTANCE
> 
> @manoj153 caps lock ? ;-)

my bad, I am used to writing title as caps, should downcase them all?

erwango, 2021-05-18:
> my bad, I am used to writing title as caps, should downcase them all?

In general all caps is seen as a bit aggressive (search for "all caps" in google),
so we use mixed or all lower cases, as one prefer.

manoj153, 2021-05-18:
> > my bad, I am used to writing title as caps, should downcase them all?
> 
> In general all caps is seen as a bit aggressive (search for "all caps" in google),
> so we use mixed or all lower cases, as one prefer.

Thank you, appreciate your feedback, I have made the changes to the title

PR #35371 (2021-05-19): drivers: uart: stm32: fix handling interrupt and async api in isr
When using the uart driver with interrupt and async api
at the same time (instance for interrupt and instance for async),
the transmission complete interrupt was handled in the async
handling section, even when interrupt driven api is used.
This caused transmission to not work properly in interrupt mode.
The fix is to move the interrupt mode handling to the begginning
of the isr. If async mode is used then interrupt mode code
will not be run.

Fixes: #35321
closed 2021-05-19 by galak