zephyrproject-rtos/zephyr #35333: RTC Time Date out of the box feature request

cederom, 2021-05-17:
**Is your feature request related to a problem? Please describe.**
* Currently there is no Time Date framework in Zephyr. We need to create by hand this functionality.

**Describe the solution you'd like**
* Time Date functionality with full POSIX support.
* Time Date set to match current time.
* Time Date is preserved across reboot.

**Describe alternatives you've considered**
* I am using nRF52 chip and NRF SDK to use RTC "by hand".

**Additional context**
* Common OS Time Date seems very important functionality not only for basic functions but also from a security standpoint.

Please consider implementing. Thank you :-)

nimarty, 2021-05-19:
I was also looking exactly for such a feature. I'm coming from Linux where I used stime() from std C Lib to set system time and sync it to hwclock. I would appreciate to see something similar in zepyhr in the future.

manoj153, 2021-12-15:
What is the plan to move this forward?
Can someone share? I can put a PR.

arbrauns, 2023-04-05:
#52618 has taken the first step here by implementing an RTC driver framework.

Still missing is the integration with POSIX time functions, and of course actual drivers for RTC peripherals.

cederom, 2024-02-08:
I have moved to NuttX RTOS some time ago so not relevant for me anymore. If not implemented yet then not relevant for anyone else. Closing. Thanks :-)

closed 2024-02-08 by cederom

arbrauns, 2024-02-09:
Please reopen, just because you don't use Zephyr anymore doesn't mean the issue has fixed itself.

cederom, 2024-02-09:
Ah, it was not completed, true :-)