zephyrproject-rtos/zephyr #35336: tests: samples: power: samples/subsys/pm/device_pm/sample.power.ospm.dev_idle_pm fails on nrf52 platforms

PerMac, 2021-05-17:
**Describe the bug**
The test test_power_state_trans from samples/subsys/pm/device_pm/ fails on-target tests on nrf52 boards

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52840dk connected
2. go to your zephyr dir
3. call `/scripts/twister -T /samples/subsys/pm/ -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See error

**Expected behavior**
The test passes

**Impact**
Not clear

**Logs and console output**
```
ERROR   - *** Booting Zephyr OS build zephyr-v2.5.0-3879-gdf691ba13ac1  ***
Device PM sample app start
open()
parent resuming..
Async wakeup request queued
child resuming..
Dummy device resumed
write()
transfer()
read()
transfer()
close()
child suspending..
Device PM sample app complete
parent suspending..
```
and hangs until timeout

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK 0.12.2
 - Commit SHA or Version used: zephyr-v2.5.0-3879-gdf691ba13ac1

ceolin, 2021-05-17:
Looking it. The test is wrong to start, looking the harness we can find:
```
        - "Device PM sample app start"
        - "parent resuming\\.\\."
        - "child resuming\\.\\."
        - "Async wakeup request queued"
        - "Dummy device resumed"
```
we got the `child resuming  before the "requested queued`, what is wrong since child resuming is printed in `dummy_device_pm_ctrl`, this function is called only after the queue has ran. The last message is also switched, the dummy driver is using an async function to put the parent so it cannot expect that this will run before the last print. This test was exploiting all the issues that this api had :/. The timeout is something that I don't know yet though. Working to fix this test.

ceolin, 2021-05-17:
ok, the timeout is just because the harness was wrong.

PR #35368 (2021-05-17): samples: pm: Fix device pm sample
"child resuming" text is printed when dummy_device_pm_ctrl is
called. The driver is using the async api, this means that this call
will happen only when the worqueue runs what will happen only when
this thread blocks waiting on the conditional variable.

The last thing is, the dummy driver was putting the parent driver
asynchronously consequently the "parent suspending" message would just
print after "Device PM sample app complete". Just use the sync API to
get these messages printed in the expected order.

Fixes #35336
commit 178cfd09 (2021-05-17): samples: pm: Fix device pm sample

"child resuming" text is printed when dummy_device_pm_ctrl is
called. The driver is using the async api, this means that this call
will happen only when the worqueue runs what will happen only when
this thread blocks waiting on the conditional variable.

The last thing is, the dummy driver was putting the parent driver
asynchronously consequently the "parent suspending" message would just
print after "Device PM sample app complete". Just use the sync API to
get these messages printed in the expected order.

Fixes #35336
closed 2021-05-17 by nashif