zephyrproject-rtos/zephyr #35344: [Coverity CID: 236075] Out-of-bounds access in subsys/bluetooth/controller/hci/hci.c

zephyrbot, 2021-05-17:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/21d1ad3762302b3e461953df59430c77e0709274/subsys/bluetooth/controller/hci/hci.c#L5154

Category: Memory - corruptions
Function: `le_ext_adv_report`
Component: Bluetooth
CID: [236075](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=236075)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/subsys/bluetooth/controller/hci/hci.c#L5154

```
5148             } else {
5149                     adv_info->direct_addr.type = 0U;
5150                     memset(&adv_info->direct_addr.a.val[0], 0, sizeof(bt_addr_t));
5151             }
5152     
5153             adv_info->length = data_len;
>>>     CID 236075:  Memory - corruptions  (OVERRUN)
>>>     Overrunning buffer pointed to by "data" of 253 bytes by passing it to a function which accesses it at byte offset 256 using argument "data_len" (which evaluates to 230). [Note: The source code implementation of the function has been overridden by a builtin model.]
5154             memcpy(&adv_info->data[0], data, data_len);
5155     
5156             node_rx_extra_list_release(node_rx->hdr.rx_ftr.extra);
5157     }
5158     
5159     static void le_adv_ext_report(struct pdu_data *pdu_data,
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v29271/p12996

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

cvinayak, 2021-05-19:
In step 197, data_len value is not considered in the coverity analysis, which should otherwise not cause data_len to be over 228 bytes. Instead data_len is assigned a value of 230 by coverity and hence causing the analysis to fail.

False positive.

carlescufi, 2021-05-19:
Marked as false positive in Coverity, closing.

cvinayak, 2021-05-20:
Reopening as a bug was identified for which PR is here https://github.com/zephyrproject-rtos/zephyr/pull/35453

PR #35453 (2021-05-21): Bluetooth: controller: Fix data length calculation in adv report
Max data length has to be adjusted for subevent code, otherwise we'll
hit an assert when trying to add data to event due to insufficient
free space.

Fixes #35344.
closed 2021-05-21 by carlescufi