zephyrproject-rtos/zephyr #35354: Adding support for measurement of Ultraviolet(UV) Light.

murali-tejeshwar, 2021-05-17:
The Zephyr sensor subsystem currently does not support the measurement of Ultraviolet(UV) light.

This can be solved by adding a new enumeration entry in the header file "include/drivers/sensor.h" so that sensors that measure UV light(such as Silicon Labs Si1133) can report UV measurements on this UV channel.

The following two lines are to be added to the  "enum sensor_channel", which is present in the above header file.
/** Ultraviolet index, in UVI. */
SENSOR_CHAN_UV,

henrikbrixandersen, 2021-05-22:
The sensors subsystem allows sensor-specific extensions to the channels enumeration. See e.g. https://github.com/zephyrproject-rtos/zephyr/blob/main/include/drivers/sensor/mcux_acmp.h

nashif, 2021-05-28:
For something generic I think it should be ok to submit this as a new channel and have it in the sensor  API. So, submit a PR please.

murali-tejeshwar, 2021-07-29:
@nashif thank you. I have submitted a PR(#37259). Requesting @MaureenHelm and @avisconti to look into this, so that I can submit a driver patch(for Si1133 sensor) which references this new channel(SENSOR_CHAN_UV).

MaureenHelm, 2021-07-29:
> I have submitted a PR(#37259). Requesting @MaureenHelm and @avisconti to look into this, so that I can submit a driver patch(for Si1133 sensor) which references this new channel(SENSOR_CHAN_UV).

Can you submit the driver in the same PR that introduces the new channel? First commit for the new channel, second commit for the driver that uses it?

avisconti, 2021-07-30:
> Can you submit the driver in the same PR that introduces the new channel? First commit for the new channel, second commit for the driver that uses it?

Yeah, that would be easier to review and to refer to it later on.

murali-tejeshwar, 2021-08-01:
@MaureenHelm @galak I have submitted a new PR(#37354) that includes both the driver and the new channel.

closed 2021-09-01 by murali-tejeshwar