zephyrproject-rtos/zephyr #35363: bt_gatt_discover() retunrs incorrect handle (offset by -1)

iceananas, 2021-05-17:
**Describe the bug**
I am build a BLE central which scans for a custom BLE peripheral with it's own custom services and characteristics. 
The central scans for peripherals, scans their services and return their characterstic handles. The identification is done with 128bit UUIDs. 

bt_gatt_discover() returns the handle which are offset by -1. E.g. The characteristic has the handle 0x20 but the returned handle is 0x19. 

**To Reproduce**
Steps to reproduce the behavior:
1. Modify a central sample (e.g. Central-HR) with custom UUIDs instead of Hearttrate service.
2. Observe console output of returned attribute handles in discover_func(),
3. Compare with the actual handle (e.g. using nRF Connect Desktop).

**Expected behavior**
bt_gatt_discover() returns the correct handle.

**Impact**
Potential showstopper, costed me a day of work debugging this.

**Logs and console output**
-

Add any other context about the problem here.
-

joerchan, 2021-05-19:
@iceananas Can you please show us your local modifications?

iceananas, 2021-05-19:
Sure!

This is how I define my target characteristic UUIDs
 ```c
static struct bt_uuid_128 control_uuid = BT_UUID_INIT_128(BT_UUID_128_ENCODE(0x1d89ca61, 0x7967, 0x4bcd, 0x9fda, 0xdda8013ada2c));
static struct bt_uuid_128 ww_uuid = BT_UUID_INIT_128(BT_UUID_128_ENCODE(0x00b12566, 0xdcb3, 0x4c4f, 0xaf5c, 0x82a20d56f2b5));
 ```
 
 This is my callback when a connection was established and I call the discover function

```c
// Handler after sucessful connection
static void connected(struct bt_conn *conn, uint8_t err) {
    
    [...]

    discover_params.uuid = &control_uuid.uuid;
    discover_params.func = discover_func;
    discover_params.start_handle = BT_ATT_FIRST_ATTTRIBUTE_HANDLE;
    discover_params.end_handle = BT_ATT_LAST_ATTTRIBUTE_HANDLE;
    discover_params.type = BT_GATT_DISCOVER_PRIMARY;

    err = bt_gatt_discover(default_conn, &discover_params);
    if (err) {
        LOG_ERR("Discover failed(err %d)", err);
        return;
    }
    
    connection_successful = true;
}
```
    
And finally this is my discover callback
    
```c
//Discovery function
static uint8_t discover_func(struct bt_conn *conn, const struct bt_gatt_attr *attr, struct bt_gatt_discover_params *params) {
    int err;

    if (!attr) {
        LOG_INF("Discover complete");
        (void)memset(params, 0, sizeof(*params));
        return BT_GATT_ITER_STOP;
    }
    LOG_INF("[ATTRIBUTE] handle %u", attr->handle);

    if (!bt_uuid_cmp(params->uuid, &control_uuid.uuid)) {
        LOG_INF("Controller Service found!");
        discover_params.uuid = &ww_uuid.uuid;
        discover_params.start_handle = attr->handle + 1;
        discover_params.type = BT_GATT_DISCOVER_CHARACTERISTIC;

        err = bt_gatt_discover(conn, &discover_params);
        if (err) {
            LOG_ERR("Discover failed (err %d)", err);
        }
    } else if (!bt_uuid_cmp(params->uuid, &ww_uuid.uuid)) {
        LOG_INF("WW Characteristic discovered at handle %u", attr->handle);
        ww_handle = attr->handle;
        //ww_handle = attr->handle+1;
        return BT_GATT_ITER_STOP;
    }
    return BT_GATT_ITER_STOP;
}
```

The commented part in the else-if loop is what I refer to. If i set ww_handle to `attr->handle+1`, I can use it to write a value for the characteristic. If not, the handle is offset by -1.

joerchan, 2021-05-19:
Thanks for the response. There is no bug, but we are lacking proper documentation and samples in this area. We have an open issue to fix this here: https://github.com/zephyrproject-rtos/zephyr/issues/27763.

The reason why are you getting the "wrong" handle is because a characteristic is split into two attributes, the characteristic declaration, and the characteristic value. When doing the Discover Characteristics by UUID procedure you will receive the characteristic declaration attribute, the value of this attribute contains the handle of the characteristic value attribute.

While doing a attr->handle + 1 would probably work in most cases it is best to get the handle properly.

The below code should fix you discovery callback.

```
	if (!bt_uuid_cmp(params->uuid, &control_uuid.uuid)) {
		struct bt_gatt_service_val *service = attr->user_data;

		printk("Controller Service found!\n");
		discover_params.uuid = &ww_uuid.uuid;
		discover_params.start_handle = attr->handle + 1;
		discover_params.end_handle = service->end_handle;
		discover_params.type = BT_GATT_DISCOVER_CHARACTERISTIC;

		err = bt_gatt_discover(conn, &discover_params);
		if (err) {
		    printk("Discover failed (err %d)\n", err);
		}
	} else if (!bt_uuid_cmp(params->uuid, &ww_uuid.uuid)) {
		struct bt_gatt_chrc *chrc = attr->user_data;

		printk("WW Characteristic discovered at handle %u\n", attr->handle);
		ww_handle = chrc->value_handle;

		return BT_GATT_ITER_STOP;
	}
```

closed 2021-05-19 by joerchan

iceananas, 2021-05-19:
Thanks a lot for clarifying this! That makes sense. 

Yes some more documentation and examples would be nice, I had to guess multiple times what to do and I also had in mind that there are probably two handles, but I couldn't find the information. This helps me definitely.