zephyrproject-rtos/zephyr #35380: sys: timeutil: inconsistent types for local times

JordanYates, 2021-05-18:
**Describe the bug**

The integer types for local times for timeutil conversion functions `timeutil_sync_ref_from_local ` and `timeutil_sync_local_from_ref` are inconsistent. One presumes the local time is a signed 64bit integer, while the other assume an unsigned 64bit integer.

https://github.com/zephyrproject-rtos/zephyr/blob/cd9421a45e7254de6be913debe153dbac4be1e91/include/sys/timeutil.h#L253-L254
https://github.com/zephyrproject-rtos/zephyr/blob/cd9421a45e7254de6be913debe153dbac4be1e91/include/sys/timeutil.h#L277-L278

pabigot, 2021-05-18:
This is not a bug.  Reference times are defined to be non-negative, hence an unsigned output from `timeutil_sync_ref_from_local` (negative values result in a `-ERANGE` error return. 

On the other hand a local time reconstructed from a reference time can be negative, if the reference time corresponds to an event that occurs before the system started.

I'll see if I can tweak the documentation so this is more clear.

pabigot, 2021-05-18:
Actually that's already documented [in the description](https://docs.zephyrproject.org/latest/reference/misc/timeutil.html?highlight=timeutil#c.timeutil_sync_local_from_ref):

> An interpolated value before local time 0 is provided without error.

closed 2021-05-18 by pabigot