zephyrproject-rtos/zephyr #35390: net.socket.tls.tls_ext: frdm_k64f test failure

hakehuang, 2021-05-18:
**Describe the bug**
all testing fails

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f ..
3. make; make flash
4. See error

**Expected behavior**
test pass

**Impact**
https supporting

**Logs and console output**
```
    W: You have 1 IPv6 net_if addresses but 2 network interfaces

    W: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUNT value.

    W: You have 1 IPv4 net_if addresses but 2 network interfaces

    W: Consider increasing CONFIG_NET_IF_MAX_IPV4_COUNT value.

    *** Booting Zephyr OS build zephyr-v2.5.0-3919-gcd9421a45e72  ***

    I: Initializing network

    I: IPv4 address: 192.0.2.1

    Running test suite tls_socket_api_extension

    ===================================================================

    START - test_tls_peer_verify_none

    E: TLS handshake error: -4e

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls_ext/src/main.c:366: test_common: (r equal to -1)

    failed to connect (113)

     FAIL - test_tls_peer_verify_none in 0.19 seconds

    ===================================================================

    START - test_tls_peer_verify_optional

    E: Identical connection handler 0x20003a70 already found.

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls_ext/src/main.c:281: test_common: (r equal to -1)

    failed to listen (120)

     FAIL - test_tls_peer_verify_optional in 0.18 seconds

    ===================================================================

    START - test_tls_peer_verify_required

    W: Failed to allocate TLS context

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/net/socket/tls_ext/src/main.c:217: test_common: (r equal to -1)

    failed to create server socket (12)

     FAIL - test_tls_peer_verify_required in 0.17 seconds

    ===================================================================

    Test suite tls_socket_api_extension failed.

    ===================================================================

    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v2.5.0-3919-gcd9421a45e72

hakehuang, 2021-05-18:
@MaureenHelm , @dleach02

jukkar, 2021-05-18:
This looks similar as #34923 but disabling the Ethernet controller did not help in this case.

jukkar, 2021-05-18:
This test is actually failing also for qemu_x86 and native_posix boards.

jukkar, 2021-05-18:
@rlubos do you know what this test is trying to do, any idea what might be the culprit?

rlubos, 2021-05-18:
AFAIR just some basic TLS handshake and exchanges, I can give a closer look at why does it fail.

jukkar, 2021-05-19:
>  as it doesn't seem to be hard to fix.

Indeed the fix itself was trivial. Thanks for excellent analysis! :+1: 

This issue is a bit similar as #34964 but that one is related to server side timeouts in `recv()` (no exact culprit found yet for it), and this one was client side when doing a `connect()`

PR #35436 (2021-05-19): net: tcp: Change locking order in connection establishment
When application calls TCP connect(), the call is blocked
by a semaphore which is then released when the connection
is established. Unfortunately the semaphore release was done
before the connection was marked as established. Depending
on the configuration options set, it is possible that after
the semaphore release, the thread that is waiting on connect()
is run immediately. Because of this, the connection bookeeping
still thought that the connection was not established even if
it was. A simple solution is to release the semaphore aft [... cut]
commit 03e1872d (2021-05-19): net: tcp: Change locking order in connection establishment

When application calls TCP connect(), the call is blocked
by a semaphore which is then released when the connection
is established. Unfortunately the semaphore release was done
before the connection was marked as established. Depending
on the configuration options set, it is possible that after
the semaphore release, the thread that is waiting on connect()
is run immediately. Because of this, the connection bookeeping
still thought that the connection was not established even if
it was. A simple solution is to release the semaphore af [... cut]
closed 2021-05-19 by galak