zephyrproject-rtos/zephyr #35395: STM32F4: Infinite reboot loop due to Ethernet initialization

mrfuchs, 2021-05-18:
This is a follow-up to issue #32771.

After porting an existing project from Zephyr 2.2 to Zephyr 2.5, I'm facing infinite reboots during initialization due to Ethernet initialization. Unfortunately and contrary to the fixes suggested in #32771, neither #35218 nor disabling auto-negotiation alone helped with my setup. I'm using a custom board based on an STM32F437VI SOC.

However, combining all three patches fixes the boot behavior for me:

1. Apply #35218
2. Apply patch re. auto-negotiation disable from https://github.com/zephyrproject-rtos/zephyr/issues/32771#issuecomment-841075466
3. Comment the calls to `HAL_Delay()` in `HAL_ETH_Init()`:

```diff
 stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c b/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c
index 2b6567b1..8fee17c7 100644
--- a/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c
+++ b/stm32cube/stm32f4xx/drivers/src/stm32f4xx_hal_eth.c
@@ -338,7 +338,7 @@ HAL_StatusTypeDef HAL_ETH_Init(ETH_HandleTypeDef *heth)
   }
 
   /* Delay to assure PHY reset */
-  HAL_Delay(PHY_RESET_DELAY);
+//  HAL_Delay(PHY_RESET_DELAY);
 
   if((heth->Init).AutoNegotiation != ETH_AUTONEGOTIATION_DISABLE)
   {
@@ -475,7 +475,7 @@ HAL_StatusTypeDef HAL_ETH_Init(ETH_HandleTypeDef *heth)
     }
 
     /* Delay to assure PHY configuration */
-    HAL_Delay(PHY_CONFIG_DELAY);
+//    HAL_Delay(PHY_CONFIG_DELAY);
   }
 
   /* Config MAC and DMA */

```

Otherwise the system crashes/reboots here (with the stack trace below):
https://github.com/zephyrproject-rtos/zephyr/blob/21d1ad3762302b3e461953df59430c77e0709274/arch/arm/core/aarch32/swap.c#L41

```
Thread #1 57005 (Suspended : Breakpoint)	
	arch_swap() at swap.c:41 0x801e7a6	
	z_swap_irqlock() at kswap.h:162 0x8034074	
	z_swap() at kswap.h:173 0x8034074	
	z_tick_sleep() at sched.c:1’268 0x8034074	
	z_tick_sleep() at sched.c:1’235 0x8034074	
	z_impl_k_sleep() at sched.c:1’296 0x80340b4	
	k_sleep() at kernel.h:86 0x80398f8	
	HAL_ETH_Init() at stm32f4xx_hal_eth.c:341 0x80292f4	
	eth_initialize() at eth_stm32_hal.c:686 0x801e38e	
	z_sys_init_run_level() at device.c:61 0x80334ba	
	bg_thread_main() at init.c:149 0x8033696	
	z_thread_entry() at thread_entry.c:29 0x8038e88	
	arch_switch_to_main_thread() at thread.c:526 0x801e9a4	
	0xaaaaaaaa
```

mrfuchs, 2021-05-18:
CC: @erwango

ABOSTM, 2021-05-18:
@mrfuchs can you detail how to reproduce the issue ? 
which sample, extra configuration, your ethernet environment , ...
eventually provide your board directory
Thanks

mrfuchs, 2021-05-21:
My bad. It turned out the device is rebooting because of watchdog timeouts. As I am porting an existing project from 2.2 to 2.5, I have a `CONFIG_WATCHDOG=y` in my configuration file (but nothing more). However, since my application is loaded by a bootloader which enables the watchdog and Zephyr's watchdog driver has changed since 2.2 (cf. `CONFIG_WDT_DISABLE_AT_BOOT` and `CONFIG_IWDG_STM32_INITIAL_TIMEOUT`) the watchdog was (re-)initialized during initialization to `IWDG_STM32_INITIAL_TIMEOUT`'s default value of 100 (ms) which is not enough time to set up the NETIF, as it seems.
Adjusting either `CONFIG_IWDG_STM32_INITIAL_TIMEOUT` or `CONFIG_WDT_DISABLE_AT_BOOT` works for me without the need of applying any other patches.

closed 2021-05-21 by mrfuchs

erwango, 2021-05-21:
> My bad. It turned out the device is rebooting because of watchdog timeouts. 

Thanks a good news, thanks for keeping us posted!