zephyrproject-rtos/zephyr #35401: Enabling POSIX_API leads to SSL handshake error

mglettig, 2021-05-18:
**Describe the bug**
As soon as CONFIG_POSIX_API=y is set SSL handshake no longer works:
`net_sock_tls: TLS handshake error: -2700`

**To Reproduce**
Steps to reproduce the behavior:
1. Build http_client sample with overlay-tls.conf for QEMU.
2. Everthing works given that networking is configured correctly
3. Rebuild http_client with `CONFIG_POSIX_API=y`. Make sure to resolve compile issues by creating libc_extensions where needed.
4. See error

**Expected behavior**
I would expect that CONFIG_POSIX_API would not affect SSL handshake. I need a system with civetweb and https client at the same time. Unfortunately civetweb needs CONFIG_POSIX_API. 

**Impact**
Showstopper.

**Logs and console output**
`[00:00:03.250,000] <err> net_sock_tls: TLS handshake error: -2700`

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Zephyr Version 2.6.0-rc1

**Additional context**
Add any other context about the problem here.

rlubos, 2021-05-19:
The issue seems to be `MBEDTLS_HAVE_DATE_TIME` option, which is enforced by the `CONFIG_POSIX_API` (https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/mbedtls/configs/config-tls-generic.h#L29). After enabling this option, mbedTLS expects to have a valid time in the system, and since there's no valid time on QEMU, the certificate validity check fails (either expired or not yet valid).

IMO we should rather provide an option in mbedTLS' Kconfig that would allow the user to select whether to use date/time validation in mbedTLS instead of depending on the POSIX_API option, what do you think @jukkar?

jukkar, 2021-05-19:
> IMO we should rather provide an option in mbedTLS' Kconfig that would allow the user to select whether to use date/time validation in mbedTLS instead of depending on the POSIX_API option

I agree, using Posix API to decide that sounds error prone as we have seen.

rlubos, 2021-05-19:
Ok, I'll send a PR then, thanks.

> Make sure to resolve compile issues by creating libc_extensions where needed.

BTW, you don't need to create anything on your own, just include appropiate POSIX headers in the sample/lib. For me the following set of headers was enough:
```
#include <posix/arpa/inet.h>
#include <posix/sys/socket.h>
#include <posix/unistd.h>
```
We should probably switch to the `zsock_` API in the library so that we're not dependent on the POSIX configuration in the system.

mglettig, 2021-05-19:
@rlubos Thanks for finding out what is causing this issue and thanks for providing a PR. I think it's a great sugestion to decouple the civetweb lib from the POSIX API. Do you thank those change would get accepted by civetweb project or would it be necessary to maintain a Zephyr specific version of civetweb?

rlubos, 2021-05-19:
> I think it's a great sugestion to decouple the civetweb lib from the POSIX API. Do you thank those change would get accepted by civetweb project or would it be necessary to maintain a Zephyr specific version of civetweb?

Oh, I think you've misunderstood me - I've meant the `http_client` library from Zephyr, used by the sample referenced in this issue. As for the civetweb, I'm not able to comment unfortunately, as I don't know the details of the library, but given that it seems to rely not only on the socket API but also other APIs provided by POSIX, I don't think it'd be a trivial task.

And actually, if the civetweb intends to be a truly portable library, there's nothing wrong in using standard POSIX APIs IMO (that's not the case for the Zephyr's native http library since it's intended for Zephyr only). It's rather Zephyr who should provide missing POSIX pieces needed for the library (but again, I know too little about the library in order to comment on how much is missing).

mglettig, 2021-05-21:
@rlubos Ok I unterstand. You are of course right. I think refactoring `http_client` library to use the `zsock_` functions makes sense.

Regarding the actual issue and your PR. I verified on my setup that this solves the issue by manually disabling both compile switches: `#define MBEDTLS_HAVE_TIME` and `#define MBEDTLS_HAVE_TIME_DATE`. Thanks again. Looking forward to see that this PR gets applied to mainline and hopefully soon to the Nordic fork of Zephyr (which will be the one I'm going to use).

PR #35461 (2021-06-01): modules: mbedTLS: Add separate Kconfig entry for MBEDTLS_HAVE_TIME_DATE
Currently the MBEDTLS_HAVE_TIME_DATE mbedTLS option is enabled based on
the CONFIG_POSIX_API option. This doesn't seem right, since enabling
the POSIX API does not guarantee that there is a valid time source in
the system. This was the case for the qemu_x86 platform, where enabling
POSIX_API caused TLS handshake failures due to certificate validation
errors caused by no valid time avaialble in the system.

Fix this by adding a specific KConfig entry for date/time configuration
in mbedTLS. Applications that need to enforce  [... cut]
commit 6ca92495 (2021-06-01): modules: mbedTLS: Add separate Kconfig entry for MBEDTLS_HAVE_TIME_DATE

Currently the MBEDTLS_HAVE_TIME_DATE mbedTLS option is enabled based on
the CONFIG_POSIX_API option. This doesn't seem right, since the enabling
the POSIX API does not guarantee that there is a valid time source in
the system. This was the case for the qemu_x86 platform, where enabling
POSIX_API caused TLS handshake failures due to certificate validation
errors caused by no valid time avaialble in the system.

Fix this by adding a specific KConfig entry for date/time configuration
in mbedTLS. Applications that need to enf [... cut]
closed 2021-06-01 by galak