zephyrproject-rtos/zephyr #35437: CAN driver for LPCXpresso55S16

Qbicz, 2021-05-19:
**Is your feature request related to a problem? Please describe.**
NXP LPCXpresso5x series has already a good support in Zephyr:
https://docs.zephyrproject.org/latest/boards/arm/lpcxpresso55s16/doc/index.html
The only thing that stops me from using it in a project is a missing CAN bus support.

- Is there any ongoing effort to support CAN on this board/MCU?
- If not, how much work it is to port the driver from LPC Xpresso SDK, or extend already existing NXP CAN driver in Zephyr? What would be the best approach?

**Describe the solution you'd like**
Have CAN bus support on NXP LPCXpresso55S16.

**Describe alternatives you've considered**
Alternatively, I can use other NXP LPC microcontrollers with CAN controller, if it's easier to add CAN support for them in Zephyr.

alexanderwachter, 2021-05-19:
It seems that it is an Bock M_CAN. If that is the case, you "only" need a fronted for the generic M_CAN driver.
(see the stm32_fdcan). @legoabram already ported to a SAM SoC.

henrikbrixandersen, 2022-01-03:
@Qbicz Sorry it took so long, but I have a working CAN-FD driver for the LPC55S1x in #35832 if you would like to test it.

commit 190c9ba3 (2022-01-10): boards: arm: lpcxpresso55s16: add MCAN support

Add NXP LPC MCAN support for the NXP lpcxpresso55s16 board definition.

Fixes #35437
closed 2022-01-10 by carlescufi