zephyrproject-rtos/zephyr #35474: The dma-stm32 driver don't build for STM32F0 MCUs

simonguinot, 2021-05-19:
Here is the error message when building the dma-stm32 driver for a STM32F0 MCU:

```
drivers/dma/dma_stm32.c: In function 'dma_stm32_configure':
drivers/dma/dma_stm32.c:470:16: error: 'LL_DMA_InitTypeDef' {aka 'struct <anonymous>'} has no member named 'PeriphRequest'
  DMA_InitStruct.PeriphRequest = config->dma_slot;
                ^
```

I encountered this issue when building a Zephyr application for a nucleo_f070rb board with SPI_STM32_DMA enabled.

simonguinot, 2021-05-20:
Looking at #34972, I think it is not related.

str4t0m, 2021-05-20:
I guess FRASTM meat to refer to #34666, which omits the assignment of DMA_InitStruct.PeriphRequest for series that are compatible to st_stm32_dma_v2bis.

PR #35473 (2021-05-20): drivers: dma: stm32: fix build for STM32F0 MCUs
The STM32F0 MCUs (except STM32F030XC) don't have a DMA channel selection
register (DMA_CSELR). This patch fixes the build of the dma_stm32 driver
for them.

Fixes #35474
closed 2021-05-20 by galak