zephyrproject-rtos/zephyr #35480: pm: device_runtime: `pm_device_request` can block forever

JordanYates, 2021-05-20:
**Describe the bug**

The core runtime power management function `pm_device_request` can hang forever with a small amount of bad luck.
The core actions taken by this function are:
```
	(void)k_work_schedule(&dev->pm->work, K_NO_WAIT);
	(void)k_condvar_wait(&dev->pm->condvar, &request_mutex, K_FOREVER);
```

The nuance is that the `condvar` API only unblocks threads that are currently blocked, there is no persistent state like `k_poll_signal`.
What this means is that if the thread calling `pm_device_request ` is interrupted between the two function calls above, it is possible (and even likely, given that the system work queue runs at priority -1) that the submitted work item completes (calling `k_condvar_broadcast`) before the submitting thread blocks on the `condvar`. 

As a result, the original call to `pm_device_request` won't return (unless a second thread also runs `pm_device_request` on the same device at a later time).

**To Reproduce**
Simulate another thread interrupting the calling thread by updating `pm_device_request` to contain:
```
	(void)k_work_schedule(&dev->pm->work, K_NO_WAIT);
	k_sleep(K_SECONDS(1));
```
Build and run the device power management sample: `samples/subsys/pm/device_pm`
```
*** Booting Zephyr OS build v2.6.0-rc1-244-gaa78a6473b16  ***
Device PM sample app start
open()
parent resuming..

```
**Expected behavior**
The power management functions should be robust in the face of arbitrary interrupts and rescheduling.

**Environment (please complete the following information):**
 - Zephyr v2.6-rc1

JordanYates, 2021-05-20:
@ceolin

ceolin, 2021-05-20:
You're absolutely right, I noticed it this week. This pr fixes it https://github.com/zephyrproject-rtos/zephyr/pull/35315
This commit https://github.com/zephyrproject-rtos/zephyr/pull/35315/commits/1039186d730d2c71cd92ab90c38965745a1adec8, in particular, introduces a function to wait correctly on the codvar

closed 2021-05-26 by nashif