zephyrproject-rtos/zephyr #35489: samples: net: gsm_modem: build fails if CONFIG_GSM_MUX=y

hwilmers, 2021-05-20:
**Describe the bug**
Building of sample app gsm_modem fails if GSM muxing is enabled
by setting CONFIG_GSM_MUX=y.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/net/gsm_modem
2. west build -b reel_board
-> OK, builds fine.

Now append to prj.conf:
CONFIG_GSM_MUX=y

Repeat build:
1. west build
-> build fails

**Expected behavior**
Build should succeed.

**Impact**
Showstopper - GSM muxing can not be used.

**Logs and console output**
...
../../../../include/sys/cbprintf_internal.h:183:19: error: invalid use of undefined type 'struct gsm_dlci'
  183 |   *_Generic((arg) + 0, \
 ...

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.11.2
 - Commit: zephyr-v2.5.0-3633-geaa2dba1a232

galak, 2021-05-20:
@hwilmers this appears fixed on the main tree.

closed 2021-05-20 by galak

hwilmers, 2021-05-20:
Someone is working faster than me. Thanks ;-)