zephyrproject-rtos/zephyr #35512: OpenThread can't find TRNG driver on nRF5340

Gunzter, 2021-05-20:
**Describe the bug**
While trying to to build the [Echo Server Example](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_server/README.html#sockets-echo-server-sample) for the nRF5340DK board the compilation fail. With message "#error OpenThread requires an entropy source for a TRNG". The issue seems rooted in Devicetree to me, but I am a complete newbie to Zephyr-OS.

**To Reproduce**
cd zephyrproject/zephyr
west build -p auto -b nrf5340dk_nrf5340_cpuapp samples/net/sockets/echo_server -- -DCONF_FILE="prj.conf overlay-ot.conf"

Building this example would be the first step in a research project using TF-M and OpenThread, so unfortunately this issue looks like a showstopper.

**Logs and console output**
```
west build -p auto -b nrf5340dk_nrf5340_cpuapp samples/net/sockets/echo_server -- -DCONF_FILE="prj.conf overlay-ot.conf"
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/martin/workspace/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/martin/workspace/zephyrproject/zephyr/samples/net/sockets/echo_server
-- Zephyr version: 2.6.0-rc1 (/home/martin/workspace/zephyrproject/zephyr), build: v2.6.0-rc1-262-gaa8e34302a18
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: nrf5340dk_nrf5340_cpuapp
-- Cache files will be written to: /home/martin/.cache/zephyr
-- Using toolchain: zephyr 0.12.4 (/home/martin/zephyr-sdk-0.12.4)
-- Found dtc: /home/martin/zephyr-sdk-0.12.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/martin/workspace/zephyrproject/zephyr/boards/arm/nrf5340dk_nrf5340/nrf5340dk_nrf5340_cpuapp.dts
-- Generated zephyr.dts: /home/martin/workspace/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/martin/workspace/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/martin/workspace/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/martin/workspace/zephyrproject/zephyr/samples/net/sockets/echo_server/Kconfig
Loaded configuration '/home/martin/workspace/zephyrproject/zephyr/build/zephyr/.config'
Configuration saved to '/home/martin/workspace/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/martin/workspace/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
CMake Warning at ../../../../subsys/random/CMakeLists.txt:11 (message):
  

      Warning: CONFIG_TEST_RANDOM_GENERATOR is not a truly random generator.
      This capability is not secure and it is provided for testing purposes only.
      Use it carefully.

CMake Deprecation Warning at /home/martin/workspace/zephyrproject/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- libmetal version: 1.0.0 (/home/martin/workspace/zephyrproject/zephyr/samples/net/sockets/echo_server)
-- Build type:  
-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: cortexm
-- open-amp version: 1.0.0 (/home/martin/workspace/zephyrproject/modules/lib/open-amp/open-amp)
-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: cortexm
-- C_FLAGS :  -Wall -Wextra
-- OpenThread Source Directory: /home/martin/workspace/zephyrproject/modules/lib/openthread
-- Package Name: OPENTHREAD
-- Setting default package version: gcdb957090
-- Package Version: gcdb957090
-- Configuring done
CMake Warning (dev) at ../../../../cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  ../../../../CMakeLists.txt:1198 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at ../../../../cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
[... 43 lines cut ...]
  518 |  return nrf_802154_time_get();
      |         ^~~~~~~~~~~~~~~~~~~
      |         nrf_802154_tx_power_get
[368/566] Building C object zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/entropy.c.obj
FAILED: zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/entropy.c.obj 
ccache /home/martin/zephyr-sdk-0.12.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=v2.6.0-rc1-262-gaa8e34302a18 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"config-tls-generic.h\" -DNRF5340_XXAA_APPLICATION -DNRF_802154_ACK_TIMEOUT_ENABLED=1 -DNRF_802154_CCA_CORR_LIMIT_DEFAULT="" -DNRF_802154_CCA_CORR_THRESHOLD_DEFAULT="" -DNRF_802154_CCA_ED_THRESHOLD_DEFAULT="" -DNRF_802154_CSMA_CA_ENABLED=1 -DNRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0 -DNRF_802154_PENDING_EXTENDED_ADDRESSES="" -DNRF_802154_PENDING_SHORT_ADDRESSES="" -DNRF_802154_RX_BUFFERS=16 -DNRF_802154_TX_STARTED_NOTIFY_ENABLED=1 -DNRF_802154_USE_RAW_API=1 -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -Izephyr/include/generated -I../soc/arm/nordic_nrf/nrf53 -I../lib/libc/newlib/include -I../lib/util/fnmatch/. -I../subsys/net/lib/sockets/. -I../subsys/net/lib/conn_mgr/. -I../subsys/net/lib/openthread/platform/. -I../subsys/settings/include -I/home/martin/workspace/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/martin/workspace/zephyrproject/modules/hal/nordic/nrfx -I/home/martin/workspace/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/martin/workspace/zephyrproject/modules/hal/nordic/nrfx/mdk -I../modules/hal_nordic/nrfx/. -Imodules/libmetal/libmetal/lib/include -I/home/martin/workspace/zephyrproject/modules/lib/open-amp/open-amp/lib/include -I/home/martin/workspace/zephyrproject/modules/crypto/mbedtls/mbedtls/include -I../modules/mbedtls/configs -I/home/martin/workspace/zephyrproject/modules/hal/nordic/drivers/nrf_802154/serialization/include -I/home/martin/workspace/zephyrproject/modules/hal/nordic/drivers/nrf_802154/serialization/include/platform -I/home/martin/workspace/zephyrproject/modules/hal/nordic/drivers/nrf_802154/serialization/include/serialization -I/home/martin/workspace/zephyrproject/modules/hal/nordic/drivers/nrf_802154/serialization/include/host -Imodules/openthread/build/etc/cmake -I/home/martin/workspace/zephyrproject/modules/lib/openthread/etc/cmake -I/home/martin/workspace/zephyrproject/modules/lib/openthread/include -isystem /home/martin/workspace/zephyrproject/modules/lib/openthread/zephyr/../include -isystem /home/martin/workspace/zephyrproject/modules/lib/openthread/zephyr/../examples/platforms -Os -imacros /home/martin/workspace/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -fdiagnostics-color=always -mcpu=cortex-m33 -mthumb -mabi=aapcs -imacros /home/martin/workspace/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/martin/workspace/zephyrproject/zephyr/samples/net/sockets/echo_server=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/martin/workspace/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/martin/workspace/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -std=c99 -MD -MT zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/entropy.c.obj -MF zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/entropy.c.obj.d -o zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/entropy.c.obj -c /home/martin/workspace/zephyrproject/zephyr/subsys/net/lib/openthread/platform/entropy.c
/home/martin/workspace/zephyrproject/zephyr/subsys/net/lib/openthread/platform/entropy.c:19:2: error: #error OpenThread requires an entropy source for a TRNG
[... cut ...]
/home/martin/workspace/zephyrproject/zephyr/subsys/net/lib/openthread/platform/entropy.c:33:28: note: each undeclared identifier is reported only once for each function it appears in
[373/566] Building C object zephyr/subsys/net/lib/openthread/platform/CMakeFiles/openthread_platform.dir/radio.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/martin/workspace/zephyrproject/zephyr/build

```

**Environment (please complete the following information):**
Ubuntu 18.04
Zephyr SDK
commit aa8e34302a184b9cd13b0aa10d1a4fd2a1b104b1

rlubos, 2021-05-24:
Hi,
OpenThread won't work for nRF5340 with vanilla Zephyr, i.e. it requires CryptoCell entropy which is only availalbe in the Zephyr-based nRF Connect SDK. Please use [nRF Connect SDK](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/getting_started.html) instead.

carlescufi, 2021-05-26:
Closing as per @rlubos' comment.

closed 2021-05-26 by carlescufi

pawelosyp-hid, 2022-09-21:
Hi all,
Has anything changed in this topic? Driver to CryptoCell entropy or other solution for that?
btw is it possible to make multiprotocol setup (BLE and OT) ? @rlubos

rlubos, 2022-09-21:
> Has anything changed in this topic? Driver to CryptoCell entropy or other solution for that?

Nothing has changed in this regard.

> btw is it possible to make multiprotocol setup (BLE and OT) ?

Yes, it is possible, but you also need NCS for that:
https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_multiprotocol_support.html#ug-multiprotocol-support

pawelosyp-hid, 2022-09-21:
Are there any chances that it will be implemented in vanilla Zephyr? Or any part of that? Thanks for responce.

rlubos, 2022-09-21:
I don't think there are plans for that as of today.