zephyrproject-rtos/zephyr #35522: doc: Current section is not shown in the side pane nor the page top cookie

yashi, 2021-05-21:
**Describe the bug**
Pages about Device tree bindings doesn't show
- side pane index
- page cookie at the top

**To Reproduce**
Open one of many binding pages at https://docs.zephyrproject.org/latest/reference/devicetree/bindings.html

e.g) https://docs.zephyrproject.org/latest/reference/devicetree/bindings/cpu/arm%2Ccortex-m3.html#dtbinding-arm-cortex-m3

**Expected behavior**
1) side pane shows the current section.
2) top cookies show the current location.

**Impact**
If you land on one of these pages, you don't know where you are.
It doesn't hurt anyone but nice to have them shown.

gmarull, 2021-05-21:
These are pages marked as `:orphan:`, so it is expected.

yashi, 2021-05-21:
Thanks. I didn't know it was intentional.  Closing.

closed 2021-05-21 by yashi