zephyrproject-rtos/zephyr #35524: tests: samples: led: LED PWM sample fails on nrf platforms

PerMac, 2021-05-21:
**Describe the bug**
The test sample.drivers.led.led_pwm from /zephyr/samples/drivers/led_pwm fails on nrf platforms. The printed output is:
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.5.0-4044-g58d3760533a6  ***
DEBUG   - DEVICE: [00:00:00.445,037] <inf> main: Found device pwmleds
DEBUG   - DEVICE: [00:00:00.445,068] <inf> main: Testing LED 0 - no label
DEBUG   - DEVICE: [00:00:00.445,068] <inf> main:   Turning on
DEBUG   - DEVICE: [00:00:01.445,159] <inf> main:   Turning off
DEBUG   - DEVICE: [00:00:02.445,251] <inf> main:   Increasing brightness gradually
DEBUG   - DEVICE: [00:00:04.462,493] <inf> main:   Blinking on: 0.1 sec, off: 0.1 sec
DEBUG   - DEVICE: [00:00:09.462,585] <inf> main:   Blinking on: 1 sec, off: 1 sec
DEBUG   - DEVICE: [00:00:09.462,615] <err> pwm_nrfx: Prescaler for period_cycles 32000000 not found.
DEBUG   - DEVICE: [00:00:09.462,615] <err> main: err=-22
DEBUG   - DEVICE: [00:00:09.462,615] <inf> main: Testing LED 0 - no label
DEBUG   - DEVICE: [00:00:09.462,615] <inf> main:   Turning on
DEBUG   - DEVICE: [00:00:10.462,707] <inf> main:   Turning off
DEBUG   - DEVICE: [00:00:11.462,768] <inf> main:   Increasing brightness gradually
DEBUG   - DEVICE: [00:00:13.480,010] <inf> main:   Blinking on: 0.1 sec, off: 0.1 sec
DEBUG   - Timed out while monitoring serial output on nrf52dk_nrf52832
DEBUG   - run status: nrf52dk_nrf52832/samples/drivers/led_pwm/sample.drivers.led.led_pwm timeout
INFO    - 1/1 nrf52dk_nrf52832          samples/drivers/led_pwm/sample.drivers.led.led_pwm FAILED Timeout (device 18.206s)
```
It seems the device name was changed to `pwmleds`. The sample.yaml tells that `"Found device LED_PWM_\\d+"` is been looked for. Changing to `LED_PWM_` to `pwmleds` makes the sample pass. 

I can fix this by myself, however I am worried about these lines in the output:
```
DEBUG   - DEVICE: [00:00:09.462,615] <err> pwm_nrfx: Prescaler for period_cycles 32000000 not found.
DEBUG   - DEVICE: [00:00:09.462,615] <err> main: err=-22
```
It reports an error, but the error is handled nowhere. Should the test be fixed to fail if there are such errors or is it "a feature"?

**To Reproduce**
Steps to reproduce the behavior:
1. connect e.g. nrf52dk_nrf52832
2. run `./scripts/twister -T samples/drivers/led_pwm/ -p nrf52dk_nrf52832 --device-testing --device-serial /dev/ttyACM0 -v -v --inline-logs`
3. See error

**Expected behavior**
Test passes. Errors are handled properly

**Environment:**
 - OS: Ubuntu 18.04
 - Toolchain zephyr-sdk 0.12.2
 - Commit SHA or Version used zephyr-v2.5.0-4044-g58d3760533a6

ABOSTM, 2021-05-21:
Same sample failure on STM32 boards (nucleo_g474re and nucleo_h743zi), but no error in the output log
```
*** Booting Zephyr OS build v2.6.0-rc1-197-gca0f31ee89fb  ***
[00:00:00.005,000] <inf> main: Found device LED_PWM_0
[00:00:00.005,000] <inf> main: Testing LED 0 - no label
[00:00:00.005,000] <inf> main:   Turning on
[00:00:01.005,000] <inf> main:   Turning off
[00:00:02.005,000] <inf> main:   Increasing brightness gradually
[00:00:04.025,000] <inf> main:   Blinking on: 0.1 sec, off: 0.1 sec
[00:00:09.025,000] <inf> main:   Blinking on: 1 sec, off: 1 sec
[00:00:14.026,000] <inf> main:   Turning off
```

MarekPieta, 2021-05-24:
It seems like the PWM period is too long and it's not supported by the pwm_nrfx driver.
Maybe we could just print information that it's not supported by the driver without marking this as an error?
I am not sure if it should be mandatory for PWM LED driver implementation to support so long periods for blink operation. Blink operation could be implemented using k_work_delayable that resubmits itself periodically aswell.

PerMac, 2021-05-24:
I belive updating the label in sample.yaml regex will make the test pass. I will provide a PR in a moment with the updated label.
It seems the error from the log (nrfx driver) is not treated as an error by twister. We will decide how to handle communicating that >0.25s periods are not supported by nrfx driver.

closed 2021-05-25 by nashif

FRASTM, 2021-05-25:
samples/drivers/led_pwm on nucleo_g474re : 
```
*** Booting Zephyr OS build v2.6.0-rc1-348-g965a39bb7f3a  ***
[00:00:00.005,000] <inf> main: Found device pwmleds
[00:00:00.005,000] <inf> main: Testing LED 0 - no label
[00:00:00.005,000] <inf> main:   Turned on
[00:00:01.005,000] <inf> main:   Turned off
[00:00:02.005,000] <inf> main:   Increasing brightness gradually
[00:00:04.025,000] <inf> main:   Blinking on: 0.1 sec, off: 0.1 sec
[00:00:09.025,000] <inf> main:   Blinking on: 1 sec, off: 1 sec
[00:00:14.026,000] <inf> main:   Turned off, loop end
```

PerMac, 2021-05-25:
@FRASTM Does your comment mean you are still seeing failure or just reporting the current output? Does the test pass now on nucleo_g474re?

FRASTM, 2021-05-26:
Yes, tests led_pwm **passed** on nucleo_g474re and nucleo_h743zi

PR #35576 (2021-05-25): tests: pwm: led: Fix pwm_led sample
The PR fixes the pwm_led sample:
- now test doesn't rely on any specific device name
- Logs which are scanned with regex in the test are printed
  only after a command passes (before failures were not
  affecting the test verdict)
- If 1 sec cycle is not supported appropriate info is printed
  and won't cause the test to failed
- Changed second "Turned off" msg so regex doesn't mix it with the
  the first one.

Fixes #35524