zephyrproject-rtos/zephyr #35532: SSL Handshake error with modified http(s) client example

mglettig, 2021-05-21:
**Describe the bug**
I successfully tested the http(s) client sample on both my target (SAME70 XPlained board) and QEMU. The sample is configured to contact a local webserver that needs to be started with python. So far so good. After the sample was successful I wanted to run a HTTPS request on a remote server (httpbin.org).

This request leads to a SSL handshake error: `TLS handshake error: -7100`

**To Reproduce**
Steps to reproduce the behavior:
1. Modify the sample with the following changes: 
```
diff --git a/samples/net/sockets/http_client/CMakeLists.txt b/samples/net/sockets/http_client/CMakeLists.txt
index 82465eda1f..42def9d4d5 100644
--- a/samples/net/sockets/http_client/CMakeLists.txt
+++ b/samples/net/sockets/http_client/CMakeLists.txt
@@ -11,6 +11,6 @@ set(gen_dir ${ZEPHYR_BINARY_DIR}/include/generated/)
 
 generate_inc_file_for_target(
     app
-    src/https-cert.der
+    src/httpbin_org_cert_bin.der
     ${gen_dir}/https-cert.der.inc
     )
diff --git a/samples/net/sockets/http_client/prj.conf b/samples/net/sockets/http_client/prj.conf
index dda4d34b88..a5a8cd1a3f 100644
--- a/samples/net/sockets/http_client/prj.conf
+++ b/samples/net/sockets/http_client/prj.conf
@@ -18,9 +18,9 @@ CONFIG_NET_CONFIG_SETTINGS=y
 CONFIG_NET_CONFIG_NEED_IPV4=y
 CONFIG_NET_CONFIG_NEED_IPV6=y
 CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
-CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
+CONFIG_NET_CONFIG_MY_IPV4_GW="2001:db8::1"
 # Address of HTTP IPv4 server
-CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
+CONFIG_NET_CONFIG_PEER_IPV4_ADDR="34.231.30.52"
 CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
 # Address of HTTP IPv6 server
 CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
diff --git a/samples/net/sockets/http_client/src/ca_certificate.h b/samples/net/sockets/http_client/src/ca_certificate.h
index 4e55d4198c..876497d47d 100644
--- a/samples/net/sockets/http_client/src/ca_certificate.h
+++ b/samples/net/sockets/http_client/src/ca_certificate.h
@@ -6,10 +6,11 @@
 
 #define CA_CERTIFICATE_TAG 1
 
-#define TLS_PEER_HOSTNAME "localhost"
+#define TLS_PEER_HOSTNAME NULL
 
 /* This is the same cert as what is found in net-tools/https-cert.pem file
  */
+
 static const unsigned char ca_certificate[] = {
 #include "https-cert.der.inc"
 };
diff --git a/samples/net/sockets/http_client/src/main.c b/samples/net/sockets/http_client/src/main.c
index 33ea253954..cc8ca50d92 100644
--- a/samples/net/sockets/http_client/src/main.c
+++ b/samples/net/sockets/http_client/src/main.c
@@ -15,7 +15,7 @@ LOG_MODULE_REGISTER(net_http_client_sample, LOG_LEVEL_DBG);
 #include "ca_certificate.h"
 
 #define HTTP_PORT 8000
-#define HTTPS_PORT 4443
+#define HTTPS_PORT 443
 
 #if defined(CONFIG_NET_CONFIG_PEER_IPV6_ADDR)
 #define SERVER_ADDR6  CONFIG_NET_CONFIG_PEER_IPV6_ADDR
@@ -382,5 +382,6 @@ void main(void)
 		k_sleep(K_FOREVER);
 	}
 
+
 	exit(0);
 }
```

2. Make sure the following certificate is located in the sample (.der file): 
[certificate.zip](https://github.com/zephyrproject-rtos/zephyr/files/6522008/certificate.zip)

3. Build for QEMU.
4. Run on QEMU (make sure access to internet is permitted otherwise run the test on a target).

**Expected behavior**
Successful HTTP GET request.

**Impact**
Secure cloud connectivity not possible.

**Logs and console output**

Logs with QEMU:
```
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)

iPXE (http://ipxe.org) 00:03.0 CA00 PCI2.10 PnP PMM+003924B0+002F24B0 CA00
                                                                               

Booting from ROM..*** Booting Zephyr OS build v2.6.0-rc1  ***

[00:00:00.110,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.110,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:03.130,000] <err> net_http_client_sample: Cannot connect to IPv4 remote (-60)
[00:00:06.140,000] <err> net_http_client_sample: Cannot connect to IPv6 remote (-60)
[00:00:06.140,000] <dbg> net_http.Data to send
                                  47 45 54 20 2f 20 48 54  54 50 2f 31 2e 31 0d 0a |GET / HT TP/1.1..
                                  48 6f 73 74 3a 20 33 34  2e 32 33 31 2e 33 30 2e |Host: 34 .231.30.
                                  35 32 0d 0a 0d 0a                                |52....           
[00:00:06.150,000] <dbg> net_http.Data to send
                                  47 45 54 20 2f 20 48 54  54 50 2f 31 2e 31 0d 0a |GET / HT TP/1.1..
                                  48 6f 73 74 3a 20 32 30  30 31 3a 64 62 38 3a 3a |Host: 20 01:db8::
                                  32 0d 0a 0d 0a                                   |2....            
[00:00:06.460,000] <err> net_sock_tls: TLS handshake error: -7100
[00:00:06.460,000] <err> net_http_client_sample: Cannot connect to IPv4 remote (-53)
[00:00:09.470,000] <err> net_http_client_sample: Cannot connect to IPv6 remote (-60)
[00:00:09.470,000] <dbg> net_http.Data to send
                                  50 4f 53 54 20 2f 66 6f  6f 62 61 72 20 48 54 54 |POST /fo obar HTT
                                  50 2f 31 2e 31 0d 0a 48  6f 73 74 3a 20 33 34 2e |P/1.1..H ost: 34.
                                  32 33 31 2e 33 30 2e 35  32 0d 0a 43 6f 6e 74 65 |231.30.5 2..Conte
                                  6e 74 2d 4c 65 6e 67 74  68 3a 20 36 0d 0a 0d 0a |nt-Lengt h: 6....
[00:00:09.480,000] <dbg> net_http.Data to send
                                  50 4f 53 54 20 2f 20 48  54 54 50 2f 31 2e 31 0d |POST / H TTP/1.1.
                                  0a 48 6f 73 74 3a 20 32  30 30 31 3a 64 62 38 3a |.Host: 2 001:db8:
                                  3a 32 0d 0a 43 6f 6e 74  65 6e 74 2d 4c 65 6e 67 |:2..Cont ent-Leng
                                  74 68 3a 20 36 0d 0a 0d  0a                      |th: 6... .       
[00:00:09.780,000] <err> net_sock_tls: TLS handshake error: -7100
[00:00:09.780,000] <err> net_http_client_sample: Cannot connect to IPv4 remote (-53)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Release 2.6-rc1

**Additional context**
I exported the certificate using Chrome on Linux in the DER binary format.

jukkar, 2021-05-21:
This sounds like an issue that would be fixed by #35461

mglettig, 2021-05-21:
Hi @jukkar. No unfortunately not. I disabled `MBEDTLS_HAVE_TIME_DATE ` and `MBEDTLS_HAVE_TIME_DATE` compile switches and it happens still. Also `CONFIG_POSIX_API` is not set in the sample.

jukkar, 2021-05-21:
@rlubos any suggestion what might be wrong here?

jukkar, 2021-05-24:
> [00:00:03.130,000] <err> net_http_client_sample: Cannot connect to IPv4 remote (-60)
> [00:00:06.140,000] <err> net_http_client_sample: Cannot connect to IPv6 remote (-60)

Perhaps the issue is related to this error, could it be that the connection cannot be established in the first place. The error values were changed recently but if I recall correctly, the 60 was ETIMEDOUT.

jukkar, 2021-05-24:
>  CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.2.1"
> -CONFIG_NET_CONFIG_MY_IPV4_GW="192.0.2.2"
> +CONFIG_NET_CONFIG_MY_IPV4_GW="2001:db8::1"
>  # Address of HTTP IPv4 server
> -CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.0.2.2"
> +CONFIG_NET_CONFIG_PEER_IPV4_ADDR="34.231.30.52"

Your own address is 192.0.2.1, and the peer address is 34.231.30.52 but you have removed the default gateway address. This will mean connection to peer will most probably not succeed.

But then in your overlay file, you enable the gateway setting back. Is the gateway able to reroute the packet from zephyr to outside system?

mglettig, 2021-05-25:
Thanks for looking into this @jukkar. As you already noticed the GW is overwritten by the overlay so all correct there. Routing to the outside system and to the internet works as I was successful with the following command from the net shell:
`net ping 8.8.8.8`
I don't think that [ETIMEDOUT](https://github.com/zephyrproject-rtos/zephyr/blob/v2.6.0-rc1/lib/libc/minimal/include/errno.h#L98) is the root cause of this issue. I guess this occurs because the sample tries to run requests before Ethernet is ready. I rather suspect some issues with the certificate. Maybe the format is not what Zephyr/mbedtls is expecting? Do you have a certificate that works for sure that I can try out with a remote server?

rlubos, 2021-05-25:
`samples/net/sockets/http_get` with `overlay-tls.conf` works out of the box. I'm not sure about the actual issue here (the mbedtls error code is too generic, `MBEDTLS_ERR_SSL_BAD_INPUT_DATA`, but I'd check mbedTLS heap size in the first place - it's often an issue that the heap is too small. 

As for the certificate, DER format should be handled out of the box, if you used PEM you'd need to enable `CONFIG_MBEDTLS_PEM_CERTIFICATE_FORMAT` option. But if the certificate is malformed somehow then we could also see errors like this I guess.

mglettig, 2021-06-02:
I figured out what was wrong here. A combination of 2 things:
- I didn't use the Root CA certificate but the server certificate. Correct would be to use the root ca certificate (the first certificate of the trust chain).
- I needed to set `CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=4096` or more.

On one hand the issue is solved for me now. On the other hand I think it would be good to provide a https client sample that speaks to a server in the internet so that other people don't need struggle with the same problems that I had.

jukkar, 2021-06-02:
> On the other hand I think it would be good to provide a https client sample that speaks to a server in the internet so that other people don't need struggle with the same problems that I had.

As there are multiple ways to configure and use mbedtls, it can be difficult to provide a setup that works for everyone. One problematic thing with mbedtls is that it does not provide good information if something goes wrong. For example the max content len really depends on data that is being sent and received. I just wonder why mbedtls is not able to negotiate the buffer size so that both parties would be able to pass data properly.

jukkar, 2021-06-02:
> On the other hand I think it would be good to provide a https client sample that speaks to a server in the internet

That is a good idea, would you be able to provide one or extend the existing samples to do this?

rlubos, 2021-06-02:
Just a small reminder that we already have `http_get` and `big_http_download` samples - both connect to the http server in the internet and both run either http or https. Not sure if we want to maintain yet another sample.

mglettig, 2021-06-08:
You are right @rlubos. I think those samples cover my use case. Sorry for missing them. Therefore I think I will close this issue without any new samples.

closed 2021-06-08 by mglettig