zephyrproject-rtos/zephyr #35539: tests: drivers: spi: spi_loopback: test failed since #34731 is merged

ABOSTM, 2021-05-21:
**Describe the bug**
Since merge of #34731 `drivers/spi: STM32: This solves SPI infinite loop on Transceive`
 tests/drivers/spi/spi_loopback/ is failed on 
* nucleo_f103rb,
* nucleo_f746zg
* stm32f3_disco 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_f746zg tests/drivers/spi/spi_loopback/
2. west flash
3. See error

**Expected behavior**
test passed

**Logs and console output**
```
*** Booting Zephyr OS build v2.6.0-rc1-284-gc00504e875e2  ***
Running test suite test_spi
===================================================================
START - test_spi_loopback
I: SPI test on buffers TX/RX 0x20010131/0x20010a1a
I: SPI test slow config
I: Start complete multiple
I: CS control inhibited (no GPIO device)
I: Passed
I: Start complete loop
I: Passed
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Start half end
E: Buffer contents are different: 0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,
E:                            vs: 0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:382: spi_rx_half_end: (1 is true)
Buffer contents are different
 FAIL - test_spi_loopback in 0.54 seconds
===================================================================
Test suite test_spi failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain Zephyr SDK
 - Commit SHA 5f61d89233

galak, 2021-05-21:
Wonder if this is effectively a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/35297

jfischer-no, 2021-05-21:
Similar on nucleo_f413zh, I came across this while testing an SPI device. On nucleo_f413zh it is a bit different, data in RX buffer is shifted up one octet, reverting 0129884062055156bc79fd8faafaf13e601438b4 fixes the issue.
```
*** Booting Zephyr OS build v2.6.0-rc1-299-gd999d5eb64f4  ***
Running test suite test_spi
===================================================================
START - test_spi_loopback
I: SPI test on buffers TX/RX 0x20000131/0x2000098e
I: SPI test slow config
I: Start complete multiple
I: CS control inhibited (no GPIO device)
I: Passed
I: Start complete loop
I: Passed
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Start half end
E: Buffer contents are different: 0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,
E:                            vs: 0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,
```

ddkn, 2021-05-24:
I have a similar issue with the nucleo_f767zi, as seen here #35561, where @jfischer-no pointed it out.

ddkn, 2021-05-25:
Does anyone know of a commit state where the nucleo-f103rb or nucleo-f767zi was working for SPI? I have both boards to test with, as well as the nucleo-L152.

ABOSTM, 2021-05-26:
@ddkn 
you can either go back to sha1: 13303c4351d9b88f0c522cf5b5a4835a42cb6403
or simply revert  #34731 (aka sha1 0129884062055156bc79fd8faafaf13e601438b4)
I hope it will fix your SPI issue.

ddkn, 2021-05-26:
Hi @ABOSTM,

Thanks for the response. I had tried the revert earlier, which had the buffer issue I mentioned. Where 013303c4 gives the same error message, but then @FRASTM in #35561 pointed out I need to connect D11 -> D12 for the loopback to work correctly (I assumed it was doing something internal). It works, now.

```
*** Booting Zephyr OS build v2.6.0-rc1-51-g13303c4351d9  ***
Running test suite test_spi
===================================================================
START - test_spi_loopback
I: SPI test on buffers TX/RX 0x20020131/0x20020a1a
I: SPI test slow config
I: Start complete multiple
I: CS control inhibited (no GPIO device)
I: Passed
I: Start complete loop
I: Passed
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Start half end
I: Passed
I: Start every 4
I: Passed
I: Start async call
I: Passed
I: SPI test fast config
I: Start complete multiple
I: CS control inhibited (no GPIO device)
I: Passed
I: Start complete loop
I: Passed
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Start half end
I: Passed
I: Start every 4
I: Passed
I: Start async call
I: Passed
I: Start complete loop
I: CS control inhibited (no GPIO device)
I: Passed
I: Start complete loop
I: CS control inhibited (no GPIO device)
I: Passed
I: All tx/rx passed
 PASS - test_spi_loopback in 0.75 seconds
===================================================================
Test suite test_spi succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

galak, 2021-05-26:
Fixed by #35661

closed 2021-05-26 by galak