zephyrproject-rtos/zephyr #35541: sockets_tls: when using dtls with sara-r4 modem, handshake hangs if no reply

emillindq, 2021-05-21:
**Describe the bug**
The scenario is as follows: user issues a sendto on a dtls socket with offloaded socket behind it on the SARA-R4 modem. This will cause a handshake to be issued where I am the dtls client. Now mbedTLS ends up calling `dtls_rx` in sockets_tls.c. As described, mbedTLS wants a recvfrom with timeout, which is done using poll() if `is_block` is true. For reasons below, `is_block` ends up false but recvfrom is blocking -> hangs forever if dtls server doesn't reply on first client hello.

I've tried to see why `is_blocking` becomes false and it turns out that fnctl is called on the underlying socket to ask for fd flags, but  `z_impl_zsock_fcntl` ends up calling `ioctl` for which F_GETFD is totally nonsense argument. `ioctl` return value is minus on error, which then sockets_tls interprets as flags since it expects fnctl to return the flags. For example SARA-R4's `ioctl` has no idea about `F_GETFL` and returns -1, which then sockets_tls interprets as flags (basically all flags set, including the nonblock flag). This leads to `is_blocking==false`

Then sockets_tls sees "oh we are not blocking" and calls recvfrom which is blocking -> hangs indefinitely.

I'd love to get @rlubos feedback on this. In particular, I wonder if it wouldn't be possible to just skip the entire check-if-blocking thing and go ahead with `k_poll` in both cases? If there is data waiting (aka. `is_block==false` in my interpretation), it will return straight away and then we will continue to call recvfrom. I tried it out and it works perfectly in my scenario.

**To Reproduce**
Has been described above

**Expected behavior**
`dtls_rx` should respect timeout value and not block indefinitely when dtls server doesn't reply to client hello.

**Impact**
Showstopper, if server misses a beat.

**Environment (please complete the following information):**
 - Zephyr: e6248db

rlubos, 2021-05-25:
> it turns out that fnctl is called on the underlying socket to ask for fd flags, but `z_impl_zsock_fcntl` ends up calling `ioctl` for which F_GETFD is totally nonsense argument.

This is not correct, Zephyr made the following assumption some time ago (comment from the sockets.c codebase):
```
/* In Zephyr, fcntl() is just an alias of ioctl(). */
```
Also if you look in the request codes list supported by Zephyr you'll see the following comment:
```
/* Codes below 0x100 are reserved for fcntl() codes. */
```
I agree it might not be that obvious on the first sight, but it's cleary the sara-r4 sockets implementation that miss the (basic) feature.

> In particular, I wonder if it wouldn't be possible to just skip the entire check-if-blocking thing and go ahead with k_poll in both cases? If there is data waiting (aka. is_block==false in my [...]

`is_block` does not indicate whether the data is waiting or not - it indicates the current mode (blocking or non-blocking), which is defined either by `O_NONBLOCK` socket flag (set permamently on a socket) or the `MSG_DONTWAIT` flag, provided in the recv()/recvfrom() call. We cannot skip this check, because that would mean we'll always end up in the `poll()` with the timeout configured by the mbedTLS. This means that we'll break the non-blocking socket behavior, as if there's no datagram available on a socket, we'll block in the `poll()`. That's why it should only be called if the socket is configured used in a blocking manner.

emillindq, 2021-05-25:
Thanks for your input!

I will create a PR for SARA-R4 driver to handle this case.

emillindq, 2021-05-26:
@rlubos I looked into this a bit more and it seems like  `fcntl` can return -1 upon error. See for example https://pubs.opengroup.org/onlinepubs/009604599/functions/fcntl.html
> On successful completion: ... F_GETFL: ... The return value is not negative
Otherwise, -1 shall be returned

I interpret this as if F_GETFL is successful, not negative. If fail (for example not implemented as in the case of SARA-R4), -1. In my humble opinion the correct way to handle this is by checking the return value from fcntl before interpreting the answer as flags in sockets_tls.c

rlubos, 2021-05-26:
> I interpret this as if F_GETFL is successful, not negative. If fail (for example not implemented as in the case of SARA-R4), -1. In my humble opinion the correct way to handle this is by checking the return value from fcntl before interpreting the answer as flags in sockets_tls.c

Ok, I agree we should check the error code instead of blindly assuming the success, this could be fixed. But what does it change in the context of SARA-R4? If `fcntl()` returns an error then the whole function should return early, indicating the error. The TLS layer should not make any assumptions on the underlying socket behaviour.

emillindq, 2021-05-27:
Yes you are right, we should do both: check for error (so it is obvious the underlying socket doesn't implementing `F_GETFL` for future potential problems) and return 0 in sara-r4 on said argument.

Please see https://github.com/zephyrproject-rtos/zephyr/pull/35712 for my proposal

PR #35712 (2021-05-27): Fix hanging dtls handshake if no reply
Fixes #35541
commit 91177eeb (2021-05-27): net: sockets: tls: check return code from fcntl

Not checking return code in fcntl can result in interpreting -1 as
flags, and cause unexpected behaviour.

Fixes #35541
closed 2021-05-27 by carlescufi