zephyrproject-rtos/zephyr #35550: nRF91: DPS310 I2C driver not working

rajeev1986, 2021-05-21:
**Describe the bug**
I have integrated a DPS310 sensor from Infineon to nRF9160 via I2C and trying to get the example DSP310 sample application provided in the samples to work. However, after checking and rechecking all the configurations, I keep getting

`Could not get DPS310 device`

which simply tells me that the application is failing to get the device binding.

```
const struct device *dev = device_get_binding(DT_LABEL(DT_INST(0, infineon_dps310)));

if (dev == NULL) {
	printk("Could not get DPS310 device\n");
	return;
}
```

To diagnose this issue, I have enabled the logs at the I2C and the DPS310 driver level as follows
```
CONFIG_LOG=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y
CONFIG_I2C_LOG_LEVEL_DBG=y
```

and found the following:

```
/* Initialize the sensor and apply the configuration */
static int dps310_init(const struct device *dev)
{
	struct dps310_data *data = dev->data;
	const struct dps310_cfg *config = dev->config;

	data->i2c_master = device_get_binding(config->i2c_bus_name);
	if (data->i2c_master == NULL) {
		LOG_ERR("Failed to get I2C device");
		return -EINVAL;
	}

	uint8_t product_id = 0;
	int res = i2c_reg_read_byte(data->i2c_master, config->i2c_addr,
				 IFX_DPS310_REG_ADDR_PRODUCT_ID, &product_id);

	if (res < 0) {
		LOG_ERR("No device found");
		return -EINVAL;
	}

	if (product_id != IFX_DPS310_PRODUCT_ID) {
		LOG_ERR("Device is not a DPS310");
		return -EINVAL;
	}
	else
	{
		LOG_DBG("DPS310 product_id: 0x%02x", product_id);
	}
	
	LOG_DBG("Init DPS310");
	/* give the sensor time to load the calibration data */
	k_sleep(K_MSEC(40));

	/* wait for the sensor to load the calibration data */
	if (!poll_rdy(data, config, REG_ADDR_MEAS_CFG,
		      IFX_DPS310_REG_ADDR_MEAS_CFG_SELF_INIT_OK)) {
		LOG_DBG("Sensor not ready");
		return -EIO;
	}

	/* read calibration coefficients */
	uint8_t raw_coef[18] = { 0 };

	res = i2c_write_read(data->i2c_master, config->i2c_addr,
			     &REG_ADDR_CALIB_COEFF_0, 1, &raw_coef, 18);
	if (res < 0) {
		LOG_WRN("I2C error: %d", res);
		return -EIO;
	}

	/* convert calibration coefficients */
	dps310_calib_coeff_creation(raw_coef, &data->comp);

	/*
	 * check which temperature sensor was used for calibration and use it
	 * for measurements.
	 */
	uint8_t tmp_coef_srce = 0;

	res = i2c_write_read(data->i2c_master, config->i2c_addr,
			     &REG_ADDR_COEF_SRCE, 1, &tmp_coef_srce, 18);
	if (res < 0) {
		LOG_WRN("I2C error: %d", res);
		return -EIO;
	}

	/* clear all other bits */
	tmp_coef_srce &= IFX_DPS310_COEF_SRCE_MASK;

	/* merge with temperature measurement configuration */
	tmp_coef_srce |= IFX_DPS310_TMP_CFG;

	/* set complete configuration in one write */
	const uint8_t config_seq[] = {
		IFX_DPS310_REG_ADDR_PRS_CFG,	/* start register address */
		IFX_DPS310_PSR_CFG,		/* PSR_CFG */
		tmp_coef_srce,			/* TMP_CFG */
		0x00,				/* MEAS_CFG */
		DPS310_CFG_REG			/* CFG_REG */
	};

	res = i2c_write(data->i2c_master, config_seq, sizeof(config_seq),
			config->i2c_addr);
	if (res < 0) {
		LOG_WRN("I2C error: %d", res);
		return -EIO;
	}

	dps310_hw_bug_fix(data, config);
	dps310_measure_tmp_psr(data, config);

	LOG_DBG("Init OK");
	return 0;
}
```
In the DPS310 driver `dps310_init()` function;
The code runs as expected up to reading the product ID and then the  `i2c_write_read()` call for reading calibration coefficients fails and gives a negative value for input/output error failing to initialize the driver properly.
```
I2C error: -5 
```
What is surprising that the I2C call for reading the product ID from the sensor register works and it gives the correct product id as `0x10h` which indicates that connections and i2c address are correct.

Looking deeper into the code, the ` i2c_reg_read_byte()` actually returns ` i2c_write_read()`. However, directly calling the ` i2c_write_read()` in the DPS310 driver layer is not working as expected. So why does one I2C API call works while the other does not? 

```
static inline int i2c_reg_read_byte(const struct device *dev,
				    uint16_t dev_addr,
				    uint8_t reg_addr, uint8_t *value)
{
	return i2c_write_read(dev, dev_addr,
			      &reg_addr, sizeof(reg_addr),
			      value, sizeof(*value));
}
```

**To Reproduce**
Add the following config to the overlay file:

```
/*
 * A DPS310 attached to the I2C pins: SDA:26, SCL:27
 */

&i2c1 {
	compatible = "nordic,nrf-twim";
	sda-pin = < 0x1a >;
	scl-pin = < 0x1b >;	
	status = "okay";
	DPS310@77 {
		status = "okay";
		compatible = "infineon,dps310";
		reg = <0x77>;
		label = "DPS310";
	};
};
```
and the following to the prj.conf file

```
CONFIG_I2C=y
CONFIG_SENSOR=y
CONFIG_DPS310=y
CONFIG_LOG=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y
CONFIG_I2C_LOG_LEVEL_DBG=y
```

Then build and flash the application as follows:

```
cd samples/sensor/dps310
west build -b nrf9160dk_nrf9160 --pristine
west flash
```

**Expected behavior**
On application startup, log message says:

```
Could not get DPS310 device
```

**Impact**
Simply annoying and unreliable as now I am unsure that which of these sensor drivers might be working or broken unless each one is retested. Also, one might have to rewrite the specific sensor driver to make it work for their specific platform which is not what one might expect. 

**Environment:**
 - MacOS
 - nRF Connect SDK + Zephyr
 - v1.5.1

**Additional context**
After reading other threads, this might the ongoing and lingering i2c issue in Zephyr as mentioned [here](https://github.com/zephyrproject-rtos/zephyr/issues/33440) and [here](https://github.com/zephyrproject-rtos/zephyr/issues/27675).

anangl, 2021-05-25:
> In the DPS310 driver `dps310_init()` function;
> The code runs as expected up to reading the product ID and then the `i2c_write_read()` call for reading calibration coefficients fails and gives a [...]

I think the problem is that the TWIM peripheral that handles I2C in nRF9160 cannot perform a transaction when the provided buffer is located in flash, and that is the case when `i2c_write_read()` is called from the DPS310 driver (`REG_ADDR_CALIB_COEFF_0` is defined as `static const`). When `i2c_write_read()` is called via `i2c_reg_read_byte()`, the buffer to be sent is `&reg_addr`, so it will be on the stack, i.e. in RAM.
You can confirm this by trying to use the following in the driver:
```
	uint8_t raw_coef[18] = { 0 };
	uint8_t reg_addr = REG_ADDR_CALIB_COEFF_0;

	res = i2c_write_read(data->i2c_master, config->i2c_addr,
			     &reg_addr, 1, &raw_coef, 18);
```
But that's obviously not a proper solution of this problem. The i2c_nrfx_twim driver needs to be improved to handle also transfers from flash (use some internal buffer and copy there the provided data from flash to overcome the limitation of the TWIM peripheral).

rajeev1986, 2021-05-26:
@anangl,
Thanks for the suggestion. So after making similar changes (or should I say hack) across the DPS310 driver code, basically, getting rid of `static const uint8_t` especially for the register addresses and the buffer to be written, I have managed the driver to work and fetch the sensor values.
Now all the` i2c_write_read()` calls within this driver are working as expected without error values.
For now, I will make the changes to the DPS310 driver and use it as a local driver rather than directly fetching it from Zephyr's driver library.
Thanks and I hope in the future release this issue will be hopefully fixed for all the related drivers or the core i2c driver itself.

carlescufi, 2021-05-26:
@anangl do you want to open an enhancement and close this?

anangl, 2021-05-26:
> @anangl do you want to open an enhancement and close this?

@carlescufi I think it is fine to keep this issue opened as a bug. This missing feature of the i2c_nrfx_twim driver actually causes the DPS310 driver to not work on nRF9160, so that's something that needs to be fixed.