zephyrproject-rtos/zephyr #35552: mbedtls: Update for better PSA API support

microbuilder, 2021-05-21:
Recent versions of mbedtls have started migrating to use Platform Security Architecture (PSA) APIs beneath the hood. This is still a work in progress, and the PSA API coverage isn't very exhaustive as of 2.26.0, and there are conflicts with the PSA APIs in mbedtls 2.26.0 that prevent it being use with TF-M (https://github.com/zephyrproject-rtos/zephyr/issues/35305).

The next release(s) of mbedtls, which may be an LTS release, should improve this situation, and will also likely occur during the Zephyr LTS2 dev cycle.

Fixing PSA API support in Zephyr 2.6.0 with mbedtls 2.26.0 isn't feasible with the current state of the codebase and build system, but the mbedtls build system integration can be improved in the 2.7.0 cycle.

## Required Changes

- During the 2.7.0 cycle, the PSA config options from mbedtls should be exposed via Kconfig similar to how the TLS config options are today, and proper integration with TF-M should be tested and co-ordinated with the upstream mbedtls project.
- mbedtls should be updated to the latest version (2.27.0, 3.0.0???), which should have improved PSA API support.

## Related Issues

Closes https://github.com/zephyrproject-rtos/zephyr/issues/35305 as something that can't reasonably be fixed with the 2.6.0 release, due to limitations of the upstream mbedtls project and current build system.

A release note for 2.26.0 has been added indicating that PSA APIs and TF-M can not be used together: https://github.com/zephyrproject-rtos/zephyr/pull/35513

microbuilder, 2021-07-07:
Note: mbedtls 3.0.0 has been officially released: https://github.com/ARMmbed/mbedtls/releases

cfriedt, 2021-07-12:
Marking this as high-priority, as aligning on next major release of mbedTLS would be desireable.

SebastianBoe, 2021-08-24:
PR for updating mbedtls to 3.0.0:
https://github.com/zephyrproject-rtos/zephyr/pull/37753

microbuilder, 2021-11-04:
Addressed with MbedTLS 3.0 support which brings improved PSA integration.

closed 2021-11-04 by microbuilder