zephyrproject-rtos/zephyr #35567: some mwdt compiler options can't be recognized by zephyr_cc_option

IRISZZW, 2021-05-24:
**Describe the bug**
some mwdt compiler options (`-Xll64` `-Xmpy_option=qmpyh` `-Xfpud_div`) can't be recognized by zephyr_cc_option.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b hsdk -d build.mwdt tests/lib/sprintf/
2. See error

**Logs and console output**
```
 west build -b hsdk -d build.mwdt tests/lib/sprintf/-- west build: generating a build systemIncluding boilerplate (Zephyr base): /SCRATCH2/Watson/zephyrproject/zephyr/cmake/app/boilerplate.cmake-- Application: /SCRATCH2/Watson/zephyrproject/zephyr/tests/lib/sprintf-- Zephyr version: 2.6.0-rc1 (/SCRATCH2/Watson/zephyrproject/zephyr), build: v2.6.0-rc1-320-g374629af906a-- Found Python3: /depot/Python/Python-3.7.0/bin/python3.7 (found suitable exact version "3.7.0") found components: Interpreter -- Found west (found suitable version "0.10.1", minimum required is "0.7.1")-- Board: hsdk
-- Cache files will be written to: /u/zhiwei/.cache/zephyr
-- Using toolchain: zephyr 0.12.3 (/SCRATCH2/svc-arcoss_auto/tools/zephyr-sdk/0.12.3)
-- Found dtc: /SCRATCH2/svc-arcoss_auto/tools/zephyr-sdk/0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /SCRATCH2/Watson/zephyrproject/zephyr/boards/arc/hsdk/hsdk.dts
-- Generated zephyr.dts: /SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt/zephyr/include/generated/device_extern.h
Parsing /SCRATCH2/Watson/zephyrproject/zephyr/Kconfig
Loaded configuration '/SCRATCH2/Watson/zephyrproject/zephyr/boards/arc/hsdk/hsdk_defconfig'
Merged configuration '/SCRATCH2/Watson/zephyrproject/zephyr/tests/lib/sprintf/prj.conf'
Configuration saved to '/SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt/zephyr/.config'
Kconfig header saved to '/SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt/zephyr/include/generated/autoconf.h'
-- The C compiler identification is Clang 10.0.1
-- The CXX compiler identification is Clang 10.0.1
-- The ASM compiler identification is unknown
-- Found assembler: /global/apps/mwdt_2021.03/MetaWare/arc/bin/ccac
-- Warning: Did not find file Compiler/-ASM
CMake Warning at ../../../cmake/linker/arcmwdt/target.cmake:137 (message):
  MWDT toolchain does not support

             CONFIG_LINKER_ORPHAN_SECTION_WARN
Call Stack (most recent call first):
  ../../../CMakeLists.txt:246 (toolchain_ld_baremetal)

-- Configuring done
-- Generating done
-- Build files have been written to: /SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt
-- west build: building application
[1/134] Preparing syscall dependency handling

[53/134] Building ASM object zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/switch.S.obj
FAILED: zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/switch.S.obj 
/global/apps/mwdt_2021.03/MetaWare/arc/bin/ccac -DBUILD_VERSION=v2.6.0-rc1-320-g374629af906a -DKERNEL -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../kernel/include -I../arch/arc/include -I../include -Izephyr/include/generated -I../soc/arc/snps_arc_hsdk -I../subsys/testsuite/include -I../subsys/testsuite/ztest/include -I../lib/libc/minimal/include -I/global/apps/mwdt_2021.03/MetaWare/arc/../../MetaWare/arc/inc -Os -imacros /SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt/zephyr/include/generated/autoconf.h -Hnocrt -fno-common -g -Hnocopyr -HL -Hnosdata -Hoff=Stackcheck_alloca -Hasmcpp -imacros /SCRATCH2/Watson/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -D_ASMLANGUAGE -Wno-sometimes-uninitialized -Wno-shift-overflow -Wno-missing-braces -Wno-self-assign -Wno-address-of-packed-member -Wno-unused-function -Wno-initializer-overrides -Wno-section -Wno-unknown-warning-option -Wno-unused-variable -Wno-format-invalid-specifier -Wno-gnu -Wno-tautological-compare -ffunction-sections -fdata-sections -arcv2hs -core2 -Xatomic -Xunaligned -Xcode_density -Xdiv_rem=radix4 -Xswap -Xbitscan -Xshift_assist -Xbarrel_shifter -Xtimer0 -Xtimer1 -Xrtc -Xfpu_mac -Hno_default_include -Hnoarcexlib -o zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/switch.S.obj -c /SCRATCH2/Watson/zephyrproject/zephyr/arch/arc/core/switch.S
[/global/apps/mwdt_2021.03/MetaWare/arc/bin/asac] "/SCRATCH2/Watson/zephyrproject/zephyr/arch/arc/core/switch.S",Line 96: error: This instruction cannot be encoded.
Memory or auxiliary register loads must have a register destination.
[/global/apps/mwdt_2021.03/MetaWare/arc/bin/asac] "/SCRATCH2/Watson/zephyrproject/zephyr/arch/arc/core/switch.S",Line 96: error: This instruction cannot be encoded.
Memory or auxiliary register loads must have a register destination.
[/global/apps/mwdt_2021.03/MetaWare/arc/bin/asac] 2 errors generated.
[56/134] Building ASM object zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/regular_irq.S.obj
FAILED: zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/regular_irq.S.obj 
/global/apps/mwdt_2021.03/MetaWare/arc/bin/ccac -DBUILD_VERSION=v2.6.0-rc1-320-g374629af906a -DKERNEL -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../kernel/include -I../arch/arc/include -I../include -Izephyr/include/generated -I../soc/arc/snps_arc_hsdk -I../subsys/testsuite/include -I../subsys/testsuite/ztest/include -I../lib/libc/minimal/include -I/global/apps/mwdt_2021.03/MetaWare/arc/../../MetaWare/arc/inc -Os -imacros /SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt/zephyr/include/generated/autoconf.h -Hnocrt -fno-common -g -Hnocopyr -HL -Hnosdata -Hoff=Stackcheck_alloca -Hasmcpp -imacros /SCRATCH2/Watson/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -D_ASMLANGUAGE -Wno-sometimes-uninitialized -Wno-shift-overflow -Wno-missing-braces -Wno-self-assign -Wno-address-of-packed-member -Wno-unused-function -Wno-initializer-overrides -Wno-section -Wno-unknown-warning-option -Wno-unused-variable -Wno-format-invalid-specifier -Wno-gnu -Wno-tautological-compare -ffunction-sections -fdata-sections -arcv2hs -core2 -Xatomic -Xunaligned -Xcode_density -Xdiv_rem=radix4 -Xswap -Xbitscan -Xshift_assist -Xbarrel_shifter -Xtimer0 -Xtimer1 -Xrtc -Xfpu_mac -Hno_default_include -Hnoarcexlib -o zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/regular_irq.S.obj -c /SCRATCH2/Watson/zephyrproject/zephyr/arch/arc/core/regular_irq.S
[/global/apps/mwdt_2021.03/MetaWare/arc/bin/asac] "/SCRATCH2/Watson/zephyrproject/zephyr/arch/arc/core/regular_irq.S",Line 277: error: This instruction cannot be encoded.
Memory or auxiliary register loads must have a register destination.
[/global/apps/mwdt_2021.03/MetaWare/arc/bin/asac] "/SCRATCH2/Watson/zephyrproject/zephyr/arch/arc/core/regular_irq.S",Line 277: error: This instruction cannot be encoded.
Memory or auxiliary register loads must have a register destination.
[/global/apps/mwdt_2021.03/MetaWare/arc/bin/asac] 2 errors generated.
[57/134] Building ASM object zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/fast_irq.S.obj
FAILED: zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/fast_irq.S.obj 
/global/apps/mwdt_2021.03/MetaWare/arc/bin/ccac -DBUILD_VERSION=v2.6.0-rc1-320-g374629af906a -DKERNEL -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../kernel/include -I../arch/arc/include -I../include -Izephyr/include/generated -I../soc/arc/snps_arc_hsdk -I../subsys/testsuite/include -I../subsys/testsuite/ztest/include -I../lib/libc/minimal/include -I/global/apps/mwdt_2021.03/MetaWare/arc/../../MetaWare/arc/inc -Os -imacros /SCRATCH2/Watson/zephyrproject/zephyr/build.mwdt/zephyr/include/generated/autoconf.h -Hnocrt -fno-common -g -Hnocopyr -HL -Hnosdata -Hoff=Stackcheck_alloca -Hasmcpp -imacros /SCRATCH2/Watson/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -D_ASMLANGUAGE -Wno-sometimes-uninitialized -Wno-shift-overflow -Wno-missing-braces -Wno-self-assign -Wno-address-of-packed-member -Wno-unused-function -Wno-initializer-overrides -Wno-section -Wno-unknown-warning-option -Wno-unused-variable -Wno-format-invalid-specifier -Wno-gnu -Wno-tautological-compare -ffunction-sections -fdata-sections -arcv2hs -core2 -Xatomic -Xunaligned -Xcode_density -Xdiv_rem=radix4 -Xswap -Xbitscan -Xshift_assist -Xbarrel_shifter -Xtimer0 -Xtimer1 -Xrtc -Xfpu_mac -Hno_default_include -Hnoarcexlib -o zephyr/arch/arch/arc/core/CMakeFiles/arch__arc__core.dir/fast_irq.S.obj -c /SCRATCH2/Watson/zephyrproject/zephyr/arch/arc/core/fast_irq.S
[... cut ...] is supported, then cmake will use these options as compile options.
https://github.com/zephyrproject-rtos/zephyr/blob/374629af906a24add294b1c3b945128cb6a486e8/soc/arc/snps_arc_hsdk/CMakeLists.txt#L13-L19

Then if you check the compile options in the error log:
```
-arcv2hs -core2 -Xatomic -Xunaligned -Xcode_density
-Xdiv_rem=radix4 -Xswap -Xbitscan
-Xshift_assist -Xbarrel_shifter -Xtimer0 -Xtimer1 -Xrtc
-Xfpu_mac
```
you will find we miss these options: `-Xll64` `-Xmpy_option=qmpyh` `-Xfpud_div`

abrodkin, 2021-05-24:
@IRISZZW it would be interesting to figure out why these particular options don't work nice with `check_c_compiler_flag`. My guess it they are not "self-contained", i.e. first require "top-level" option like CPU family (AKA `-arcv2hs` or `-arcv2em`) to exist. Though these are just y guesses which would be good to prove or dismiss.

IRISZZW, 2021-05-24:
> @IRISZZW it would be interesting to figure out why these particular options don't work nice with `check_c_compiler_flag`. My guess it they are not "self-contained", i.e. first require "top-level" [...]

you are right, they are not "self-contained":
```
$ cat main.c 
int main(void)
{
        return 0;
}

$ ccac -Xll64 main.c 
[ccac] Argument `-Xll64' is invalid: This feature only applies to AV2HS and AV3HS
Aborting...

$ ccac -Xfpud_div main.c
[ccac] Argument `-Xfpud_div' is invalid: This feature only applies to AV2HS
Aborting...

$ ccac -Xmpy_option=qmpyh  main.c 
[ccac] Argument `-Xmpy_option=qmpyh' is invalid: The "MPY OPTION" value for the ARC EM processor is limited to 1 through 8 and w,wlh1,wlh2,wlh3,wlh4,wlh5,mpy16,mpy,mac,mpyd
Aborting...
```

abrodkin, 2021-05-24:
@evgeniy-paltsev @tejlmand any hints on how can we test options which have dependencies?
I.e. if option `-option1` is only being recognized when `-option2` also passed to the compiler.

evgeniy-paltsev, 2021-05-24:
Actually I don't see any purpose of this options test - given that we don't provide any kind of alternate options if the original options are not supported by compiler. So in my point of view the replacement of `zephyr_cc_option` by `zephyr_compile_options` looks valid.

closed 2021-05-25 by galak