zephyrproject-rtos/zephyr #35596: [BLE] Boards cannot connect if extended advertising is enabled in `prj.conf`

chrisswinchatt-arm, 2021-05-25:
**Describe the bug**

A Zephyr program compiled with the following in `prj.conf`:

```
CONFIG_BT=y
CONFIG_BT_CENTRAL=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_EXT_ADV=y
CONFIG_BT_PER_ADV=y
```

Calling `bt_conn_le_create` after discovering a peer returns -5 (`-EIO`) when compiled with the above config. Removing or setting to `n` the `CONFIG_BT_EXT_ADV` and `CONFIG_BT_PER_ADV` options and recompiling corrects the behaviour.

I noticed different results with different combinations of `CONFIG_BT_EXT_ADV` and `CONFIG_BT_PER_ADV`:

CONFIG_BT_EXT_ADV | CONFIG_BT_PER_ADV | Return value of `bt_conn_le_create`
:---:|:---:|:---:
y | y | -EIO
y | n | -EIO
n | y | 2
n | n | 0

**To Reproduce**
Steps to reproduce the behavior:
1. Compile the `central` and `peripheral` samples using the above config
2. Flash them to boards; I tested with `nucleo_wb55rg`, `disco_l475_iot1` and `nrf52840_mdk` 
3. Try to establish a connection
4. See error

**Expected behavior**
It should be possible for a single program to support both connect & periodic sync.

**Impact**
Showstopper

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) Windows
 - Toolchain (e.g Zephyr SDK, ...): Zephyr / West
 - Commit SHA or Version used: 9a14f627426ba1e25e75d8ea3ebbdcccc4d9b2a7

**Additional context**
N/A

joerchan, 2021-05-25:
Support for extended advertising needs to be enabled in the zephyr controller.
`CONFIG_BT_CTLR_ADV_EXT=y`

carlescufi, 2021-05-25:
@chrisswinchatt-arm if you objective is to come up with a [power consumption comparison](https://github.com/chrisswinchatt-arm/bluetooth-power-consumption-benchmark/tree/feature-zephyr) between different RTOS, may I point out that you need to take into account not only Bluetooth, but also things like other peripherals (the UART in particular) as well as the behavior of the application itself.

joerchan, 2021-05-26:
Did some digging because it was weird that only creating connections was affected, but advertising and scanning worked fine.

Turns out the feature bit checking was inconsistent, I've created a fix for it.

chrisswinchatt-arm, 2021-05-26:
Thanks for speedy response @all.

PR #35670 (2021-05-26): Bluetooth: host: Fix extended advertising feature bit check
Fix CONFIG_BT_EXT_ADV_LEGACY_SUPPORT option which optimizes the host
to skip checking the feature bit of the controller to check for
extended advertising commands.
This was broken because of how this was implented using an undef of the
feature bit, which was not replicated in scan.c, adv.c and id.c once
this was split out from hci_core.

Instead of doing this wierd way of redefining the feature check macro
do it in a proper way by defining a new macro.

Fixes: #35596
closed 2021-05-26 by carlescufi