zephyrproject-rtos/zephyr #35633: Out of bound read: Multiple Coverity sightings in generated code

nashif, 2021-05-25:
```
1#ifdef CONFIG_USERSPACE
  	1. Condition z_syscall_trap(), taking true branch.
42        if (z_syscall_trap()) {
  	CID 219506: Out-of-bounds read (OVERRUN) [select issue]
  	
CID 236005 (#1 of 1): Out-of-bounds read (OVERRUN)
2. overrun-local: Overrunning array of 1 bytes at byte offset 3 by dereferencing pointer (uintptr_t *)&bypass.
43                return (bool) arch_syscall_invoke1(*(uintptr_t *)&bypass, K_SYSCALL_LOG_PROCESS);
44        }
45#endif
```

```

67#ifdef CONFIG_USERSPACE
  	1. Condition z_syscall_trap(), taking true branch.
68        if (z_syscall_trap()) {
  	CID 232755: Out-of-bounds read (OVERRUN) [select issue]
  	
CID 235951 (#1 of 1): Out-of-bounds read (OVERRUN)
2. overrun-local: Overrunning array of 2 bytes at byte offset 3 by dereferencing pointer (uintptr_t *)&source_id.
69                return (uint32_t) arch_syscall_invoke4(*(uintptr_t *)&backend, *(uintptr_t *)&domain_id, *(uintptr_t *)&source_id, *(uintptr_t *)&level, K_SYSCALL_LOG_FILTER_SET);
70        }
71#endif
```

All issues:

CID | Impact | Type | Status | First Detected | Owner | Classification | Severity | Action | Component | Category | Function | File | Line Number | Standard: CERT C
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
236005 | High | Out-of-bounds read | New | 05/10/2021 | Unassigned | Unclassified | Unspecified | Undecided | Other | Memory - illegal accesses | log_process | /twister-out/qemu_x86/tests/subsys/logging/log_api/logging.log2_api_immediate_64b_timestamp/zephyr/include/generated/syscalls/log_ctrl.h | 43 | None
235951 | High | Out-of-bounds read | New | 05/10/2021 | Unassigned | Unclassified | Unspecified | Undecided | Other | Memory - illegal accesses | log_filter_set | /twister-out/qemu_x86/tests/subsys/logging/log_api/logging.log2_api_immediate_64b_timestamp/zephyr/include/generated/syscalls/log_ctrl.h | 69 | None
235933 | High | Out-of-bounds read | New | 05/10/2021 | Unassigned | Unclassified | Unspecified | Undecided | Drivers | Memory - illegal accesses | gpio_pin_configure | /twister-out/nucleo_l4r5zi/tests/drivers/build_all/drivers.modem.build/zephyr/include/generated/syscalls/gpio.h | Various | None
235923 | High | Out-of-bounds read | New | 05/10/2021 | Unassigned | Unclassified | Unspecified | Undecided | Other | Memory - illegal accesses | z_log_msg2_runtime_vcreate | /twister-out/qemu_x86/tests/subsys/logging/log_api/logging.log2_api_immediate_64b_timestamp/zephyr/include/generated/syscalls/log_msg2.h | 47 | None
235917 | High | Out-of-bounds read | New | 05/10/2021 | Unassigned | Unclassified | Unspecified | Undecided | Other | Memory - illegal accesses | z_log_msg2_runtime_vcreate | /twister-out/qemu_x86/tests/subsys/logging/log_api/logging.log2_api_immediate_64b_timestamp/zephyr/include/generated/syscalls/log_msg2.h | 47 | None
232755 | High | Out-of-bounds read | New | 05/04/2021 | Unassigned | Unclassified | Unspecified | Undecided | Other | Memory - illegal accesses | log_filter_set | /twister-out/qemu_x86/tests/subsys/logging/log_core_additional/logging.add.user/zephyr/include/generated/syscalls/log_ctrl.h | Various | None
230223 | High | Out-of-bounds read | New | 05/04/2021 | Unassigned | Unclassified | Unspecified | Undecided | Various | Memory - illegal accesses | z_log_msg2_runtime_vcreate | /twister-out/qemu_x86/tests/subsys/logging/log_api/logging.log2_api_immediate_64b_timestamp/zephyr/include/generated/syscalls/log_msg2.h | Various | None
229325 | High | Out-of-bounds read | New | 05/04/2021 | Unassigned | Unclassified | Unspecified | Undecided | Various | Memory - illegal accesses | z_log_msg2_runtime_vcreate | /twister-out/qemu_x86/tests/subsys/logging/log_api/logging.log2_api_immediate_64b_timestamp/zephyr/include/generated/syscalls/log_msg2.h | Various | None
219725 | High | Out-of-bounds read | New | 03/13/21 | Unassigned | Unclassified | Unspecified | Undecided | Other | Memory - illegal accesses | k_busy_wait | /twister-out/ehl_crb/tests/subsys/edac/ibecc/edac.ibecc.injection.user/zephyr/include/generated/syscalls/kernel.h | 101 | None
219586 | High | Out-of-bounds read | New | 03/06/2021 | Unassigned | Unclassified | Minor | Ignore | Networking | Memory - illegal accesses | net_if_ipv4_addr_add_by_index | /twister-out/qemu_x86/tests/net/iface/net.iface/zephyr/include/generated/syscalls/net_if.h | 81 | None
219568 | High | Out-of-bounds read | New | 03/06/2021 | Unassigned | Unclassified | Minor | Ignore | Networking | Memory - illegal accesses | net_if_ipv6_addr_add_by_index | /twister-out/qemu_x86/tests/net/iface/net.iface/zephyr/include/generated/syscalls/net_if.h | 42 | None
219506 | High | Out-of-bounds read | New | 03/04/2021 | Unassigned | Unclassified | Minor | Ignore | Other | Memory - illegal accesses | log_process | /twister-out/qemu_x86/tests/subsys/logging/log_core_additional/logging.add.user/zephyr/include/generated/syscalls/log_ctrl.h | Various | None
219496 | High | Out-of-bounds read | New | 03/04/2021 | Unassigned | Unclassified | Minor | Ignore | Kernel | Memory - illegal accesses | ztest_set_fault_valid | /twister-out/qemu_x86/tests/kernel/threads/thread_error_case/kernel.threads.error.case/zephyr/include/generated/syscalls/ztest_error_hook.h | Various | None
219482 | High | Out-of-bounds read | New | 03/04/2021 | Unassigned | Unclassified | Minor | Ignore | Drivers | Memory - illegal accesses | pwm_pin_capture_cycles | /twister-out/frdm_k64f/tests/drivers/pwm/pwm_loopback/drivers.pwm.loopback/zephyr/include/generated/syscalls/pwm.h | 74 | None
219480 | High | Out-of-bounds read | New | 03/04/2021 | Unassigned | Unclassified | Unspecified | Undecided | Other | Memory - illegal accesses | cbvprintf | /lib/os/cbprintf_nano.c | Various | None
219477 | High | Out-of-bounds read | New | 03/04/2021 | Unassigned | Unclassified | Minor | Ignore | Drivers | Memory - illegal accesses | pwm_pin_set_cycles | /twister-out/frdm_k64f/tests/drivers/pwm/pwm_loopback/drivers.pwm.loopback/zephyr/include/generated/syscalls/pwm.h | 29 | None
215238 | High | Out-of-bounds read | New | 11/01/2020 | Unassigned | Unclassified | Minor | Ignore | Various | Memory - illegal accesses | net_addr_ntop | /twister-out/qemu_x86/tests/net/utils/net.util/zephyr/include/generated/syscalls/net_ip.h | Various | None
215223 | High | Out-of-bounds read | New | 11/01/2020 | Unassigned | Unclassified | Minor | Ignore | Various | Memory - illegal accesses | net_addr_pton | /twister-out/qemu_x86/tests/net/utils/net.util/zephyr/include/generated/syscalls/net_ip.h | Various | None
214226 | High | Out-of-bounds read | New | 09/08/2020 | Unassigned | Unclassified | Minor | Ignore | Various | Memory - illegal accesses | uart_poll_out | /twister-out/qemu_x86/samples/subsys/debug/gdbstub/sample.debug.gdbstub/zephyr/include/generated/syscalls/uart.h | Various | None
211042 | High | Out-of-bounds read | New | 06/20/20 | Unassigned | Unclassified | Minor | Ignore | Networking | Memory - illegal accesses | zsock_inet_pton | /twister-out/qemu_x86/tests/net/socket/udp/net.socket.udp/zephyr/include/generated/syscalls/socket.h | Various | None
210588 | High | Out-of-bounds read | New | 05/28/20 | Unassigned | Unclassified | Minor | Ignore | Drivers | Memory - illegal accesses | dac_write_value | /twister-out/frdm_k64f/tests/drivers/dac/dac_api/drivers.dac/zephyr/include/generated/syscalls/dac.h | 42 | None
208195 | High | Out-of-bounds read | New | 02/09/2020 | Unassigned | Unclassified | Minor | Ignore | Drivers | Memory - illegal accesses | gpio_pin_interrupt_configure | /twister-out/nucleo_l4r5zi/tests/drivers/build_all/drivers.modem.build/zephyr/include/generated/syscalls/gpio.h | Various | None
207976 | High | Out-of-bounds read | New | 02/01/2020 | Unassigned | Unclassified | Minor | Ignore | Drivers | Memory - illegal accesses | counter_set_channel_alarm | /twister-out/nucleo_l4r5zi/tests/drivers/counter/counter_basic_api/drivers.counter/zephyr/include/generated/syscalls/counter.h | Various | None
207968 | High | Out-of-bounds read | New | 02/01/2020 | Unassigned | Unclassified | Minor | Ignore | Drivers | Memory - illegal accesses | counter_cancel_channel_alarm | /twister-out/nucleo_l4r5zi/tests/drivers/counter/counter_basic_api/drivers.counter/zephyr/include/generated/syscalls/counter.h | Various | None
[... cut ...]
205610 | High | Out-of-bounds read | New | 11/08/2019 | Unassigned | Unclassified | Minor | Ignore | Kernel | Memory - illegal accesses | k_futex_wake | /twister-out/qemu_x86/tests/kernel/mem_protect/futex/kernel.futex/zephyr/include/generated/syscalls/kernel.h | 601 | None
199438 | High | Out-of-bounds read | Triaged | 06/26/19 | Unassigned | Bug | Unspecified | Fix Submitted | Drivers | Memory - illegal accesses | stm32_exti_enable | /drivers/interrupt_controller/intc_exti_stm32.c | 119 | None

dcpleung, 2021-05-25:
Ignored all as it is as designed for the generated syscall stubs.

closed 2021-05-25 by dcpleung

galak, 2021-05-26:
@dcpleung are we going to keep getting such coverity issues in the future?  Do we need the generator to surpress future warnings?  (@nordic-krch did this for logging at some point).

https://doclazy.wordpress.com/2011/07/14/coverity-suppressing-false-positives-with-cod/

dcpleung, 2021-05-26:
Coverity doesn't know how we do syscalls, so this will appear every time new syscalls are introduced (or even modified). One very undesirable way to avoid this is to force all function arguments to be pointers, which will probably make drivers/subsys APIs really awkward.

Let me look into the coverity tag.

dcpleung, 2021-05-26:
#35690 has the coverity tag added.