zephyrproject-rtos/zephyr #35656: arch.interrupt.arm.arm_interrupt: hangs on mimxrt685_evk_cm33

hakehuang, 2021-05-26:
**Describe the bug**
test application test_arm_interrupt hangs with below
Caught system error -- reason 4
Wrong crash type got 4 expected 0

**To Reproduce**
Steps to reproduce the behavior:
1.cd tests/arch/arm/arm_interrupt;  mkdir build; cd build
2. cmake -DBOARD=mimxrt685_evk_cm33
3. make; make flash
4. See error

**Expected behavior**
test pass

**Impact**
interrupt handling

**Logs and console output**

```
*** Booting Zephyr OS build v2.6.0-rc1-323-gae4fb20f6733  ***
Running test suite arm_interrupt
===================================================================
START - test_arm_null_pointer_exception
Skipped
PASS - test_arm_null_pointer_exception in 0.1 seconds
===================================================================
START - test_arm_interrupt
Available IRQ line: 59
E: &gt;&gt;&gt; ZEPHYR FATAL ERROR 1: Unhandled interrupt on CPU 0
E: Current thread: 0x30180140 (test_arm_interrupt)
Caught system error -- reason 1
E: r0/a1:  0x00000003  r1/a2:  0x301831d8  r2/a3:  0x00000003
E: r3/a4:  0x30182fc0 r12/ip:  0xa0000000 r14/lr:  0x18003f0f
E:  xpsr:  0x6100004b
E: Faulting instruction address (r15/pc): 0x1800190a
E: &gt;&gt;&gt; ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x30180140 (test_arm_interrupt)
Caught system error -- reason 3
E: r0/a1:  0x00000004  r1/a2:  0x301831d8  r2/a3:  0x00000004
E: r3/a4:  0x30182fc0 r12/ip:  0x00000000 r14/lr:  0x18003f0f
E:  xpsr:  0x6100004b
E: Faulting instruction address (r15/pc): 0x18001928
E: &gt;&gt;&gt; ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x30180140 (test_arm_interrupt)
Caught system error -- reason 4
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:216
Intentional assert
E: r0/a1:  0x00000004  r1/a2:  0x000000d8  r2/a3:  0x80000000
E: r3/a4:  0x0000004b r12/ip:  0xa0000000 r14/lr:  0x18003f0f
E:  xpsr:  0x4100004b
E: Faulting instruction address (r15/pc): 0x1800d0a0
E: &gt;&gt;&gt; ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x30180140 (test_arm_interrupt)
Caught system error -- reason 4
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/drivers/timer/sys_clock_init.c:23
E: r0/a1:  0x00000004  r1/a2:  0x00000017  r2/a3:  0x80000000
E: r3/a4:  0x0000000f r12/ip:  0xa0000000 r14/lr:  0xfffffffd
E:  xpsr:  0x4100000f
E: Faulting instruction address (r15/pc): 0x1800d0a0
E: &gt;&gt;&gt; ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x30180140 (test_arm_interrupt)
Caught system error -- reason 4
Wrong crash type got 4 expected 0
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.6.0-rc1-158-gcd9421a45e72

ioannisg, 2021-05-27:
Looks like a stack overflow

mmahadevan108, 2021-05-27:
The test is failing the below lines
https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/arch/arm/arm_interrupt/src/arm_interrupt.c#L336-L353

The test is expecting a CPU exception and RT600 is throwing a Kernel Panic. I am not clear how to address this.

ioannisg, 2021-05-27:
I'll take a look.

mmahadevan108, 2021-06-02:
@ioannisg Any guidance on this issue?

hakehuang, 2021-06-23:
lpcxpresso55s28 has the same problem.
'''
?*** Booting Zephyr OS build zephyr-v2.6.0-286-g46029914a7ac  ***
Running test suite arm_interrupt
===================================================================
START - test_arm_null_pointer_exception
Skipped
PASS - test_arm_null_pointer_exception in 0.1 seconds
===================================================================
START - test_arm_interrupt
Available IRQ line: 59
E: >>> ZEPHYR FATAL ERROR 1: Unhandled interrupt on CPU 0
E: Current thread: 0x30000138 (test_arm_interrupt)
Caught system error -- reason 1
E: r0/a1:  0x00000003  r1/a2:  0x300031e0  r2/a3:  0x00000003
E: r3/a4:  0x30002fc0 r12/ip:  0xa0000000 r14/lr:  0x100025f7
E:  xpsr:  0x6100004b
E: EXC_RETURN: 0x0
E: Faulting instruction address (r15/pc): 0x1000070a
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x30000138 (test_arm_interrupt)
Caught system error -- reason 3
E: r0/a1:  0x00000004  r1/a2:  0x300031e0  r2/a3:  0x00000004
E: r3/a4:  0x30002fc0 r12/ip:  0x00000000 r14/lr:  0x100025f7
E:  xpsr:  0x6100004b
E: EXC_RETURN: 0x0
E: Faulting instruction address (r15/pc): 0x10000728
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x30000138 (test_arm_interrupt)
Caught system error -- reason 4
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/tests/arch/arm/arm_interrupt/src/arm_interrupt.c:216
Intentional assert
E: r0/a1:  0x00000004  r1/a2:  0x000000d8  r2/a3:  0x80000000
E: r3/a4:  0x0000004b r12/ip:  0xa0000000 r14/lr:  0x100025f7
E:  xpsr:  0x4100004b
E: EXC_RETURN: 0x0
E: Faulting instruction address (r15/pc): 0x1000b9a4
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Fault during interrupt handling
E: Current thread: 0x30000138 (test_arm_interrupt)
Caught system error -- reason 4
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
ESF could not be retrieved successfully. Shall never occur.
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
ESF could not be retrieved successfully. Shall never occur.

'''

mmahadevan108, 2021-06-23:
Yes it would, thank you for looking into this.

mmahadevan108, 2021-06-23:
I have tested on mimxrt685_evk. This test passes with this fix.

PR #36225 (2021-06-24): tests: arm_interrupt: remove redundant test case part
The test case part in test_arm_interrupt, which is
triggering a SysTick IRQ (when SysTick is not used
for system timer implementation), to catch a spurious
interrupt is not really testing what it is designed
for. The reason is that the z_arm_exc_spurious has
be re-designed to be invoked only if the kernel is
built without support for system clock, i.e. with
CONFIG_SYS_CLOCK_EXISTS=n, but this configuration is
not compatible with ZTESTs and is, therefore, never
tested in this test, in any configuration. When the
SysTick is not implemented in [... cut]
closed 2021-06-24 by ioannisg