zephyrproject-rtos/zephyr #35658: arch.interrupt.arm.irq_vector_table.arm_irq_vector_table: Halting system for mximxrt685_evk_cm33

hakehuang, 2021-05-26:
**Describe the bug**
irq_vector_table test hangs

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -Dmimxrt685_evk_cm33 ..
3. make; make flash
4. See error

**Expected behavior**
PASS

**Impact**
irq vector teable

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.6.0-rc1-323-gae4fb20f6733

hakehuang, 2021-05-26:
@MaureenHelm , @dleach02 , @mmahadevan108 , this issue also existing in RC1

MaureenHelm, 2021-05-26:
git bisect landed on commit ff398d81a46596cdc18b7fe5ea4ae24d8599940c:
```
commit ff398d81a46596cdc18b7fe5ea4ae24d8599940c
Author: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>
Date:   Fri Apr 23 16:50:13 2021 -0500

    drivers: timer: rt600: Update SYS_CLOCK_TICKS_PER_SEC
    
    Use the default value as we were failing kernel
    timer tests with the increased value
    
    Signed-off-by: Mahesh Mahadevan <mahesh.mahadevan@nxp.com>
```

mmahadevan108, 2021-05-27:
This issue seems to be related to switching the kernel tick timer away from SYSTICK to OS Event timer. The test is failing the below lines
https://github.com/zephyrproject-rtos/zephyr/blob/main/tests/arch/arm/arm_interrupt/src/arm_interrupt.c#L336-L353
because the test is expecting a CPU exception instead and RT600 is throwing a Kernel Panic. 

I am not clear how to address this.

MaureenHelm, 2021-06-04:
I think the test needs to be updated similar to Nordic platforms:
https://github.com/zephyrproject-rtos/zephyr/blob/45bd3dee08ef9804fd4684ae20fde4e53af2c0f2/tests/arch/arm/arm_irq_vector_table/src/arm_irq_vector_table.c#L177-L196

dleach02, 2021-06-24:
#36278 points to the underlying issue here.

After some discussion with @ioannisg it may be that we are doing this test correctly but that the original test/nordic board exhibited a different behavior because of #36278

PR #36526 (2021-07-07): tests: arch: arm_irq_vector_table: Update to run on MXRT685
Fixes #35658
Update the custom vector table to add the OS Event timer interrupt which is used on RT685 as the kernel system timer
closed 2021-07-07 by nashif