zephyrproject-rtos/zephyr #35671: support llvm/lld linker

tejlmand, 2021-05-26:
Support for lld instead of ld.bfd was introduced here:
 #31706, in this commit: 2a59fa52b8e8fe2923a5295cf6c2a71a5eb837c8

by setting:
```
find_program(CMAKE_LINKER     ld.lld)
```

However, CMake doesn't use the `CMAKE_LINKER` variable for linking, instead it uses the compiler which then calls the linker.

This means that although this code indicates the lld is used, in fact clang will call the `ld` per default.

If setting `/usr/bin/ld --> /use/bin/lld` as described in: #32237 then the following linking error is seen:
```
ccache /usr/bin/clang   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/projects/github/ncs/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/posix/core/libarch__posix__core.a  zephyr/soc/posix/inf_clock/libsoc__posix__inf_clock.a  zephyr/boards/posix/native_posix/libboards__posix__native_posix.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/posix/core/offsets/offsets.c.obj  -L/projects/github/ncs/zephyr/build/zephyr  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -m32  -ldl  -pthread  -lm && cd /projects/github/ncs/zephyr/build/zephyr && /opt/cmake-3.20.2-linux-x86_64/bin/cmake -E echo
ld: error: unknown argument '--sort-common=descending'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

another way to reproduce is to manually force the use of `lld` by adding `-fuse-ld=lld` to linker invocation, as example by appying:
```diff
diff --git a/cmake/linker/lld/target.cmake b/cmake/linker/lld/target.cmake
index a4d0b2f915..9fa92d1214 100644
--- a/cmake/linker/lld/target.cmake
+++ b/cmake/linker/lld/target.cmake
@@ -84,6 +84,7 @@ function(toolchain_ld_link_elf)
   target_link_libraries(
     ${TOOLCHAIN_LD_LINK_ELF_TARGET_ELF}
     ${TOOLCHAIN_LD_LINK_ELF_LIBRARIES_PRE_SCRIPT}
+    -fuse-ld=lld
     ${TOPT}
     ${TOOLCHAIN_LD_LINK_ELF_LINKER_SCRIPT}
     ${TOOLCHAIN_LD_LINK_ELF_LIBRARIES_POST_SCRIPT}
```
**Describe the solution you'd like**
Proper support for lld.

**Describe alternatives you've considered**
N/A

**Additional context**

XiaZhouZero, 2021-08-15:
Does anybody know how to build Zephyr and generate the whole IR file of an application, e.g., sample/hello_world?
I tried to add compiler flags in the application CmakeLists.txt by add this command `zephyr_library_compile_options(-flto -fuse-ld=lld)` . However, the building process failed when linking the libapp.a file.

Thanks for your help in advance!

tejlmand, 2021-09-03:
> Does anybody know how to build Zephyr with LLVM/Clang and generate the whole IR file of an application, e.g., sample/hello_world?

@XiaZhouZero Could you open a dedicated question on this topic, as it is unrelated to this issue.
Issue if there is something you think should be improved.
Question if you just need to know more.

That being said, are you talking about the final image or each individual object file ?
As compile option you may use the `COMPILER_OPT` in Kconfig to add additional compiler options, like `-emit-llvm`.

And that will give you the llvm bytecode, however there are python scripts that relies on the elf object file as output, so it's not really useful as such as those scripts will then fail.

So it could be we should have some kind of possibility to allow users to also output an additional format, bytecode, assembly, pre-processed file, etc.

PR #63396 (2023-10-03): cmake: LLVM LLD minimum version 14.0.0
Fixes: #35671

Add minimal version required for LLVM LLD linker.
Linking fails with older LLVM LLD, such as v10.0.0.

LLVM v14.0.0 was released in 2022, and latest LLVM is v17.0.1. Zephyr currently doesn't have a strict minimum version of LLVM specified, but based on LLVM development and known issues on older releases, then a minimum version of v14.0.0 has been chosen in this commit.
commit 026dbdf5 (2023-10-03): cmake: LLVM LLD minimum version 14.0.0

Fixes: #35671

Add minimal version required for LLVM LLD linker.
Linking fails with older LLVM LLD, such as v10.0.0.

LLVM v14.0.0 was released in 2022, and latest LLVM is v17.0.1.
Zephyr currently doesn't have a strict minimum version of LLVM
specified, but based on LLVM development and known issues on older
releases, then a minimum version of v14.0.0 has been chosen in this
commit.
closed 2023-10-03 by carlescufi