zephyrproject-rtos/zephyr #35677: samples/subsys/console/getchar and samples/subsys/console/getline build breaks for arduino_nano_33_ble build

nageshshamnur, 2021-05-26:
**Describe the bug**
when try to build console getchar and getline examples with west build, it fails due to undefined symbol reference

**To Reproduce**
Steps to reproduce the behavior:
```west build -p auto -b arduino_nano_33_ble samples/subsys/console/getline```
```west build -p auto -b arduino_nano_33_ble samples/subsys/console/getchar```
error as below:
```
ohos-bld@root1-Nitro-AN515-55:~/zephyrproject/zephyr$ west build -p auto -b arduino_nano_33_ble samples/subsys/console/getline
-- west build: making build dir /home/ohos-bld/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/ohos-bld/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/ohos-bld/zephyrproject/zephyr/samples/subsys/console/getline
-- Zephyr version: 2.6.0-rc1 (/home/ohos-bld/zephyrproject/zephyr), build: v2.6.0-rc1-336-g457a28bf78a4
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: arduino_nano_33_ble
-- Cache files will be written to: /home/ohos-bld/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.3 (/home/ohos-bld/zephyr-sdk-0.12.3)
-- Found dtc: /home/ohos-bld/zephyr-sdk-0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/ohos-bld/zephyrproject/zephyr/boards/arm/arduino_nano_33_ble/arduino_nano_33_ble.dts
-- Generated zephyr.dts: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/ohos-bld/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ohos-bld/zephyrproject/zephyr/boards/arm/arduino_nano_33_ble/arduino_nano_33_ble_defconfig'
Merged configuration '/home/ohos-bld/zephyrproject/zephyr/samples/subsys/console/getline/prj.conf'
Configuration saved to '/home/ohos-bld/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'

warning: The choice symbol CONSOLE_GETLINE (defined at subsys/console/Kconfig:21) was selected (set
=y), but no symbol ended up as the choice selection. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_CONSOLE_GETLINE.html and/or look up
CONSOLE_GETLINE in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ohos-bld/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Deprecation Warning at /home/ohos-bld/zephyrproject/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
CMake Warning (dev) at ../../../../cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  ../../../../CMakeLists.txt:1253 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at ../../../../cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  ../../../../CMakeLists.txt:1198 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/ohos-bld/zephyrproject/zephyr/build
-- west build: building application
[1/137] Preparing syscall dependency handling

[130/137] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map 
: && ccache /home/ohos-bld/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/home/ohos-bld/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"/home/ohos-bld/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/thumb/v7e-m/nofp"  -L/home/ohos-bld/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/ohos-bld/zephyrproject/zephyr/build/zephyr && /usr/bin/cmake -E echo
/home/ohos-bld/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: app/libapp.a(main.c.obj): in function `main':
/home/ohos-bld/zephyrproject/zephyr/samples/subsys/console/getline/src/main.c:14: undefined reference to `console_getline_init'
/home/ohos-bld/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: /home/ohos-bld/zephyrproject/zephyr/samples/subsys/console/getline/src/main.c:19: undefined reference to `console_getline'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/ohos-bld/zephyrproject/zephyr/build
```

**Expected behavior**
A clear and concise description of what you expected towes happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
west build needs to be success

galak, 2021-05-26:
This is expected behavior on this board since the "console" is on USB_UART_CONSOLE which does not support input according to the Kconfig.

galak, 2021-05-26:
See https://github.com/zephyrproject-rtos/zephyr/blob/4084242a71d0a93d97a83776b995785bd28ec883/drivers/console/Kconfig#L105

If you look at https://github.com/zephyrproject-rtos/zephyr/blob/4084242a71d0a93d97a83776b995785bd28ec883/subsys/console/Kconfig#L9 you'll see that CONSOLE_GETCHAR & CONSOLE_GETLINE depend on UART_CONSOLE

closed 2021-05-26 by galak