zephyrproject-rtos/zephyr #35681: Unable to get ouput for samples/subsys/logging/logger and samples/philosophers

nageshshamnur, 2021-05-26:
**Describe the bug**
Env details:
Ubuntu 20.04.2 LTS running zephyr project from master and zephyr sdk version: zephyr-sdk-0.12.3
Hardware: Arduino nano 33 ble sense. 
I have setup the zephyr environment as explained in the page: https://docs.zephyrproject.org/latest/getting_started/index.html. 
I have connected by Arduino nano 33 ble sense board to my ubuntu laptop via USB to mini-USB cable which powers my board and as well as help in flashing it via /dev/ttyACM0 port. I have also connected USB to TTL cable to my board in parallel to USB cable with only TX/RX pins  on my arduino nano 33 ble so that i can watch the output on minicom. 

**To Reproduce**
Steps to reproduce the behavior:
1. ```west build -p auto -b arduino_nano_33_ble samples/subsys/logging/logger```
```
$ west build -p auto -b arduino_nano_33_ble samples/subsys/logging/logger
-- west build: making build dir /home/ohos-bld/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/ohos-bld/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/ohos-bld/zephyrproject/zephyr/samples/subsys/logging/logger
-- Zephyr version: 2.6.0-rc1 (/home/ohos-bld/zephyrproject/zephyr), build: v2.6.0-rc1-336-g457a28bf78a4
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: arduino_nano_33_ble
-- Cache files will be written to: /home/ohos-bld/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.3 (/home/ohos-bld/zephyr-sdk-0.12.3)
-- Found dtc: /home/ohos-bld/zephyr-sdk-0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/ohos-bld/zephyrproject/zephyr/boards/arm/arduino_nano_33_ble/arduino_nano_33_ble.dts
-- Generated zephyr.dts: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/ohos-bld/zephyrproject/zephyr/samples/subsys/logging/logger/Kconfig
Loaded configuration '/home/ohos-bld/zephyrproject/zephyr/boards/arm/arduino_nano_33_ble/arduino_nano_33_ble_defconfig'
Merged configuration '/home/ohos-bld/zephyrproject/zephyr/samples/subsys/logging/logger/prj.conf'
Configuration saved to '/home/ohos-bld/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/ohos-bld/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ohos-bld/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Deprecation Warning at /home/ohos-bld/zephyrproject/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
CMake Warning (dev) at ../../../../cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  ../../../../CMakeLists.txt:1253 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at ../../../../cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  ../../../../CMakeLists.txt:1198 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/ohos-bld/zephyrproject/zephyr/build
-- west build: building application
[1/146] Preparing syscall dependency handling

[139/146] Linking C executable zephyr/zephyr_prebuilt.elf

[146/146] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       24760 B       960 KB      2.52%
            SRAM:        8928 B       256 KB      3.41%
        IDT_LIST:          0 GB         2 KB      0.00%
```
2. ```west flash --bossac="/home/ohos-bld/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2/bossac"```
```
$ west flash --bossac="/home/ohos-bld/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2/bossac"
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner bossac
Erase flash

Done in 0.002 seconds
Write 24760 bytes to flash (7 pages)
[==============================] 100% (7/7 pages)
Done in 1.082 seconds
Verify 24760 bytes of flash
[==============================] 100% (7/7 pages)
Verify successful
Done in 0.038 seconds
Set boot flash true
```
3. In another shell connect minicom ```sudo minicom -s" with configuration as baud_rate=115200 partiy=None stopbits=1 but prints are visisble on the board.
```$ sudo minicom -s```
```
Welcome to minicom 2.7.1                                                                                                               
                                                                                                                                       
OPTIONS: I18n                                                                                                                          
Compiled on Dec 23 2019, 02:06:26.                                                                                                     
Port /dev/ttyUSB0, 18:59:12                                                                                                            
                                                                                                                                       
Press CTRL-A Z for help on special keys                                                                                                
```
**Expected behavior**
console display should be visible on minicom.

**Impact**
Unable to proceed with using arduino nano 33 ble board with zephyr 

**Logs and console output**
as explained above

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.2 LTS
 - Toolchain: Zephyrproject: master and zephyr-sdk: zephyr-sdk-0.12.3
Hardware: Arduino nano 33 ble sense. 
 - Commit SHA or Version used: 457a28bf78a4cdd2b17f3983c50a0183f2f71700 for zephyrproject.

**Additional context**
I have connected by Arduino nano 33 ble sense board to my ubuntu laptop via USB to mini-USB cable which powers my nano board and as well as help in flashing it via /dev/ttyACM0 port. I have also connected USB to TTL cable to my board in parallel to USB cable with only TX/RX pins  on my arduino nano 33 ble boad.
flashing on usb port: /dev/ttyACM0
serial console (minicom): /dev/ttyUSB0

i even tried chaning prj.conf as explained in the issue:  https://github.com/zephyrproject-rtos/zephyr/issues/26104 that also did not solve my problem.
Thanks!!!

nageshshamnur, 2021-05-27:
This issue was happening due to incorrect connection of Rx/Tx pins in USB-TTL cable. Adding the image here for the benefit of all people. 
Closing this issue as incorrect one.
[image]

nageshshamnur, 2021-06-09:
Waiting for PR to be merged which is related to this issue. This is required since CDC_ACM gets disconnected once flashing is done.

jfischer-no, 2021-06-09:
> This issue was happening due to incorrect connection of Rx/Tx pins in USB-TTL cable. Adding the image here for the benefit of all people.
> Closing this issue as incorrect one.

If it is not a bug then it does not need to be open.

closed 2021-06-09 by jfischer-no