zephyrproject-rtos/zephyr #35707: zephyr/tests/kernel/common test case is failing with gcc-11 (Yocto)

saininav, 2021-05-27:
I am building stable zephyr v2.5.0 with Yocto toolchain using [meta-zephyr](http://git.yoctoproject.org/cgit/cgit.cgi/meta-zephyr) layer. 

GCC : 11.1.1
Python: 3.9.5
Board: up_squared

No issue observed with GCC 10.2.

Observed that few other [kernel testcases](https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v2.5.0/tests/kernel) are also failing with same error:
zephyr/tests/kernel/common
zephyr/tests/kernel/common
zephyr/tests/kernel/common

**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

nashif, 2021-05-27:
I doubt this is board specific. This is a new toolchain that is not supported, not tested, etc. Please use the Zephyr SDK or a toolchain that is well-tested, for example 10.2

nashif, 2021-05-27:
oops, closed by mistake

nashif, 2021-05-27:
I can reproduce this with fedora 34 and `export ZEPHYR_TOOLCHAIN_VARIANT=host`

nashif, 2021-05-27:
@dcpleung  take a look at the syscall script thing, you might have an idea what is going on.

dcpleung, 2021-06-01:
#35852 takes care of some issues but not all of them.

There is an issue with linker section.

Another one is about `popcountdi2` being undefined, where we are actually using GCC's builtin popcount function. So this need some investigation.

dcpleung, 2021-07-26:
@saininav Could you try again with the latest main branch? The DWARF error should have been fixed.

saininav, 2021-07-27:
> @saininav Could you try again with the latest main branch? The DWARF error should have been fixed.

Yes I can build with latest main. Thanks.

Would this fix, be backported to release branch 'v2.6-branch'  ?

dcpleung, 2021-08-04:
Backporting via #37447 and #37448

dcpleung, 2021-08-30:
Both backport PRs have been merged.

dcpleung, 2021-09-14:
@saininav Could you confirm that the backport works for you in v2.6?

saininav, 2021-09-24:
It works. Thanks for backporting fixes.

closed 2021-09-24 by dcpleung

vindicatorr, 2021-12-09:
I've got the latest `main` branch today as well as `Zephyr SDK 0.13.2`, however the issue also occurs for me with `arm-none-eabi-gcc (Arch Repository) 11.2.0` as well as the older `arm-zephyr-eabi-gcc (crosstool-NG 1.24.0.378_e011758) 10.3.0`

```
> Executing task: ninja -C ./build footprint <

ninja: Entering directory `./build'
[1/151] Preparing syscall dependency handling

[140/151] Linking C executable zephyr/zephyr_pre0.elf

[144/151] Linking C executable zephyr/zephyr_pre1.elf

[150/151] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       13076 B      1020 KB      1.25%
            SRAM:        4192 B       256 KB      1.60%
        IDT_LIST:          0 GB         2 KB      0.00%
[150/151] cd <pathTo>/<project>/build...<pathTo>/<project>/build -d 99 all -q
Traceback (most recent call last):
  File "<pathTo>/zephyr//scripts/footprint/size_report", line 803, in <module>
    main()
  File "<pathTo>/zephyr//scripts/footprint/size_report", line 779, in main
    do_simple_name_matching(elf, symbol_dict, processed)
  File "<pathTo>/zephyr//scripts/footprint/size_report", line 268, in do_simple_name_matching
    for compile_unit in dwarfinfo.iter_CUs():
  File "/usr/lib/python3.9/site-packages/elftools/dwarf/dwarfinfo.py", line 353, in _parse_CUs_iter
    cu = self._cached_CU_at_offset(offset)
  File "/usr/lib/python3.9/site-packages/elftools/dwarf/dwarfinfo.py", line 383, in _cached_CU_at_offset
    cu = self._parse_CU_at_offset(offset)
  File "/usr/lib/python3.9/site-packages/elftools/dwarf/dwarfinfo.py", line 417, in _parse_CU_at_offset
    cu_structs = DWARFStructs(
  File "/usr/lib/python3.9/site-packages/elftools/dwarf/structs.py", line 92, in __init__
    assert address_size == 8 or address_size == 4
AssertionError
FAILED: zephyr/cmake/reports/CMakeFiles/footprint <pathTo>/<project>/build/zephyr/cmake/reports/CMakeFiles/footprint 
cd <pathTo>/<project>/build && /usr/bin/python3.9 <pathTo>/zephyr//scripts/footprint/size_report -k <pathTo>/<project>/build/zephyr/zephyr.elf -z <pathTo>/zephyr/ -o <pathTo>/<project>/build -d 99 all -q
```