zephyrproject-rtos/zephyr #35711: net: sockets: dtls: handshake not reset as it ought to be

emillindq, 2021-05-27:
**Describe the bug**
The bug is as follows: I am a client in the DTLS handshake and the remote is not replying. I am using `sendto` to start the DTLS handshake. My end will retransmit the client hello message until timeout. This works well.

I will receive error on `sendto` since remote is gone. When I call0 `sendto` again, I would expect the DTLS handshake to be reset and start over if it's not complete. But what happens is that nothing comes out of the underlying socket because the mbedTLS session is not reset as it ought to be.

This happens:
1. sockets_tls.c calls `mbedtls_ssl_handshake` and gets back `MBEDTLS_ERR_SSL_TIMEOUT`. We return error and continue on our day. According to mbedTLS manual (paraphrased)
>If this function returns `MBEDTLS_ERR_SSL_TIMEOUT`, you must either free it or call mbedtls_ssl_session_reset() on it before re-using it for a new connection
https://tls.mbed.org/api/ssl_8h.html#a4a37e497cd08c896870a42b1b618186e

2. When I call `sendto` again, `mbedtls_ssl_handshake` will be called without being reset, and the internal state within mbedTLS will be "I'm waiting on data" -> nothing comes out and handshake is not retried.

Am I supposed to call close() if sendto returns error? This doesn't make any sense to me since I am operating on an UDP socket (not really, but I as a user would at least expect it to work in a similar way).

Is my expectations  wrong?

**To Reproduce**
Steps to reproduce the behavior:
1. Call sendto twice without remote.
2. On second call no client hello will come out.

**Expected behavior**
A new handshake should be started.

**Impact**
Annoyance if I have to call `close()` every time handshake fails.

**Environment (please complete the following information):**
 - Commit SHA 6e2522d

**Additional context**
Add any other context about the problem here.

rlubos, 2021-06-15:
Sorry for late reply, I somehow missed this issue.

> Am I supposed to call close() if sendto returns error? This doesn't make any sense to me since I am operating on an UDP socket (not really, but I as a user would at least expect it to work in a similar way).

Actually, that was my original assumption - if an error occurs on the TLS/DTLS socket (other than EAGAIN), you need to close it before reuse. I wouldn't compare it with the UDP in terms of error recovery - ok, it's still datagram underneath, but mbedTLS itself is a complex beast, and I wouldn't be confident enough to re-use its context (although it seems to be possible though from what you quote).

> 1. sockets_tls.c calls `mbedtls_ssl_handshake` and gets back `MBEDTLS_ERR_SSL_TIMEOUT`. We return error and continue on our day. According to mbedTLS manual (paraphrased)

This actually confirms that our approach isn't wrong, so you currently need to "free" the TLS context (i. e. close the socket) to make it available again for a new connection. I think though we could also implement it the other way, it should be pretty straightforward to add `tls_mbedtls_reset()` in vulnerable places (i. e. when error occurs) so that the socket is ready to start over the handshake process w/o closing it. Specifically I'd see it inside `tls_mbedtls_handshake()` or `send_tls()`, perhaps other places as well, would need a closer look. 

Are you willing to send a patch?

emillindq, 2021-06-29:
@rlubos sorry for late reply! Unfortunately I've been busy for the past two weeks with critical stuff needed to be done prior vacations...

I can definitely look at it, but as for testing I feel that there's a gazillion amount of cases that would be needed to be tested to make sure I didn't break something. I know about zephyr/tests/net/socket/tls/ but this doesn't seem exhaustive since it doesn't have a link partner to test against. Is there something already existing in Zephyr that I can use to verify this properly?

rlubos, 2021-06-29:
`zephyr/tests/net/socket/tls_ext` verifies the TLS handshake/data exchange, it uses loopback interface. Unfortunately it only supports TLS, no DTLS, so it'd be a little help in this case.

For basic testsing, I typically use the `echo_client/server` samples, built with `overlay-tls.conf`, running against each other. In default configuration they exchange both TLS and DTLS packets. This particular case however (re-running handshake w/o reopening a socket) might need some custom test though.

emillindq, 2021-07-02:
Leaving a notice I will go on vacation now. Somebody else can take over, but I will get back to this in 4 weeks time.

PR #37433 (2021-08-06): net: sockets: dtls: reset mbedtls session on timed out handshake
This PR fixes #35711

According to MbedTLS API documentation, its session must be reset if mbedtls_ssl_handshake returns timeout error. This commit resets the session for said return value, and that allows us to call send() multiple times even if handshake times out for previous calls.

Please note that `k_sem_init` resets the semaphores internal list of polls waiting for it, and therefore a poll waiting on the `tls_established` sem will block forever if first handshake fails. This was changed to `k_sem_reset` in this PR.

This P [... cut]
commit f9023d2c (2021-08-06): net: sockets: dtls: reset mbedtls session on timed out handshake

According to MbedTLS API documentation, its session must be
reset if mbedtls_ssl_handshake returns timeout error. This
commit resets the session for said return value, and that
allows us to call send() multiple times even if handshake
times out for previous calls.

Fixes #35711
closed 2021-08-06 by cfriedt