zephyrproject-rtos/zephyr #35718: Error messages from fs

emillindq, 2021-05-27:
**Is your enhancement proposal related to a problem? Please describe.**
The problem I'm having when using the file system is that I get spammed by error messages when doing operations whose rc gives me the information I want, or that is not error at all. For example:
* `fs_mkdir()` prints an error if the directory already exists. settings_file.c calls this function to create a directory on boot, and causes an error to be printed on every boot. You might think "oh well just check if it exists", well,
* `fs_stat()` will also print an error if a non-existing dir or file tries to be statted. But that's one purpose to call it, to check if the file exists and therefore is not an error. In addition, how can we create a directory without getting an error? Not possible it seems like.

**Describe the solution you'd like**
FS subsystem should either check the return code and not print error if for example `-ENOENT` in `fs_stat()`, or not print errors at all and let the caller interpret the return code to interpret if it's an error.

**Additional context**
I am using FAT file system. I am a software developer so I can implement whatever the consensus is.

jfischer-no, 2021-05-28:
> **Describe the solution you'd like**
> FS subsystem should either check the return code and not print error if for example `-ENOENT` in `fs_stat()`, or not print errors at all and let the caller interpret the return code to interpret if it's an error.

What is preventing you from turning FS logging off with `CONFIG_FS_LOG_LEVEL_OFF=y`?

emillindq, 2021-05-28:
> What is preventing you from turning FS logging off with `CONFIG_FS_LOG_LEVEL_OFF=y`?

Yes of course I could do that, but there are errors that are actual errors (for example "mount point not found") that I want to see. My point is that it should be possible to do all types of fs accesses without triggering errors, when there is no error. As mentioned, how to find out if a folder exists?

nashif, 2021-05-29:
Can you point to where this log message is in the code? Difficult to find the code concerned based on the above.

emillindq, 2021-05-31:
> Can you point to where this log message is in the code? Difficult to find the code concerned based on the above.

I'm sorry, should have been more explicit.
If I call fs_mkdir then if  the dir exists, this error will be printed with rc=-17 (-EEXST)
https://github.com/zephyrproject-rtos/zephyr/blob/3d39f72a88b3100ac83e7c6c285ae567aca640d0/subsys/fs/fs.c#L501-L504

If I try to stat it to avoid mentioned error, the following error is printed if it doesn't exist, with rc=-2 (-ENOENT)
https://github.com/zephyrproject-rtos/zephyr/blob/3d39f72a88b3100ac83e7c6c285ae567aca640d0/subsys/fs/fs.c#L603-L607

de-nordic, 2021-06-30:
The problem is that the VFS layer sometimes directly returns error, defined by it's own actions and logic, and sometimes passes error from underlying system, with no translation.
The log messages after API calls are often the only indication that the error source is not the VFS layer, but it has been passed up from the file system driver.
When call to underlying API fails, this is actually error because the operation has not been completed; logging the result as "ERROR" is correct, although annoying: it basically logs where the error came from.

emillindq, 2021-06-30:
> The problem is that the VFS layer sometimes directly returns error, defined by it's own actions and logic, and sometimes passes error from underlying system, with no translation.
> The log messages [...]

Alright this makes sense I guess, but at least it should be possible to do all file system operations without generating error if everything is working *as expected* (see example above with creating folder). Do you agree? With the omission of the LOG_ERR in `fs_stat` if file/folder doesn't exist as in PR then this becomes possible.

de-nordic, 2021-06-30:
> Alright this makes sense I guess, but at least it should be possible to do all file system operations without generating error if everything is working _as expected_ (see example above with creating [...]

Yes.  I do agree.
I was actually going to log proposal to get rid of all log messages from the VFS layer, but this would require some rework in flie system drivers, the Zephyr layer.
Logging information from VFS is pointless, in my opinion, because it does not really help; user probably wants to log own messages, depending on results of `fs_*` functions or posix calls, so that just bloats logs.
The file system drivers would have to translate errors to strictly defined set of errno codes and a file system driver would be responsible to log info when translation error code is the best the driver could decide it to be.

commit 1e9f1231 (2021-07-26): logging: fs: treat absent file on stat as no error

There are multiple reasons to want to find out if file or
directory exists, for example to create it. Stating and
finding out it doesn't exist should not cause an LOG_ERR call as
this gives information to the user in a normal call case.

Fixes #35718
closed 2021-07-26 by cfriedt