zephyrproject-rtos/zephyr #35719: WiFi Management expects networking to be offloaded

allopatin, 2021-05-27:
**Is your enhancement proposal related to a problem? Please describe.**

Currently I am working on porting the Silicon Labs WFM200 Wifi NCP to Zeyphr. That device is capable of managing network connections like connecting to a station, creating an AP or scanning for networks. However it does not implement an IP stack etc. The problem is that when using `CONFIG_WIFI_OFFLOAD` for enabling the wifi management API `CONFIG_NET_OFFLOAD` is automatically selected. That is in conflict to the NCP's functionality and it seems like I am not able to initilize a L2 driver.

**Describe the solution you'd like**

It would be great if the Wifi management API could be used without network offload. Hopefully this would allow to initialize the L2 driver without any conflicts. Alternatively the Wifi management API could be enhanced with additional function pointers to L2 driver.

**Describe alternatives you've considered**

Injecting my own L2 functions into the net_if struct by modifying the net_l2 member of net_if_dev to avoid using ETH_NET_DEVICE_INIT. But what will happen with the net_offload member? It might still override the net_l2 functions.
Alternatively it could be possible to write a full L2 driver and add the connection management functionality of the Wifi management API without calling `NET_DEVICE_OFFLOAD_INIT`.

An tips or ideas on how this issue could be solved would be greatly appreciated!
Thanks in advance!

rlubos, 2021-05-27:
@allopatin Hi, we're aware of the problem and we've already discussed this on one of the recent Networking Forums. The idea was to introduce WiFi L2 (based on ethernet L2), and use it instead of `struct net_if_api` inside `struct net_wifi_mgmt_offload`. This should allow implementing L2 functionality with the wifi management layer, w/o offloading. @sylvioalves Do we have any progress in this area?

allopatin, 2021-06-07:
@sylvioalves @rlubos Hello, would it be possible to get access to an early version of the API?

sylvioalves, 2021-06-10:
@allopatin Hi, sorry for the delay. I have not got any further yet. Do you think you can contribute with this change?

allopatin, 2021-06-14:
@sylvioalves I'd be happy to help if possible. Is there any starting point where I can get involved?

rlubos, 2021-06-21:
@allopatin I've sketched the initial idea on the following branch: https://github.com/rlubos/zephyr/commits/wifi/mgmt-integration-rework

The idea was rather simple, introduce `struct wifi_api`, based on `struct ethernet_api`, and use it within `struct net_wifi_mgmt_offload` instead of direct `struct net_if_api` pointer. This would allow to define devices that'd use both, the offloaded `wifi_mgmt` API, while still enabling L2 functionality (the assumption was that WiFi devcies would be re-using the Ethernet L2 implementation). 

This has also minimal impact for the existing fully offloaded devices, as they are still able to use `NET_DEVICE_OFFLOAD_INIT()` macro (or similar) to define the fully offloaded device. The devices that only offload the WiFi configuration would use `NET_DEVICE_INIT()` macro (or similar) instead, and would need to provide appropriate driver functions from the `ethernet/wifi_api` in order to integrate with the Ethernet L2 (and thus the native networking stack).

Note though, that I've only summarized the idea we've come up with for the native stack integration, haven't tested this with any actual WiFi device that'd use this new scheme, so some adjustments might be needed.

rlubos, 2022-09-27:
Resolved with https://github.com/zephyrproject-rtos/zephyr/pull/49464

closed 2022-09-27 by rlubos