zephyrproject-rtos/zephyr #35720: tests:kernel timer fails on test_sleep_abs with TICKLESS_KERNEL and PM on nucleo_wb55rg

FRASTM, 2021-05-27:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
   `west build -p auto -b nucleo_wb55rg tests/kernel/timer/timer_api/
See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**```

** Booting Zephyr OS build v2.6.0-rc1  ***                                     
Running test suite timer_api                                                    
===================================================================             
START - test_time_conversions                                                   
 PASS - test_time_conversions in 0.115 seconds                                  
===================================================================             
START - test_timer_duration_period                                              
 PASS - test_timer_duration_period in 0.326 seconds                             
===================================================================             
START - test_timer_restart                                                      
 PASS - test_timer_restart in 0.277 seconds                                     
===================================================================             
START - test_timer_period_0                                                     
 PASS - test_timer_period_0 in 0.202 seconds                                    
===================================================================             
START - test_timer_period_k_forever                                             
 PASS - test_timer_period_k_forever in 0.203 seconds                            
===================================================================             
START - test_timer_expirefn_null                                                
 PASS - test_timer_expirefn_null in 0.327 seconds                               
===================================================================             
START - test_timer_periodicity                                                  
 PASS - test_timer_periodicity in 0.254 seconds                                 
===================================================================             
START - test_timer_status_get                                                   
 PASS - test_timer_status_get in 0.2 seconds                                    
===================================================================             
START - test_timer_status_get_anytime                                           
 PASS - test_timer_status_get_anytime in 0.327 seconds                          
===================================================================             
START - test_timer_status_sync                                                  
 PASS - test_timer_status_sync in 0.353 seconds                                 
===================================================================             
START - test_timer_k_define                                                     
 PASS - test_timer_k_define in 0.702 seconds                                    
===================================================================             
START - test_timer_user_data                                                    
 PASS - test_timer_user_data in 0.302 seconds                                   
===================================================================             
START - test_timer_remaining                                                    
 PASS - test_timer_remaining in 0.52 seconds                                    
===================================================================             
START - test_timeout_abs                                                        
 PASS - test_timeout_abs in 0.2 seconds                                         
===================================================================             
START - test_sleep_abs                                                          
                                                                                
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/mai)
expected wakeup at 14921, got 14922 (late 1)                                    
 FAIL - test_sleep_abs in 0.32 seconds                                          
===================================================================             
Test suite timer_api failed.                                                    
===================================================================             
PROJECT EXECUTION FAILED  
```
**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

FRASTM, 2021-05-27:
It can pass when increasing the "time slop" up to 250µs in the zassert_true ( ... k_us_to_ticks_ceil32(250)) of the test_sleep_abs()

FRASTM, 2021-06-01:
It seems like the commit 63972d6df7121f5133ded61d3cde56d30e2b708b from #32683 is involved in the test failure on this config on this nucleo_wb55rg board

affrinpinhero-2356, 2021-07-26:
Hi all, 
I have done some diggingg on this issue. I found it takes  exactly 1 tick extra to get back from sleep. For reference I have put my logs. For test purpose I have printed start and stop values also. 

```
*** Booting Zephyr OS build zephyr-v2.6.0-1390-gbc92b966aa8e  ***
Running test suite timer_api
===================================================================
START - test_time_conversions
 PASS - test_time_conversions in 0.114 seconds
===================================================================
START - test_timer_duration_period
 PASS - test_timer_duration_period in 0.327 seconds
===================================================================
START - test_timer_restart
 PASS - test_timer_restart in 0.277 seconds
===================================================================
START - test_timer_period_0
 PASS - test_timer_period_0 in 0.203 seconds
===================================================================
START - test_timer_period_k_forever
 PASS - test_timer_period_k_forever in 0.203 seconds
===================================================================
START - test_timer_expirefn_null
 PASS - test_timer_expirefn_null in 0.327 seconds
===================================================================
START - test_timer_periodicity
 PASS - test_timer_periodicity in 0.254 seconds
===================================================================
START - test_timer_status_get
 PASS - test_timer_status_get in 0.2 seconds
===================================================================
START - test_timer_status_get_anytime
 PASS - test_timer_status_get_anytime in 0.327 seconds
===================================================================
START - test_timer_status_sync
 PASS - test_timer_status_sync in 0.354 seconds
===================================================================
START - test_timer_k_define
 PASS - test_timer_k_define in 0.703 seconds
===================================================================
START - test_timer_user_data
 PASS - test_timer_user_data in 0.302 seconds
===================================================================
START - test_timer_remaining
 PASS - test_timer_remaining in 0.52 seconds
===================================================================
START - test_timeout_abs
 PASS - test_timeout_abs in 0.3 seconds
===================================================================
START - test_sleep_abs
Start : 14889, End : 14940, Tick Value : 1
    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:810: test_sleep_abs: (late >= 0 && late < k_us_to_ticks_ceil32(100) is false)
expected wakeup at 14939, got 14940 (late 1)
 FAIL - test_sleep_abs in 0.36 seconds
===================================================================
Test suite timer_api failed.
===================================================================
PROJECT EXECUTION FAILED
```

And I have found the following comment in _tests/kernel/timer/timer_api/src/main.c +802_.

```
/* Systems with very high tick rates and/or slow idle resume
	 * (I've seen this on intel_adsp) can occasionally take more
	 * than a tick to return from k_sleep().  Set a 100us real
	 * time slop.
	 */
```

erwango, 2021-07-27:
> And I have found the following comment in tests/kernel/timer/timer_api/src/main.c +802.

Looking to the [commit](https://github.com/zephyrproject-rtos/zephyr/commit/63972d6df7121f5133ded61d3cde56d30e2b708b) that introduced this comment. There are some more details in commit description:
```
On an intel_adsp (which has 50 kHz ticks, a
comparatively slow idle resume and interrupt entry, and even has two
CPUs to mess with latency measurements) I would occasionally see the
k_sleep() take more than a tick to wake up from the interrupt handler
until the return to application code. On an intel_adsp (which has 50 kHz ticks, a
comparatively slow idle resume and interrupt entry, and even has two
CPUs to mess with latency measurements) I would occasionally see the
k_sleep() take more than a tick to wake up from the interrupt handler
until the return to application code.
```

nucleo_wb55rg tick rate is around 2KHz, I'm not sure this case could be compared.
Besides, this is the same on disco_l475_iot1, so I don't think this is the factor to blame in that case.

One difference is the sysclock (32MHz on nucleo_wb55rg vs 80MHz on disco_l475_iot1),
which may potentially increase the latency in interrupt handler (x2).
It could be interesting to verify this hypothesis. 
@affrinpinhero-2356 Can you have a try changing nucleo_wb55rg clock-frequency to the maximum (64MHz) and see if it has impact ?

affrinpinhero-2356, 2021-07-28:
I have tried with 64MHz. still, it has the issue.

prasannaven, 2021-07-30:
@erwango can you suggest.

erwango, 2021-07-30:
> @erwango can you suggest.

As mentionned [here](https://github.com/zephyrproject-rtos/zephyr/issues/35720#issuecomment-887296881), if test is still failing with 64MHz sysclock, then I think the comment suggested here dosesn't apply to this target:
```
/* Systems with very high tick rates and/or slow idle resume
	 * (I've seen this on intel_adsp) can occasionally take more
	 * than a tick to return from k_sleep().  Set a 100us real
	 * time slop.
	 */
```
So investigation is still required

affrinpinhero-2356, 2021-08-02:
Hi all, 

I have done some debug. Following were my observation.
1.  _CONFIG_TEST_USERSPACE_ have some influence in this test. 
2. Increasing _k_usleep()_ also making this test pass. 

Log snip for Ksleep is _2_
```
START - test_sleep_abs
value before ksleep 36072
start-36084, end-36134
 PASS - test_sleep_abs in 0.8 seconds
```
Log snip for Ksleep is _1_ and _userspace_ disable
```
START - test_sleep_abs
value before ksleep 36206
start-36218, end-36268
 PASS - test_sleep_abs in 0.8 seconds
```

affrinpinhero-2356, 2021-08-04:
Looks like the issue was due to privilage modes in ARM. 

 

In ksleep implementation.

 

```
#ifdef CONFIG_USERSPACE
    if (z_syscall_trap()) {
        /* coverity[OVERRUN] */
        return (int32_t) arch_syscall_invoke1(*(uintptr_t *)&us, K_SYSCALL_K_USLEEP);
    }
#endif
    compiler_barrier();
    return z_impl_k_usleep(us);
```

 

And also some system registers were read in _arch_syscall_invoke1_ see the below code snip. 

 

```
static inline uintptr_t arch_syscall_invoke1(uintptr_t arg1,
                         uintptr_t call_id)
{
    register uint32_t ret __asm__("r0") = arg1;
    register uint32_t r6 __asm__("r6") = call_id;

 

    __asm__ volatile("svc %[svid]\n"
             : "=r"(ret)
             : [svid] "i" (_SVC_CALL_SYSTEM_CALL),
               "r" (ret), "r" (r6)
             : "r8", "memory", "r1", "r2", "r3", "ip");
    return ret;
}
```

 

ARM core which was in STm32WB55RG was in a mode that can't access system registers. Kindly find the below screenshot. 

 
[image]

Since these registers weren't called, k_sleep() were not acting the way it was to.

FRASTM, 2021-08-05:
interesting. 
It means that  when  CONFIG_USERSPACE is defined, the stm32WB55RG, cannot read the system registers as the `arch_syscall_invoke1()` function asks for.

affrinpinhero-2356, 2021-08-06:
Since this was at the Kernel side. 
Can you suggest how to proceed with the same?

FRASTM, 2021-08-24:
In the configuration of the testcase /tests/kernel/timer/timer_api  with  **prj_tickless.conf** 
where CONFIG_PM is selected, the LPTIMer is use as the kernel systick 
then the `CONFIG_SYS_CLOCK_TICKS_PER_SEC 4096` 
The comment of the /tests/kernel/timer/timer_api/src/main.c shows that, for nucleo_stm32wb55eg, a real time slop of about 250us is required:
`zassert_true(late >= 0 && late < k_us_to_ticks_ceil32(`**250**`),`

FRASTM, 2021-08-30:
When the nucleo_wb55rg has MSI clock source at 32MHz,  instead of HSE 32MHz, the test is PASSED. DTS of the nucleo_wb55rg is then 
```
+&clk_msi {
+	status = "okay";
+	msi-range = <10>; /* 32MHz */
+};

 &rcc {
-	clocks = <&clk_hse>;
+	clocks = <&clk_msi>;
 	clock-frequency = <DT_FREQ_M(32)>;

```

closed 2021-09-10 by nashif